evcc-io/evcc #11548: Sunny Home Manager 2.0 / SMA Hybrid Wechselrichter: EVCC verarbeitet die DC-Eingangsleistung anstelle der AC-Ausgangsleistung des Wechselrichters

marsu81, 2024-01-07:
### Describe the bug

Ich habe EVCC als Plugin innerhalb von Home Assistant installiert und habe festgestellt, dass zur Visualisierung des Eigenverbrauchs in EVCC die Summe der DC-Eingangsleistungen aller Strings aus dem Wechselrichter verwendet werden (in meinem Fall "PV Power A" und "PV Power B"). Korrekterweise müsste zur Berechnung jedoch die AC-Ausgangsleistung des Wechselrichters ("PV Power") verwendet werden, da hier bereits die Wirkungsgradverluste des Wechselrichters berücksichtigt sind.
Das Verhalten tritt in der EVCC Version 0.123.7 und älter auf.

### Steps to reproduce

1. Erzeugung von Entitäten, welche die AC-Ausgangsleistung, die DC-Eingangsleistung von Kanal A sowie die DC-Eingangsleistung von Kanal B des Wechselrichters zurückgeben
2. Vergleich der Werte von Punkt 1 mit dem seitens EVCC angezeigten Eigenverbrauchsanteil
3. EVCC zeigt anstelle der AC-Ausgangsleistung die Summe der der DC-Eingangsleistungen von Kanal A und Kanal B als Eigenverbrauch an

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 10s # control cycle interval

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: x.x.x.x  
  name: grid1
- type: template
  template: sma-hybrid 
  id: 3  
  host: x.x.x.x  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sma-hybrid 
  id: 3  
  host: x.x.x.x  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- type: template
  template: ellipro 
  ski: xxx  
  ip: xxx  
  name: wallbox4

loadpoints:
- title: Elli Wallbox pro
  charger: wallbox4
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

hems:
  type: sma
  AllowControl: false
  
eebus:
  certificate:
    public: |
      -----BEGIN CERTIFICATE-----
      xxx
      -----END CERTIFICATE-----

    private: |
      -----BEGIN EC PRIVATE KEY-----
      xxx
      -----END EC PRIVATE KEY-----

# mqtt message broker
mqtt:
  broker: x.x.x.x:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  #clientid: 2
  user: xxx
  password: xxx
```

### Log details

```text
Da EVCC innerhalb Homeassistant läuft, kann ich evcc --log debug offenbar nicht ausführen (analog zur Erfordernis, auch die evcc.yaml auf einem "externen" System über "evcc configure" zu erzeugen).
Daher hier der Auszug aus dem log innerhalb von Homeassistant: Die zurückgegebenen Werte von pv power sind nicht korrekt, da eine vorhandene Eingangsleistung auf der DC-Seite nicht automatisch eine nutzbare Leistung auf der AC-Seite abbildet. 

[lp-1  ] DEBUG 2024/01/07 17:13:34 charge power: 0W
[site  ] DEBUG 2024/01/07 17:13:34 pv power: 6W
[site  ] DEBUG 2024/01/07 17:13:34 battery soc: 14%
[site  ] DEBUG 2024/01/07 17:13:34 battery power: 0W
[site  ] DEBUG 2024/01/07 17:13:34 grid power: 4398W
[site  ] DEBUG 2024/01/07 17:13:34 grid powers: [2476 1054 868]W
[site  ] DEBUG 2024/01/07 17:13:34 grid currents: [11.8 5.63 5.22]A
[site  ] DEBUG 2024/01/07 17:13:34 site power: 4398W
[lp-1  ] DEBUG 2024/01/07 17:13:34 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/07 17:13:34 charger status: A
[site  ] DEBUG 2024/01/07 17:13:44 ----
[lp-1  ] DEBUG 2024/01/07 17:13:44 charge power: 0W
[site  ] DEBUG 2024/01/07 17:13:44 pv power: 13W
[site  ] DEBUG 2024/01/07 17:13:44 battery soc: 14%
[site  ] DEBUG 2024/01/07 17:13:44 battery power: 0W
[site  ] DEBUG 2024/01/07 17:13:44 grid power: 4429W
[site  ] DEBUG 2024/01/07 17:13:44 grid powers: [2513 1050 866]W
[site  ] DEBUG 2024/01/07 17:13:44 grid currents: [11.9 5.68 5.28]A
[site  ] DEBUG 2024/01/07 17:13:44 site power: 4429W
[lp-1  ] DEBUG 2024/01/07 17:13:44 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/07 17:13:44 charger status: A
[site  ] DEBUG 2024/01/07 17:13:54 ----
[lp-1  ] DEBUG 2024/01/07 17:13:54 charge power: 0W
[site  ] DEBUG 2024/01/07 17:13:54 pv power: 14W
[site  ] DEBUG 2024/01/07 17:13:54 battery soc: 14%
[site  ] DEBUG 2024/01/07 17:13:54 battery power: 0W
[site  ] DEBUG 2024/01/07 17:13:54 grid power: 4463W
[site  ] DEBUG 2024/01/07 17:13:54 grid powers: [2547 1053 863]W
[site  ] DEBUG 2024/01/07 17:13:54 grid currents: [12 5.69 5.28]A
[site  ] DEBUG 2024/01/07 17:13:54 site power: 4463W
[lp-1  ] DEBUG 2024/01/07 17:13:54 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/07 17:13:54 charger status: A
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.123.7

premultiply, 2024-01-07:
Es gibt bei Hybrid-WR keine AC-Erzeugungsleistung.

closed 2024-01-07 by premultiply

marsu81, 2024-01-07:
Wenn ich mich an der Weboberfläche des WR anmelde, dann kann ich dort auf der AC-Seite die aktuelle PV-Power ablesen, im Wert 
identisch zur Anzeige der aktuellen PV-Erzeugung in der SMA App sowie auch der Anzeige der entsprechenden Entität in Home Assistant. Beim WR handelt es sich um einen SUNNY TRIPOWER 10.0 SE. Was ist das dann für ein Wert, wenn nicht die ausgehende PV-Leistung auf der AC-Seite?

[image]

premultiply, 2024-01-07:
Woher soll ich das wissen?
Schau dir gerne alle Modbus-Register in der SMA-Doku an...
Falls du einen besser passenden Wert findest lass es mich gerne wissen.
Alles mit AC scheidet bei Hybrid aber prinzipbedingt aus.

Man kann bei einem Hybrid-WR keine AC-Erzeugungsleistung angeben kann, da diese nirgendwo einzeln messbar ist. Es ist halt ein Hybrid-WR. Die einzelnen internen Leistungsflüsse über die DC/DC-Konverter lassen sich eben prinziptbedingt nur als DC-Leistung messen. Der AC-Anschluss ist eine undefinierbare Gemengelage aus PV-Erzeugung, Batterie rein/raus, Eigenverbrauch, Netzbezug usw.

marsu81, 2024-01-07:
Ok, ich werde mich mal mit der SMA-Doku auseinandersetzen. Wenn ich dort weiterkomme, melde ich mich noch mal zum Thema. Vielen Dank für die schnelle Rückmeldung & Grüße! :)

marsu81, 2024-01-08:
Guten Abend,

ich habe mir heute die SMA-Modbus Doku für den SUNNY TRIPOWER 10.0 SE angeschaut und kann das Thema jetzt nachvollziehen. Daher hier noch ein alternativer Ansatz, wie man den Wirkungsgrad und damit die AC-Seite der PV-Erzeugung & Batterie aus den bekannten, per Modbus auslesbaren Messwerten heraus annähernd berechnen könnte (ausgehend von der Annahme, dass der Netzbezug im WR keine Verluste erzeugt, was vermutlich nicht ganz korrekt ist). Voraussetzung ist, dass am WR sowohl der Netzbezug als auch die tatsächliche Last als Messwerte vorliegen (da bin ich mir nicht ganz sicher). Aus der Bilanzierung unten ergibt sich der Wirkungsgrad und damit kann man da die AC-Leistung auf Basis der DC-Eingangswerte berechnen.

Es ist richtig, dass das Ganze am Ende praktisch nicht relevant ist. Mir war es nur aufgefallen, weil innerhalb evcc auch dann ein Eigenverbrauch angezeigt wird, obwohl seitens WR gar keine Ausgangsleistung erzeugt wird (für die Zustände, wo die Einschaltschwelle für den WR auf der DC-Seite aufgrund zu geringer PV-Leistung nicht erreicht wird).

Viele Grüße,
André

[image]