evcc-io/evcc #11550: verwirrende `maxcurrent` Anzeige im UI Loadpoint Settings

VolkerK62, 2024-01-07:
Ich habe in der Config beim Fahrzeug `maxcurrent: 20` und beim Loadpoint `maxcurrent: 32` definiert.
Wenn das Fahrzeug angesteckt ist liefert `/api/state` 
- effectiveMaxCurrent: 20
- maxCurrent: 32

Wenn geladen wird, findet das richtigerweise mit max 20A statt.
Bis hierhin alles okay.

In den UI Loadpoint-Einstellungen wird jedoch `max. Ladestrom  32A` angezeigt.

[image]

Da kann ich einstellen, was ich will, solange es über 20A ist, wird es richtigerweise nicht berücksichtigt.
Das könnte zu Verwirrung führen.
Ich denke, in den Loadpoint Einstellungen sollte das angezeigt werden und auswählbar sein, was aufgrund der Kombination (Loadpoint und angestecktes Fahrzeug) möglich ist. 

In dem Zusammenhang:
Der Ladeleistungsbalken im UI ist auch auf 32A ausgerichtet.

[image]

andig, 2024-01-08:
@naltatis vielleicht in dem Dialog ein Absprung zu den Fahrzeugeinstellungen oder ggf. die Möglichkeit, entweder Fahrzeug oder Charger einstellen zu können? Beim Ladebalken müssten wir uns fragen, welches Maximum die Skalierung darstellen soll. Evtl. lieber `effectiveMaxCurrent`?

naltatis, 2024-01-08:
Ja, beim Balken unter der Ladeleistung sollten wir auf jeden Fall `effectiveMaxCurrent` verwenden.

Im Einstellungsdialog ist das wie jetzt eigentlich korrekt. Damit verstellt man ja die min/max-Currents des Ladepunkts. Aktuell nur nicht persistent und redundant zur yaml (daher 🧪). Das trägt sicher auch noch zur Verwirrung bei. Die Vehicle min/maxcurrents könnten wir auch gleich mit ins UI mirgieren, wenn wir das ändern.

Easee macht das mMn relativ gut. Die zeigen am Charger während eines Ladevorgangs alle Parameter an, die zur aktuellen Leistung beitragen:
[image]

So ausführlich brauchen wir das sicher nicht. Aber aktuell fehlen hier Infos, damit da klar ist. Vielleicht tuts hier im ersten Schritt ein textlicher Hinweis im Loadpoint-Settingsdialog oder ein Tooltip an Leistung und Stromstärke-Balken (evtl. etwas versteckt).

closed 2024-02-04 by github-actions[bot]