evcc-io/evcc #11552: Keine Restzeit bei Audi A3 Hybrid

Tedfierro, 2024-01-07:
### Describe the bug

Schon seit einigen Versionen wird mir für meinen A3 Hybrid keine Restzeit im GUI mehr angezeigt. Sie ist auch nicht auswählbar im Dropdown-Menü. 
Eine Abfrage von "evcc vehicle audi" ergibt folgendes:
[image]

### Steps to reproduce

-

### Configuration details

```yaml
vehicles:
- type: template
  template: audi
  title: Audi A3
  user: email
  password: password
  vin: WAUZZZ***********
  name: audi
  phases: 1
  capacity: 12
  cache: 10m
```

### Log details

```text
[main  ] INFO 2024/01/07 21:33:06 evcc 0.123.7
[main  ] INFO 2024/01/07 21:33:06 using config file: /etc/evcc.yaml
[main  ] INFO 2024/01/07 21:33:06 starting ui and api at :7070
[db    ] INFO 2024/01/07 21:33:06 using sqlite database: /home/pi/.evcc/evcc.db
[lp-1  ] DEBUG 2024/01/07 21:33:09 charge total import: 1452.282kWh
[lp-2  ] DEBUG 2024/01/07 21:33:09 charge total import: 1.368kWh
[site  ] WARN 2024/01/07 21:33:09 vehicle 'Lader' does not support automatic detection
[site  ] INFO 2024/01/07 21:33:09 site config:
[site  ] INFO 2024/01/07 21:33:09   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/01/07 21:33:09     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/07 21:33:09     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/07 21:33:09     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/01/07 21:33:09   vehicles:
[site  ] INFO 2024/01/07 21:33:09     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2024/01/07 21:33:09     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/01/07 21:33:09 loadpoint 1:
[lp-1  ] INFO 2024/01/07 21:33:09   mode:        off
[lp-1  ] INFO 2024/01/07 21:33:09   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/01/07 21:33:09   meters:      charge ✓
[lp-1  ] INFO 2024/01/07 21:33:09     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2024/01/07 21:33:09 loadpoint 2:
[lp-2  ] INFO 2024/01/07 21:33:09   mode:        off
[lp-2  ] INFO 2024/01/07 21:33:09   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2024/01/07 21:33:09   meters:      charge ✓
[lp-2  ] INFO 2024/01/07 21:33:09     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] DEBUG 2024/01/07 21:33:09 phase timer inactive
[lp-1  ] DEBUG 2024/01/07 21:33:09 pv timer inactive
[lp-1  ] DEBUG 2024/01/07 21:33:09 guard timer inactive
[lp-1  ] INFO 2024/01/07 21:33:09 vehicle updated: unknown -> Audi A3
[lp-2  ] DEBUG 2024/01/07 21:33:09 phase timer inactive
[lp-2  ] DEBUG 2024/01/07 21:33:09 pv timer inactive
[lp-2  ] DEBUG 2024/01/07 21:33:09 guard timer inactive
[lp-2  ] INFO 2024/01/07 21:33:09 vehicle updated: unknown -> Lader
[lp-2  ] DEBUG 2024/01/07 21:33:09 max charge current: 0.125A
[site  ] WARN 2024/01/07 21:33:09 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/01/07 21:33:09 ----
[lp-1  ] DEBUG 2024/01/07 21:33:09 charge power: 2772W
[lp-2  ] DEBUG 2024/01/07 21:33:10 charge power: 0W
[site  ] DEBUG 2024/01/07 21:33:10 pv power: 0W
[site  ] DEBUG 2024/01/07 21:33:10 battery soc: 61%
[site  ] DEBUG 2024/01/07 21:33:10 battery power: -13W
[site  ] DEBUG 2024/01/07 21:33:10 grid power: 3435W
[site  ] DEBUG 2024/01/07 21:33:10 site power: 3722W
[lp-1  ] DEBUG 2024/01/07 21:33:11 vehicle odometer: 25921km
[lp-1  ] DEBUG 2024/01/07 21:33:11 charge currents: [12 0.007 0.012]A
[lp-1  ] DEBUG 2024/01/07 21:33:11 charge total import: 1452.282kWh
[lp-1  ] DEBUG 2024/01/07 21:33:11 charger status: C
[lp-1  ] INFO 2024/01/07 21:33:11 car connected
[lp-1  ] DEBUG 2024/01/07 21:33:11 pv timer elapse
[lp-1  ] DEBUG 2024/01/07 21:33:11 pv timer inactive
[lp-1  ] DEBUG 2024/01/07 21:33:11 charger: guard elapse
[lp-1  ] DEBUG 2024/01/07 21:33:11 guard timer inactive
[lp-1  ] DEBUG 2024/01/07 21:33:11 charge total import: 1452.282kWh
[lp-1  ] INFO 2024/01/07 21:33:11 start charging ->
[lp-1  ] DEBUG 2024/01/07 21:33:11 wake-up timer: stop
[lp-1  ] DEBUG 2024/01/07 21:33:11 charger vehicle id: 04445E22D16E80
[lp-1  ] DEBUG 2024/01/07 21:33:11 vehicle status: C
[lp-1  ] DEBUG 2024/01/07 21:33:11 vehicle soc: 19%
[lp-1  ] DEBUG 2024/01/07 21:33:11 vehicle soc limit: 0%
[lp-1  ] DEBUG 2024/01/07 21:33:11 vehicle range: 8km
[lp-1  ] WARN 2024/01/07 21:33:11 charger logic error: disabled but charging
[lp-1  ] DEBUG 2024/01/07 21:33:14 set audi plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] DEBUG 2024/01/07 21:33:18 charger disable
[lp-1  ] DEBUG 2024/01/07 21:33:18 wake-up timer: stop
[site  ] DEBUG 2024/01/07 21:33:19 ----
[lp-1  ] DEBUG 2024/01/07 21:33:19 charge power: 1077W
[lp-2  ] DEBUG 2024/01/07 21:33:19 charge power: 0W
[site  ] DEBUG 2024/01/07 21:33:19 pv power: 0W
[site  ] DEBUG 2024/01/07 21:33:19 battery soc: 61%
[site  ] DEBUG 2024/01/07 21:33:19 battery power: -13W
[site  ] DEBUG 2024/01/07 21:33:19 grid power: 644W
[site  ] DEBUG 2024/01/07 21:33:19 site power: 931W
[lp-2  ] DEBUG 2024/01/07 21:33:19 charge total import: 1.368kWh
[lp-2  ] DEBUG 2024/01/07 21:33:19 charger status: A
[lp-2  ] INFO 2024/01/07 21:33:19 car disconnected
[lp-2  ] DEBUG 2024/01/07 21:33:19 set session soc limit: 0
[lp-2  ] DEBUG 2024/01/07 21:33:19 set session energy limit: 0
[lp-2  ] DEBUG 2024/01/07 21:33:19 guard enable in 4m50s
[site  ] DEBUG 2024/01/07 21:33:29 ----
[lp-1  ] DEBUG 2024/01/07 21:33:29 charge power: 0W
[lp-2  ] DEBUG 2024/01/07 21:33:29 charge power: 0W
[site  ] DEBUG 2024/01/07 21:33:29 pv power: 0W
[site  ] DEBUG 2024/01/07 21:33:29 battery soc: 61%
[site  ] DEBUG 2024/01/07 21:33:29 battery power: -13W
[site  ] DEBUG 2024/01/07 21:33:29 grid power: 651W
[site  ] DEBUG 2024/01/07 21:33:29 site power: 938W
[lp-1  ] DEBUG 2024/01/07 21:33:29 vehicle odometer: 25921km
[lp-1  ] DEBUG 2024/01/07 21:33:30 charge currents: [0.006 0.007 0.012]A
[lp-1  ] DEBUG 2024/01/07 21:33:30 charge total import: 1452.284kWh
[lp-1  ] DEBUG 2024/01/07 21:33:30 charger status: B
[lp-1  ] INFO 2024/01/07 21:33:30 stop charging <-
[lp-1  ] DEBUG 2024/01/07 21:33:30 charge total import: 1452.284kWh
[lp-1  ] DEBUG 2024/01/07 21:33:31 soc estimated: 19.01% (vehicle: 19.00%)
[lp-1  ] DEBUG 2024/01/07 21:33:31 vehicle soc: 19%
[lp-1  ] DEBUG 2024/01/07 21:33:31 vehicle soc limit: 0%
[lp-1  ] DEBUG 2024/01/07 21:33:31 vehicle range: 8km
[lp-1  ] DEBUG 2024/01/07 21:33:31 set audi plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[site  ] DEBUG 2024/01/07 21:33:39 ----
[lp-1  ] DEBUG 2024/01/07 21:33:39 charge power: 0W
[lp-2  ] DEBUG 2024/01/07 21:33:39 charge power: 0W
[site  ] DEBUG 2024/01/07 21:33:39 pv power: 0W
[site  ] DEBUG 2024/01/07 21:33:39 battery soc: 61%
[site  ] DEBUG 2024/01/07 21:33:39 battery power: -13W
[site  ] DEBUG 2024/01/07 21:33:39 grid power: 675W
[site  ] DEBUG 2024/01/07 21:33:39 site power: 962W
[lp-2  ] DEBUG 2024/01/07 21:33:40 charge total import: 1.368kWh
[lp-2  ] DEBUG 2024/01/07 21:33:40 charger status: A
[lp-2  ] DEBUG 2024/01/07 21:33:40 guard enable in 4m29s
[site  ] DEBUG 2024/01/07 21:33:49 ----
[lp-1  ] DEBUG 2024/01/07 21:33:49 charge power: 0W
[lp-2  ] DEBUG 2024/01/07 21:33:49 charge power: 0W
[site  ] DEBUG 2024/01/07 21:33:49 pv power: 0W
[site  ] DEBUG 2024/01/07 21:33:49 battery soc: 61%
[site  ] DEBUG 2024/01/07 21:33:49 battery power: -13W
[site  ] DEBUG 2024/01/07 21:33:50 grid power: 675W
[site  ] DEBUG 2024/01/07 21:33:50 site power: 962W
[lp-1  ] DEBUG 2024/01/07 21:33:50 charge currents: [0.006 0.007 0.012]A
[lp-1  ] DEBUG 2024/01/07 21:33:50 charge total import: 1452.284kWh
[lp-1  ] DEBUG 2024/01/07 21:33:50 charger status: B
[lp-1  ] DEBUG 2024/01/07 21:33:50 set audi plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[site  ] DEBUG 2024/01/07 21:33:59 ----
```

### What type of operating system are you running?

Linux

### Version

0.123.7

Tedfierro, 2024-01-08:
Vehicle log trace: 
```
[main  ] INFO 2024/01/08 12:31:40 evcc 0.123.7
[main  ] INFO 2024/01/08 12:31:40 using config file: /etc/evcc.yaml
[db    ] INFO 2024/01/08 12:31:41 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2024/01/08 12:31:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT * FROM `settings` 12 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%   device_id       %") -1 <nil>
[db    ] TRACE 2024/01/08 12:31:41 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[audi  ] TRACE 2024/01/08 12:31:41 GET https://identity.vwgroup.io/oidc/v1/authorize?client_id=f4d0934f-32bf-4ce4-b3c4-699a7049ad26%40apps_vw-dilab_com&code_challenge=hvlkArF9OViIG1e6UwbBHu7eh1iK8O7DZ7RESTpAkp0&code_challenge_method=S256&prompt=login&redirect_uri=myaudi%3A%2F%2F%2F&response_type=code+id_token+token&scope=openid+profile+mbb&ui_locales=de-DE&nonce=zLZsOVjEZyULxewllWdervQugHFkeapxaGcsWUGsZtJ&state=a9c348d6-4e7c-4709-b57d-82473d872634
[audi  ] TRACE 2024/01/08 12:31:41 GET https://identity.vwgroup.io/signin-service/v1/signin/f4d0934f-32bf-4ce4-b3c4-699a7049ad26@apps_vw-dilab_com?relayState=0387f3c6983a1077cb801173bac6d96778d100e4
[audi  ] TRACE 2024/01/08 12:31:41 <!DOCTYPE html>
<html lang="de" xmlns="http://www.w3.org/1999/html">
<head>
    <meta charset="utf-8"/>
    <meta name="viewport"
          content="width=device-width, height=device-height, initial-scale=1, maximum-scale=1, user-scalable=no"/>
    <meta name="identitykit" content="loginIdentifier"/>

<script type="text/javascript">(window.NREUM||(NREUM={})).init={privacy:{cookies_enabled:true},ajax:{deny_list:["bam.eu01.nr-data.net"]},distributed_tracing:{enabled:true}};(window.NREUM||(NREUM={})).loader_config={agentID:"535875755",accountID:"2547829",trustKey:"2217462",xpid:"VgMDVl5RARAEUlFQBQQOXlY=",licenseKey:"NRJS-ecd5929e8859410bbd9",applicationID:"460225982"};;/*! For license information please see nr-loader-spa-1.249.0.min.js.LICENSE.txt */
(()=>{var e,t,r={234:(e,t,r)=>{"use strict";r.d(t,{P_:()=>m,Mt:()=>b,C5:()=>s,DL:()=>A,OP:()=>N,lF:()=>j,Yu:()=>E,Dg:()=>v,CX:()=>c,GE:()=>x,sU:()=>R});var n=r(8632),i=r(9567);const o={beacon:n.ce.beacon,errorBeacon:n.ce.errorBeacon,licenseKey:void 0,applicationID:void 0,sa:void 0,queueTime:void 0,applicationTime:void 0,ttGuid:void 0,user:void 0,account:void 0,product:void 0,extra:void 0,jsAttributes:{},userAttributes:void 0,atts:void 0,transactionName:void 0,tNamePlain:void 0},a={};function s(e){if(!e)throw new Error("All info objects require an agent identifier!");if(!a[e])throw new Error("Info for ".concat(e," was never set"));return a[e]}function c(e,t){if(!e)throw new Error("All info objects require an agent identifier!");a[e]=(0,i.D)(t,o);const r=(0,n.ek)(e);r&&(r.info=a[e])}const u=e=>{if(!e||"string"!=typeof e)return!1;try{document.createDocumentFragment().querySelector(e)}catch{return!1}return!0};var d=r(7056),l=r(50);const f="[data-nr-mask]",h=()=>{const e={mask_selector:"*",block_selector:"[data-nr-block]",mask_input_options:{color:!1 [... cut ...]
[audi  ] TRACE 2024/01/08 12:31:42 {"query":"{userVehicles{vin,type,nickname}}"}
--
[... 139 lines cut ...]
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
##    type: tibber
##    token: *****
    #price: 0.325 # [currency]/kWh
    type: awattar
    charges: 0.1596
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0793 # price p. kWh

site:
  title: Home
  meters:
    pv: pv
    grid: grid
    battery: battery
  residualpower: 300

Version: `0.123.7`

```

andig, 2024-01-08:
> api/state:

Magst Du das mal mit jsonlint.com formatieren? Danke :)

andig, 2024-01-08:
Kann das einfach Zufall sein? Der Code für Restladedauer sieht so aus:

		var d time.Duration
		if lp.charging() {
			d = lp.socEstimator.RemainingChargeDuration(limitSoc, lp.chargePower)
		}
		lp.SetRemainingDuration(d)

...der wird aber nur aufgerufen, wenn auch gerade der Soc abgefragt wurde. Wenn Du jetzt nicht gerade lädst *und* den Soc abrufst wird das auch nicht ausgeführt. Evtl. nur Verkettung unglücklicher Umstände?

Tedfierro, 2024-01-08:
> > api/state:
> 
> Magst Du das mal mit jsonlint.com formatieren? Danke :)

Natürlich :)
```
{
    "result": {
        "auth": {
            "vehicles": {}
        },
        "battery": [
            {
                "power": -129,
                "soc": 51,
                "capacity": 11,
                "controllable": false
            }
        ],
        "batteryCapacity": 11,
        "batteryConfigured": true,
        "batteryDischargeControl": false,
        "batteryEnergy": 0,
        "batteryMode": "unknown",
        "batteryPower": -129,
        "batterySoc": 51,
        "bufferSoc": 100,
        "bufferStartSoc": 0,
        "currency": "EUR",
        "greenShareHome": 0.15502553603224695,
        "greenShareLoadpoints": 0,
        "gridConfigured": true,
        "gridPower": 3445,
        "homePower": 587.0000667572021,
        "loadpoints": [
            {
                "chargeCurrent": 16,
                "chargeCurrents": [
                    12.390000343322754,
                    0.00800000037997961,
                    0.004999999888241291
                ],
                "chargeDuration": 115000000000,
                "chargePower": 2819.999933242798,
                "chargeTotalImport": 1461.7894944444415,
                "chargedEnergy": 0,
                "chargerFeatureHeating": false,
                "chargerFeatureIntegratedDevice": false,
                "chargerIcon": null,
                "charging": true,
                "connected": true,
                "connectedDuration": 0,
                "disableThreshold": 0,
                "effectiveLimitSoc": 100,
                "effectiveMaxCurrent": 16,
                "effectiveMinCurrent": 6,
                "effectivePlanSoc": 0,
                "effectivePlanTime": "0001-01-01T00:00:00Z",
                "effectivePriority": 0,
                "enableThreshold": 0,
                "enabled": true,
                "guardAction": "inactive",
                "guardRemaining": 0,
                "limitEnergy": 0,
                "limitSoc": 0,
                "maxCurrent": 16,
                "minCurrent": 6,
                "mode": "now",
                "phaseAction": "inactive",
                "phaseRemaining": 0,
[... 116 lines cut ...]
            },
            "total": {
                "avgCo2": 152.30372915086474,
                "avgPrice": 0.1711880371544765,
                "chargedKWh": 929.2793667909256,
                "solarPercentage": 55.372088903670395
            }
        },
        "tariffFeedIn": 0.0793,
        "tariffGrid": 0.2551,
        "tariffPriceHome": 0.22784651076553095,
        "tariffPriceLoadpoints": 0.2551,
        "vehicles": {
            "audi": {
                "title": "Audi A3",
                "limitSoc": 100
            },
            "plug-garage": {
                "title": "Lader"
            }
        },
        "version": "0.123.7"
    }
}
```

andig, 2024-01-09:
Da fehlt tatsächlich `chargeRemainingDuration`. Kannst Du das bitte einfach mal beobachten ob das immer so ist oder temporär? Falls immer bräuchte ich bitte Zugangsdaten fürs Auto- das müsste ich dann ausführlich analysieren.

Tedfierro, 2024-01-10:
Das ist immer so. Im Mai/Juni als ich mit evcc loslegte noch ohne PV war eine Restzeit sichtbar.

andig, 2024-01-12:
Ich hab nur eine Idee: Zugangsdaten Auto mit info@evcc.io teilen und ich muss das Auto mal bei mir einbauen um zu schauen warum wir das nicht abrufen. Hab keine bessere Idee.

closed 2024-01-12 by andig

fL4sH3r, 2024-03-01:
Gibt es hierzu schon Neuigkeiten? Ich habe exakt das gleich Problem mit meinem VW.