evcc-io/evcc #11554: VW eGolf --> car connected fehlerhaft

bananajoe86, 2024-01-07:
### Describe the bug

EVCC zeigt dauerhaft "connected = true" und "Vehiclepresent = true". VW-ID funktioniert. Der "Plug-Status" der ID-Api funktioniert (VW-Mqtt zeigt den connect status korrekt an). 
In Evcc wird dieser Status aber wohl nicht abgerufen und Status des Vehicle bleibt dauerhaft auf present, egal ob der Stecker gesteckt ist oder nicht. 
Durch den fehlerhaften Status schaltet Evcc nicht korrekt ab, wenn das Fahrzeug nicht disconnected ist. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 228338ad890436cb326541d6d857494277eded7d521e1e70090ad271b127a919

interval: 10s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.178.108  
  name: grid1234
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.178.108  
  name: pv1234

chargers:
- type: template
  template: fritzdect 
  uri: http://192.168.178.1/  
  user: xxx
  password: xxx
  ain: xxx
  standbypower: 5  
  name: wallbox1234
  
vehicles:
- type: template
  template: id 
  title: eGolf  
  user: xxx  
  password: xxx  
  vin: xxx
  capacity: 35.8  
  name: eGolf
  #mode: pv
  #minSoC: 40
  #targetSoC: 58

loadpoints:
- title: Ladepunkt
  charger: wallbox1234
  vehicle: eGolf
  mode: pv
  soc:
    poll:
      mode: always
      interval: 60m
    estimate: true

  enable:
    threshold: -1000
    delay: 1m
  disable:
    threshold: 1000 # maximum import power (W)
    delay: 10m
  guardduration: 1m
  
site:
  title: Ladepunkt
  meters:
    grid: grid1234
    pv: pv1234

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3099 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.122 # [currency]/kWh
    
# mqtt message broker
mqtt:
  broker: 192.168.178.103:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  # user:
  # password:
```

### Log details

```text
evcc  | [main  ] INFO 2024/01/07 21:40:05 evcc 0.123.7
evcc  | [main  ] INFO 2024/01/07 21:40:05 using config file: /etc/evcc.yaml
evcc  | [main  ] INFO 2024/01/07 21:40:05 starting ui and api at :7070
evcc  | [db    ] INFO 2024/01/07 21:40:05 using sqlite database: /root/.evcc/evcc.db
evcc  | [mqtt  ] INFO 2024/01/07 21:40:05 connecting evcc-1975050073 at tcp://192.168.178.103:1883
evcc  | [lp-1  ] WARN 2024/01/07 21:40:07 poll mode '{always 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
evcc  | [lp-1  ] WARN 2024/01/07 21:40:07 phases not configured, assuming 3p
evcc  | [site  ] INFO 2024/01/07 21:40:09 site config:
evcc  | [site  ] INFO 2024/01/07 21:40:09   meters:      grid ✓ pv ✓ battery ✗
evcc  | [site  ] INFO 2024/01/07 21:40:09     grid:      power ✓ energy ✗ currents ✗
evcc  | [site  ] INFO 2024/01/07 21:40:09     pv 1:      power ✓ energy ✗ currents ✗
evcc  | [site  ] INFO 2024/01/07 21:40:09   vehicles:
evcc  | [site  ] INFO 2024/01/07 21:40:09     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
evcc  | [lp-1  ] INFO 2024/01/07 21:40:09 loadpoint 1:
evcc  | [lp-1  ] INFO 2024/01/07 21:40:09   mode:        pv
evcc  | [lp-1  ] INFO 2024/01/07 21:40:09   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
evcc  | [lp-1  ] INFO 2024/01/07 21:40:09   meters:      charge ✓
evcc  | [lp-1  ] INFO 2024/01/07 21:40:09     charge:    power ✓ energy ✓ currents ✗
evcc  | [lp-1  ] INFO 2024/01/07 21:40:09 vehicle updated: unknown -> eGolf
evcc  | [site  ] WARN 2024/01/07 21:40:09 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
evcc  | [lp-1  ] INFO 2024/01/07 21:40:12 car connected
```

### What type of operating system are you running?

Linux

### Version

0.123.7

StefanSchoof, 2024-01-08:
Der Connected status ist etwas das die Wallbox liefern muss und deine Steckdose kann das nicht.

bananajoe86, 2024-01-08:
Wäre es vorstellbar, dass das als neue Funktion in EVCC implementiert werden kann; also der Car-Status von extern kommen kann (MQTT oder ähnliches)? Das Problem ohne den korrekten Status ist, dass Evcc immer die Ladung anschmeißt, obwohl kein Fahrzeug angesteckt ist.

StefanSchoof, 2024-01-08:
Mit einem custom charger https://docs.evcc.io/docs/devices/chargers/#manuell und mqtt Plugin https://docs.evcc.io/docs/reference/plugins#mqtt-lesenschreiben kann man da was machen. Aber bedenke bitte, dass nur weil die VW API eine Verbindung meldet es nicht an der Steckdose sein muss, sondern auch an irgendeiner anderen Wallbox sein kann.

VolkerK62, 2024-01-08:
>Das Problem ohne den korrekten Status ist, dass Evcc immer die Ladung anschmeißt, obwohl kein Fahrzeug angesteckt ist.

Solange der "Verbrauch" unterhalb von `standbypower` liegt, wird keine Ladung gestartet (Status bleibt auf "B"). Im UI sollte dann "ladebereit, warte auf Fahrzeug" angezeigt werden.

andig, 2024-01-08:
Das hat mit dem Golf nichts zu tun sondern ist eine prinzipbedingte Einschränkung der Schaltsteckdosen. Der Verbindungsstatus kann nicht erkannt werden.