evcc-io/evcc #11557: Lademodus Schnell lädt nur mit max. 3,4 kW (OpenWB Pro)

TobiasHuber1980, 2024-01-08:
### Describe the bug

lädt (seit update auf 123.7?) nur noch mit 3,4 kW im Lademodus Schnell (auch bei Timer Ankunft)

### Steps to reproduce

1. evcc starten
2. lademodus schnell starten
3.
...

### Configuration details

```yaml
meters:
- type: template
  template: rct-power 
  usage: grid  
  host: 192.168.178.10  
  name: grid1
- type: template
  template: rct-power 
  usage: pv  
  host: 192.168.178.10  
  name: pv2
- type: template
  template: rct-power 
  usage: battery  
  host: 192.168.178.10  
  name: battery3
  capacity: 11.5 # in kWh
  
chargers:
- name: openwb_pro
  type: template
  template: openwb-pro
  host: 192.168.178.11 

loadpoints:
- title: Garage
  charger: openwb_pro
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
    
    
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3094 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh
    
vehicles:
  - name: my_car_1
    type: template
    template: id
    title: VW ID.5
    user: xxx
    password: xxx
    vin: xxx
    capacity: 82 # Akku in kWh
   
  - name: my_car_2
    type: template
    template: bmw
    title: BMW i3
    user: xxx
    password: xxx
    vin: xxx
    capacity: 42 # Akku in kWh
```

### Log details

```text
pi@raspberrypi:~ $ evcc --log debug
[main  ] INFO 2024/01/08 10:59:32 evcc 0.123.7 (a55ce941)
[main  ] INFO 2024/01/08 10:59:32 using config file: /etc/evcc.yaml
[main  ] INFO 2024/01/08 10:59:32 starting ui and api at :7070
[lp-1  ] DEBUG 2024/01/08 10:59:34 charge total import: 702.801kWh
[site  ] WARN 2024/01/08 10:59:34 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/01/08 10:59:34 site config:
[site  ] INFO 2024/01/08 10:59:34   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/01/08 10:59:34     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/01/08 10:59:34     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/08 10:59:34     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/01/08 10:59:34   vehicles:
[site  ] INFO 2024/01/08 10:59:34     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2024/01/08 10:59:34     vehicle 2: range ✓ finish ✗ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/01/08 10:59:34 loadpoint 1:
[lp-1  ] INFO 2024/01/08 10:59:34   mode:        pv
[lp-1  ] INFO 2024/01/08 10:59:34   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/01/08 10:59:34   meters:      charge ✓
[lp-1  ] INFO 2024/01/08 10:59:34     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/01/08 10:59:34 phase timer inactive
[lp-1  ] DEBUG 2024/01/08 10:59:34 pv timer inactive
[lp-1  ] DEBUG 2024/01/08 10:59:34 guard timer inactive
[site  ] WARN 2024/01/08 10:59:34 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/01/08 10:59:34 ----
[lp-1  ] DEBUG 2024/01/08 10:59:34 charge power: 0W
[site  ] DEBUG 2024/01/08 10:59:34 pv power: 0W
[site  ] DEBUG 2024/01/08 10:59:35 battery soc: 6%
[site  ] DEBUG 2024/01/08 10:59:35 battery power: -100W
[site  ] DEBUG 2024/01/08 10:59:35 grid power: 292W
[site  ] DEBUG 2024/01/08 10:59:35 site power: 192W
[lp-1  ] DEBUG 2024/01/08 10:59:35 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/08 10:59:35 charge total import: 702.801kWh
[lp-1  ] DEBUG 2024/01/08 10:59:35 charger status: B
[lp-1  ] DEBUG 2024/01/08 10:59:35 charge start energy: 703kWh
[lp-1  ] INFO 2024/01/08 10:59:35 car connected
[lp-1  ] DEBUG 2024/01/08 10:59:35 vehicle api refresh
[lp-1  ] DEBUG 2024/01/08 10:59:35 pv timer elapse
[lp-1  ] DEBUG 2024/01/08 10:59:35 pv timer inactive
[lp-1  ] DEBUG 2024/01/08 10:59:35 charger: guard elapse
[lp-1  ] DEBUG 2024/01/08 10:59:35 guard timer inactive
[lp-1  ] DEBUG 2024/01/08 10:59:35 charge total import: 702.801kWh
[main  ] DEBUG 2024/01/08 10:59:37 vehicle status: B (VW ID.5)
[main  ] DEBUG 2024/01/08 10:59:38 vehicle status: A (BMW i3)
[lp-1  ] INFO 2024/01/08 10:59:38 vehicle updated: unknown -> VW ID.5
[lp-1  ] DEBUG 2024/01/08 10:59:38 vehicle status: B
[lp-1  ] DEBUG 2024/01/08 10:59:38 vehicle soc: 25%
[lp-1  ] DEBUG 2024/01/08 10:59:38 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/01/08 10:59:38 vehicle range: 69km
[lp-1  ] DEBUG 2024/01/08 10:59:38 set my_car_1 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] DEBUG 2024/01/08 10:59:38 available power -192W < 4140W min 3p threshold
[lp-1  ] DEBUG 2024/01/08 10:59:38 phase scale1p in 0s
[telegram] DEBUG 2024/01/08 10:59:38 sending to ***
[lp-1  ] DEBUG 2024/01/08 10:59:38 switched phases: 1p
[lp-1  ] DEBUG 2024/01/08 10:59:38 phase timer inactive
[lp-1  ] DEBUG 2024/01/08 10:59:38 pv charge current: 0A = 0A + -0.833A (192W @ 1p)
[lp-1  ] DEBUG 2024/01/08 10:59:38 pv enable timer reset
[lp-1  ] DEBUG 2024/01/08 10:59:38 pv timer inactive
[site  ] DEBUG 2024/01/08 10:59:49 ----
[lp-1  ] DEBUG 2024/01/08 10:59:49 charge power: 0W
[site  ] DEBUG 2024/01/08 10:59:50 pv power: 0W
[site  ] DEBUG 2024/01/08 10:59:50 battery soc: 6%
[site  ] DEBUG 2024/01/08 10:59:50 battery power: -100W
[site  ] DEBUG 2024/01/08 10:59:50 grid power: 282W
[site  ] DEBUG 2024/01/08 10:59:50 site power: 182W
[lp-1  ] DEBUG 2024/01/08 10:59:50 vehicle odometer: 10349km
[lp-1  ] DEBUG 2024/01/08 10:59:50 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/08 10:59:50 charge total import: 702.801kWh
[lp-1  ] DEBUG 2024/01/08 10:59:50 charger status: B
[lp-1  ] DEBUG 2024/01/08 10:59:50 set my_car_1 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] DEBUG 2024/01/08 10:59:50 pv charge current: 0A = 0A + -0.79A (182W @ 1p)
[lp-1  ] DEBUG 2024/01/08 10:59:54 set charge mode: now
[site  ] DEBUG 2024/01/08 10:59:54 ----
[lp-1  ] DEBUG 2024/01/08 10:59:54 charge power: 0W
[site  ] DEBUG 2024/01/08 10:59:54 pv power: 0W
[site  ] DEBUG 2024/01/08 10:59:54 battery soc: 6%
[site  ] DEBUG 2024/01/08 10:59:54 battery power: -100W
[site  ] DEBUG 2024/01/08 10:59:54 grid power: 284W
[site  ] DEBUG 2024/01/08 10:59:54 site power: 183W
[lp-1  ] DEBUG 2024/01/08 10:59:54 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/08 10:59:54 charge total import: 702.801kWh
[lp-1  ] DEBUG 2024/01/08 10:59:54 charger status: B
[lp-1  ] DEBUG 2024/01/08 10:59:54 set my_car_1 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] DEBUG 2024/01/08 10:59:54 switched phases: 3p
[lp-1  ] DEBUG 2024/01/08 10:59:54 max charge current: 16A
[lp-1  ] DEBUG 2024/01/08 10:59:55 charger enable
[lp-1  ] DEBUG 2024/01/08 10:59:55 wake-up timer: start
[site  ] DEBUG 2024/01/08 11:00:04 ----
[lp-1  ] DEBUG 2024/01/08 11:00:04 charge power: 181W
[site  ] DEBUG 2024/01/08 11:00:04 pv power: 0W
[site  ] DEBUG 2024/01/08 11:00:05 battery soc: 6%
[site  ] DEBUG 2024/01/08 11:00:05 battery power: -100W
[site  ] DEBUG 2024/01/08 11:00:05 grid power: 289W
[site  ] DEBUG 2024/01/08 11:00:05 site power: 189W
[lp-1  ] DEBUG 2024/01/08 11:00:05 charge currents: [0.515 0.474 0.577]A
[lp-1  ] DEBUG 2024/01/08 11:00:05 charge total import: 702.801kWh
[lp-1  ] DEBUG 2024/01/08 11:00:05 charger status: C
[lp-1  ] DEBUG 2024/01/08 11:00:05 charge start energy: 703kWh
[lp-1  ] INFO 2024/01/08 11:00:05 start charging ->
[lp-1  ] DEBUG 2024/01/08 11:00:05 wake-up timer: stop
[telegram] DEBUG 2024/01/08 11:00:05 sending to ***
[lp-1  ] DEBUG 2024/01/08 11:00:05 soc estimated: 25.00% (vehicle: 25.00%)
[lp-1  ] DEBUG 2024/01/08 11:00:05 vehicle soc: 25%
[lp-1  ] DEBUG 2024/01/08 11:00:05 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/01/08 11:00:05 vehicle range: 69km
[lp-1  ] DEBUG 2024/01/08 11:00:05 set my_car_1 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[site  ] DEBUG 2024/01/08 11:00:19 ----
[lp-1  ] DEBUG 2024/01/08 11:00:19 charge power: 3415W
[site  ] DEBUG 2024/01/08 11:00:20 pv power: 0W
[site  ] DEBUG 2024/01/08 11:00:20 battery soc: 6%
[site  ] DEBUG 2024/01/08 11:00:20 battery power: -100W
[site  ] DEBUG 2024/01/08 11:00:20 grid power: 3719W
[site  ] DEBUG 2024/01/08 11:00:20 site power: 3618W
[lp-1  ] DEBUG 2024/01/08 11:00:20 charge currents: [5.03 5.08 5.06]A
[lp-1  ] DEBUG 2024/01/08 11:00:20 charge total import: 702.809kWh
[lp-1  ] DEBUG 2024/01/08 11:00:20 charger status: C
[lp-1  ] DEBUG 2024/01/08 11:00:20 soc estimated: 25.01% (vehicle: 25.00%)
[lp-1  ] DEBUG 2024/01/08 11:00:20 vehicle soc: 25%
[lp-1  ] DEBUG 2024/01/08 11:00:20 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/01/08 11:00:20 vehicle range: 69km
[lp-1  ] DEBUG 2024/01/08 11:00:20 set my_car_1 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
```

### What type of operating system are you running?

Linux

### Version

a55ce941

andig, 2024-01-08:
...und du hast nicht zufällig noch im UI für Fahrzeug oder WB eine Strombegrenzung konfiguriert?

TobiasHuber1980, 2024-01-08:
{"result":{"auth":{"vehicles":{}},
"battery":[{"power":-100.08904266357422,
"soc":6.193016096949577,"capacity":11.5,
"controllable":false}],
"batteryCapacity":11.5,
"batteryConfigured":true,
"batteryDischargeControl":false,
"batteryEnergy":0,"batteryMode":"unknown",
"batteryPower":-100.08904266357422,
"batterySoc":6.193016096949577,
"bufferSoc":100,"bufferStartSoc":0,
"currency":"EUR","greenShareHome":0,
"greenShareLoadpoints":0,
"gridConfigured":true,"gridEnergy":0,
"gridPower":3742.531982421875,
"homePower":242.1429397583006,
"loadpoints":[{"chargeCurrent":16,
"chargeCurrents":[5.062,5.055,5.062],
"chargeDuration":15000000000,
"chargePower":3400.3,
"chargeRemainingDuration":53041000000000,
"chargeRemainingEnergy":50099.11111111117,
"chargeTotalImport":702.895,
"chargedEnergy":11.999999999943611,
"chargerFeatureHeating":false,
"chargerFeatureIntegratedDevice":false,
"chargerIcon":null,"charging":true,
"connected":true,
"connectedDuration":0,
"disableThreshold":0,
"effectiveLimitSoc":100,
"effectiveMaxCurrent":16,
"effectiveMinCurrent":6,
"effectivePlanSoc":0,
"effectivePlanTime":"0001-01-01T00:00:00Z",
"effectivePriority":0,
"enableThreshold":0,
"enabled":true,
"guardAction":"inactive",
"guardRemaining":0,
"limitEnergy":0,
"limitSoc":0,
"maxCurrent":16,
"minCurrent":6,
"mode":"now",
"phaseAction":"inactive",
"phaseRemaining":0,
"phasesActive":3,
"phasesConfigured":0,
"phasesEnabled":3,
"planEnergy":0,
"planProjectedStart":"0001-01-01T00:00:00Z",
"planTime":"0001-01-01T00:00:00Z",
"priority":0,"pvAction":"inactive",
"pvRemaining":0,
"sessionCo2PerKWh":null,
"sessionEnergy":11.999999999943611,
"sessionPrice":0.0037127999999825535,
"sessionPricePerKWh":0.3094,
"sessionSolarPercentage":0,
"title":"Garage",
"vehicleCapacity":82,
"vehicleClimaterActive":null,
"vehicleDetectionActive":false,
"vehicleFeatureOffline":false,
"vehicleIcon":"car",
"vehicleName":"my_car_1",
"vehicleOdometer":10349,
"vehiclePresent":true,
"vehicleRange":70,
"vehicleSoc":25.013170731707255,
"vehicleTargetSoc":80,
"vehicleTitle":"VW ID.5"}],
"prioritySoc":95,
"pv":[{"power":0}],
"pvConfigured":true,
"pvEnergy":0,
"pvPower":0,
"residualPower":0,
"siteTitle":"Zuhause",
"smartCostActive":false,
"smartCostLimit":0,
"smartCostType":
"pricestatic",
"sponsor":"TobiasHuber1980",
"sponsorTokenExpires":0,
"statistics":{"30d":{"avgCo2":0,
"avgPrice":0.21591884574343265,
"chargedKWh":196.43100000000027,
"solarPercentage":40.75028520338598},
"365d":{"avgCo2":0,
"avgPrice":0.13833799576583095,
"chargedKWh":1485.5906176755832,
"solarPercentage":76.98056736641009},
"total":{"avgCo2":0,
"avgPrice":0.13833799576583095,
"chargedKWh":1485.5906176755832,
"solarPercentage":76.98056736641009}},
"tariffFeedIn":0.08,
"tariffGrid":0.3094,
"tariffPriceHome":0.3094,
"tariffPriceLoadpoints":0.3094,
"vehicles":{"my_car_1":{"title":"VW ID.5"},
"my_car_2":{"title":"BMW i3","minSoc":75}},
"version":"0.123.7 (a55ce941)"}}

TobiasHuber1980, 2024-01-08:
> ...und du hast nicht zufällig noch im UI für Fahrzeug oder WB eine Strombegrenzung konfiguriert? Bitte zeig mal die Ausgabe von /api/state wenn im im Schnelllademodus.

weder noch, keine Strombegrenzung.

TobiasHuber1980, 2024-01-08:
update:
auto abgesteckt, neu angesteckt..... geht wieder normal.... 
closed.

closed 2024-01-08 by TobiasHuber1980