evcc-io/evcc #11558: Warning "vehicle soc: EOF (ignored by estimator)" when Porsche API is unable to return data

MarkusGH, 2024-01-08:
### Describe the bug

There are situations where the porsche API is unable to return data (see log).
It seems the error handling for this case inside the porsche API code is not sufficient, so the error bubbles up to the loadpoint.
Possible improvements:
1) Provide better error message
2) Try to wake up car (request current vehicle status)

### Steps to reproduce

Unfortunately this is not easy to reproduce, but the log should provide enough information.

### Configuration details

```yaml
chargers:
- name: openwb1
  type: openwb
  phases1p3p: true
  broker: openwb.xxx.de # openWB IP
  id: 1                      # loadpoint ID
- name: openwb2
  type: openwb
  phases1p3p: true
  broker: openwb.xxx.de # openWB IP
  id: 2                      # loadpoint ID

loadpoints:
- title: Carport
  charger: openwb2
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 15m
    estimate: true
  phases: 0
  mincurrent: 6
  maxcurrent: 20
  enable:
    threshold: -600
    delay: 1m
  disable:
    threshold: 800
    delay: 3m
  guardduration: 5m

vehicles:
- name: ev1
  type: template
  template: porsche
  title: xxx
  user: xxx
  password: xxx
  vin: xxx
  capacity: 17.9  # Akku-Kapazität in kWh
  phases: 2
  minCurrent: 6
  maxCurrent: 16
  cache: 15m
```

### Log details

```text
Jan  7 19:03:40 homebridge evcc[2000978]: [porsche] TRACE 2024/01/07 19:03:40 GET https://api.porsche.com/e-mobility/de/de_DE/E3/***?tim
ezone=Europe/Berlin
Jan  7 19:03:41 homebridge evcc[2000978]: [porsche] TRACE 2024/01/07 19:03:41 {
Jan  7 19:03:41 homebridge evcc[2000978]:   "batteryChargeStatus" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:   "directCharge" : {
Jan  7 19:03:41 homebridge evcc[2000978]:     "disabled" : true,
Jan  7 19:03:41 homebridge evcc[2000978]:     "isActive" : false
Jan  7 19:03:41 homebridge evcc[2000978]:   },
Jan  7 19:03:41 homebridge evcc[2000978]:   "directClimatisation" : {
Jan  7 19:03:41 homebridge evcc[2000978]:     "climatisationState" : "UNKNOWN",
Jan  7 19:03:41 homebridge evcc[2000978]:     "remainingClimatisationTime" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:     "targetTemperature" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:     "climatisationWithoutHVpower" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:     "heaterSource" : null
Jan  7 19:03:41 homebridge evcc[2000978]:   },
Jan  7 19:03:41 homebridge evcc[2000978]:   "timers" : [ ],
Jan  7 19:03:41 homebridge evcc[2000978]:   "climateTimer" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:   "chargingProfiles" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:   "departureInformation" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:   "errorInfo" : [ {
Jan  7 19:03:41 homebridge evcc[2000978]:     "pcckErrorKey" : "ERROR_RETRIEVAL_DEPARTURE_TIMER",
Jan  7 19:03:41 homebridge evcc[2000978]:     "pcckErrorMessage" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:     "pcckErrorCode" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:     "pcckIsBusinessError" : false
Jan  7 19:03:41 homebridge evcc[2000978]:   }, {
Jan  7 19:03:41 homebridge evcc[2000978]:     "pcckErrorKey" : "ERROR_RETRIEVAL_BATTERY_CHARGING_INFO",
Jan  7 19:03:41 homebridge evcc[2000978]:     "pcckErrorMessage" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:     "pcckErrorCode" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:     "pcckIsBusinessError" : false
Jan  7 19:03:41 homebridge evcc[2000978]:   }, {
Jan  7 19:03:41 homebridge evcc[2000978]:     "pcckErrorKey" : "ERROR_RETRIEVAL_CLIMATISATION_STATUS",
Jan  7 19:03:41 homebridge evcc[2000978]:     "pcckErrorMessage" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:     "pcckErrorCode" : null,
Jan  7 19:03:41 homebridge evcc[2000978]:     "pcckIsBusinessError" : false
Jan  7 19:03:41 homebridge evcc[2000978]:   } ]
Jan  7 19:03:41 homebridge evcc[2000978]: }
Jan  7 19:03:41 homebridge evcc[2000978]: [porsche] TRACE 2024/01/07 19:03:41 GET https://api.porsche.com/vehicle-data/de/de_DE/status/***
Jan  7 19:03:41 homebridge evcc[2000978]: [lp-1  ] WARN 2024/01/07 19:03:41 vehicle soc: EOF (ignored by estimator)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.123.7

andig, 2024-01-08:
> It seems the error handling for this case inside the porsche API code is not sufficient, so the error bubbles up to the loadpoint.

Inwiefern? Fehler werden zurück gegeben- es gibt im Code nichts, dass sie abfangen würde.

MarkusGH, 2024-01-09:
Genau - es gibt im Code nichts was sie abfangen würde. So wie ich das verstehe, entspricht die Antwort des Servers nicht dem, was erwartet wird und deswegen kommt beim Dekodieren ein Fehler, der dann an den Loadpoint durchgereicht wird. Das führt zu einer sehr "generischen" Fehlermeldung ("EOF" - welcher Anwender soll damit etwas anfangen?). Besser wäre aus meiner Sicht, wenn die Situation im Porsche API abgehandelt wird - zum Beispiel durch Erkennen und beheben des Problems, wenigstens eine besserer Fehlermeldung ("Unexpected answer from Porsche API"?).

andig, 2024-01-12:
Das ist ein generelles Thema für alle Schnittstellen für das wir aktuell keine gute Lösung haben, insbesondere da die Fehler zwischendurch nochmal "gewrapped" werden. Leider erstmal wontfix.

closed 2024-01-12 by andig

MarkusGH, 2024-01-12:
Kein Problem für mich - wollte es nur dokumentieren.

rogatec, 2025-08-05:
@andig da ich das Problem heute mit meinem Tesla habe/hatte - ich vermute mal das ist weiterhin ein generelles Problem mit `wontfix` status? 

`vehicle soc: timeout (ignored by estimator)`