evcc-io/evcc #11562: SMA Hybrid falsche ausgelesene Werte der PV Erzeugung

n0cturne, 2024-01-08:
### Describe the bug

Nach dem Update auf EVCC 0.123.7 sind die Warte aus dem Hybridwechselrichter falsch, Laut den Werten geht mehr in die Batterie als beide Wechselrichter generieren:
[image]
Laut SMA wird aber deutlich mehr generit:
[image]

Problem tritt bei der Version 0.123.5 nicht auf, könnte daher mit der Änderung [SMA: add battery control](https://github.com/evcc-io/evcc/pull/11368) zusammen hängen.

### Steps to reproduce

1. Update auf 0.123.7
2. Problem liegt direkt nach Systemstart an
3. Downgrade auf 0.123.5 und Problem liegt nicht mehr an
...

### Configuration details

```yaml
PV Anlage:
   SMA Sunny Homemanager 2.0
   SMA Sunny Tripower 5.0 
   SMA Sunny Tripower 10 SE 
   BYD HVS 7.7kWh

Wallboxen
   2x Heidelberg Energy Control
   Eastron SDM72D-M 

EVCC Server
   Raspberry Pi 4 Model B Rev 1.5 -> Raspbian GNU/Linux 11 (bullseye)
```

### Log details

```text
evcc meter bei 0.123.7:
[main  ] INFO 2024/01/08 13:02:01 evcc 0.123.7
[main  ] INFO 2024/01/08 13:02:01 using config file: /etc/evcc.yaml
[db    ] INFO 2024/01/08 13:02:01 using sqlite database: /home/pi/.evcc/evcc.db
grid1
-----
Power:          4W
Energy:         2591.9kWh
Current L1..L3: 1.06A 0.876A -0.881A
Voltage L1..L3: 235V 235V 236V
Power L1..L3:   48.2W 2.3W -46.9W

pv1
---
Power:  791W
Energy: 8752.4kWh

battery1
--------
Power:  -1876W
Energy: 959.7kWh
Soc:    34%

pv2
---
Power:          760W
Energy:         4330.6kWh
Current L1..L3: 1.14A 1.13A 1.14A
Voltage L1..L3: 233V 232V 233V
Power L1..L3:   253W 253W 254W

my_charger
----------
Power:  0W
Energy: 1772.6kWh

evcc meter bei 0.123.7:
[main  ] INFO 2024/01/08 12:59:00 evcc 0.123.5
[main  ] INFO 2024/01/08 12:59:00 using config file: /etc/evcc.yaml
[db    ] INFO 2024/01/08 12:59:00 using sqlite database: /home/pi/.evcc/evcc.db
grid1
-----
Power:          -9W
Energy:         2591.9kWh
Current L1..L3: 1.02A 0.868A -0.884A
Voltage L1..L3: 236V 236V 237V
Power L1..L3:   36.5W 0.1W -45.6W

pv1
---
Power: 1644W

battery1
--------
Power: -1961W
Soc:   32%

pv2
---
Power:          777W
Energy:         4330.6kWh
Current L1..L3: 1.15A 1.15A 1.16A
Voltage L1..L3: 234V 233V 234V
Power L1..L3:   259W 258W 260W

my_charger
----------
Power:  0W
Energy: 1772.6kWh
```

### What type of operating system are you running?

Linux

### Version

0.123.7

premultiply, 2024-01-08:
Ok, da ist offensichtlich die PV-Erzeugung falsch. :(
Da ich bei mir nur einen String am STP belegt habe konnte ich das nicht richtig validieren.
Mit einem hat es gestimmt.

buennerbernd, 2024-01-08:
Ich habe das Problem auch mit dem STP 10.0
Bekomme zu wenig PV-Erzeugung, z.B. 89 statt 4498

n0cturne, 2024-01-08:
was mir auch aufgefallen ist, dass ein SMA STP SE (pv1) anscheinend weniger Informationen liefert als ein "normaler" SMA STP (pv2). In der neuen Version (0.123.7) zumindest Power und Energy:
  _Jan 08 13:01:07 raspberrypi evcc[23149]: [main  ] INFO 2024/01/08 13:01:07 evcc 0.123.7
  Jan 08 13:01:13 raspberrypi evcc[23149]: [site  ] INFO 2024/01/08 13:01:13     pv 1:      power ✓ energy ✓ currents ✗
  Jan 08 13:01:13 raspberrypi evcc[23149]: [site  ] INFO 2024/01/08 13:01:13     pv 2:      power ✓ energy ✓ currents ✓_

Damals (0.123.5) lediglich nur Power:
  _Jan 08 13:16:25 raspberrypi evcc[23666]: [main  ] INFO 2024/01/08 13:16:25 evcc 0.123.5
  Jan 08 13:16:30 raspberrypi evcc[23666]: [site  ] INFO 2024/01/08 13:16:30     pv 1:      power ✓ energy ✗ currents ✗
  Jan 08 13:16:30 raspberrypi evcc[23666]: [site  ] INFO 2024/01/08 13:16:30     pv 2:      power ✓ energy ✓ currents ✓_

andig, 2024-01-08:
@premultiply das dürfte hier dran liegen: https://github.com/evcc-io/evcc/pull/11368/files#diff-0526d3b583173fd1a1d2cff4ec8ad8dc1401205821f9c41a852e09181515b666R28. Ist mir beim Review durch gerutscht- mir ist nicht ganz klar, warum du das geändert hast?

premultiply, 2024-01-08:
Weil es auch WR mit mehr als zwei MPPTs und das Template das so nicht abbildete.

premultiply, 2024-01-08:
Fixed by https://github.com/evcc-io/evcc/commit/2ecb4df4521744435ed599d4602cd06b5499771a

closed 2024-01-08 by premultiply

buennerbernd, 2024-01-09:
Ich kenne die Internals von EVCC nicht, aber 
die Summe der DC Werte ist immer größer als was AC seitig herauskommt.
Wenn die PV überdimensioniert ist kommt es bei Volllast auch mal zu größeren Differenzen.

premultiply, 2024-01-09:
Der Unterschied beträgt nur die Verlustleistung des WR.
!! Bei allen Hybrid-WR nicht anders möglich !!

Das hat nichts mit Überbelegung etc. zu tun.

Der Bug das beim SMA STP SE nur MPPT1 berücksichtigt wurde ist im Nightly gefixt.

Uhrenradio, 2024-01-10:
Problem liegt bei mir auch an: SMA STP Hybrid-Wechselrichter zeigt in EVCC zu wenig PV-Leistung an. PV-Meter falsch.
Battery-Meter ist korrekt.
Ebenfalls passt der Hausverbrauch welcher aus dem Grid-Meter nicht: Zu gering.

Wenn ich es korrekt verstehe muss ich auf das nächste Release warten, korrekt?

Braucht jemand meine config und ein Trace zur Fehlersuche?

premultiply, 2024-01-10:
Aktuelles Nightly verwenden.

Uhrenradio, 2024-01-10:
> Aktuelles Nightly verwenden.

Check! In meinem Fall werde ich auf das nächste offizielle Release warten (müssen): Mein EVCC läuft auf einem Home-Assistant-System.

andig, 2024-01-10:
Der Fluch von HA ;)

buennerbernd, 2024-01-10:
> Der Unterschied beträgt nur die Verlustleistung des WR. !! Bei allen Hybrid-WR nicht anders möglich !!
> 
> Das hat nichts mit Überbelegung etc. zu tun.
> 
> Der Bug das beim SMA STP SE nur MPPT1 berücksichtigt wurde ist im Nightly gefixt.

Ich habe mich nur gewundert, weil der WR doch auch seine AC Leistung ausgibt.
Ich hatte aber nur den Betrieb ohne Speicher im Kopf. Da finde ich das simpler.

premultiply, 2024-01-10:
Mag sein, aber einen Hybrid kauf man eigentlich nur freiwillig wenn man genau in das eng definierte Nutzungsszenario (mit Speicher) fällt.

Uhrenradio, 2024-01-10:
Stichwort Verlustleistung:
Rein theoretisch könnte man hier noch per Offset (Eigenverbrauch WR) und Prozent-Wert für die Verlustleistung optimieren um vom PV-Ertrag auf AC umzurechnen. Wäre aber wahrscheinlich „Over-Engineered“.

Dann lieber etwas mehr ResidualPower konfigurieren und alles ist gut…: Oder habe ich da einen Denkfehler?

premultiply, 2024-01-11:
Der PV-Ertrag spielt für die Steuerung keine Rolle. Ein reiner Anzeigewert für das gute Gefühl.

Bei allen Hybrid-WR nicht anders als via DC möglich.

Wer unbedingt die AC-Leistung verwenden will darf sich keine Hybrid-WR installieren.

buennerbernd, 2024-01-11:
Du hast Recht, für das gute Gefühl ist muss man keine Korrektur einbauen. Da ist der WR einfach ein weiterer Verbraucher im Haus.

Nur aus Interesse. Wenn man noch keinen Speicher hat, würde dann auch ein Template für Nicht-Hybrid-SMA-WR funktionieren?

premultiply, 2024-01-11:
Ja.