evcc-io/evcc #11566: GUI: Loadpoint Aus geht wieder auf PV

C64Axel, 2024-01-08:
### Describe the bug

Wenn ich im GUI das Laden von PV auf AUS stelle wird automatisch wieder PV eingestellt.

### Steps to reproduce

1. Laden PV eingestellt
2. Tippen auf AUS
3. Laden PV wird wieder eingestellt
...

### Configuration details

```yaml
# log: debug

sponsortoken: XXXX
plant: XXXX
site:
  title: Zu Hause
  meters:
    grid: Hauptzaehler
    pv: sun2000

loadpoints:
  - title: Carport
    charger: autoladen
    vehicle: Mazda_CX60
    mode: pv
    phases: 1
    soc:
      poll:
        mode: connected
        interval: 10m

chargers:
  - name: autoladen
    type: template
    template: tasmota
    host: autoladen-0580
    user: admin
    password: XXXX
    channel: 1
    standbypower: 5

meters:
  - name: Hauptzaehler
    type: custom
    power:
      source: mqtt
      topic: stromzaehler/tele/SENSOR
      jq: .SML.Power_curr
  - name: sun2000
    type: template
    template: huawei-dongle
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.200.27
    port: 502
  - name: solarlog
    type: template
    template: solarlog
    usage: pv
    host: 192.168.50.5
    port: 502

vehicles:
  - name: Mazda_CX60
    type: template
    template: mazda2mqtt
    title: Mazda CX60
    vin: XXXX
    phases: 1
    capacity: 17.8
    mode: pv
    timeout: 720h

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.333
  feedin:
    type: fixed
    price: 0.08

mqtt:
  broker: mosquitto:1883
  topic: evcc
  user: XXXX
  password: XXXX

influx:
  url: http://influxdb:8086
  database: evcc
  token: XXXX
  org: Hausautomation

modbusproxy:
  - port: 5021
    uri: 192.168.200.27:502
```

### Log details

```text
[lp-1  ] INFO 2024/01/08 17:53:23 car disconnected
[lp-1  ] INFO 2024/01/08 17:53:24 car connected
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.123.7

naltatis, 2024-01-08:
Magst du mal nen größeren Logauszug mir reinnehmen in dem man die Modusumschaltungen auch sehen kann?

VolkerK62, 2024-01-08:
Das liegt am charger (schaltbare steckdose).
"aus" ist wie Auto abstecken und dann greift der Modus vom Loadpoint. Also "pv". Das ist dann wieder wie Auto anstecken.

C64Axel, 2024-01-08:
Danke Dir, hat funktioniert.
Hat aber in älteren Versionen auch schon mal mit den Einträgen geklappt.

closed 2024-01-08 by C64Axel

VolkerK62, 2024-01-08:
Ja. Das mit dem "aus" bei schlatbaren Steckdosen ist reingekommen um damit eine session beenden zu können. Was ja vorher nur durch neustart möglich war.
Und dann ist durch 0.123.0 `resetondisconnect` rausgeflogen und praktisch durch loadpoint-mode ersetzt worden.
Das führt dann zu dem beschriebenen Effekt.