evcc-io/evcc #11569: VW (offline). Keine information vom Auto über Ladezustand

f-lehmann-gmx-de, 2024-01-08:
### Describe the bug

Ich benutze EVCC im Docker und hatte die Version latest installiert was zu dem Zeitpunkt die 0.123.1 war. Alles lief soweit ok. Als ich sah das es ein neues latest gibt hatte ich die version aktualisiert. Danach konnte sich EVCC nicht mehr mit dem Auto verbinden um den SOC abzufragen.
Ich hatte nichts an der Konfiguration geändert nur die version von evcc erneuert. Was mich wundert. In meiner alten config musste ich für VW einmal template beim vehicle angeben "id" angeben. In der aktuellen doc steht bei VW das dort "vw" stehen soll. Ich habe beides ausprobiert ohne Erfolg. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.1.35
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
# modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
[... 347 lines cut ...]
2024/01/08 20:55:14	stdout	[site  ] INFO 2024/01/08 20:55:14   vehicles:
2024/01/08 20:55:14	stdout	[site  ] INFO 2024/01/08 20:55:14     pv 1:      power ✓ energy ✓ currents ✗
2024/01/08 20:55:14	stdout	[site  ] INFO 2024/01/08 20:55:14     grid:      power ✓ energy ✓ currents ✗
2024/01/08 20:55:14	stdout	[site  ] INFO 2024/01/08 20:55:14   meters:      grid ✓ pv ✓ battery ✗
2024/01/08 20:55:14	stdout	[site  ] INFO 2024/01/08 20:55:14 site config:
2024/01/08 20:55:14	stdout	[lp-1  ] DEBUG 2024/01/08 20:55:14 charge total import: 1488.428kWh
2024/01/08 20:55:14	stdout	[main  ] ERROR 2024/01/08 20:55:14 failed configuring co2 tariff: cannot create tariff type 'grünstromindex': Get "https://api.corrently.io/v2.0/gsi/prediction?zip=66424": dial tcp: lookup api.corrently.io: i/o timeout
2024/01/08 20:55:14	stdout	[gsi   ] ERROR 2024/01/08 20:55:14 Get "https://api.corrently.io/v2.0/gsi/prediction?zip=***": dial tcp: lookup api.corrently.io: i/o timeout
2024/01/08 20:54:19	stdout	[lp-1  ] WARN 2024/01/08 20:54:19 phases not configured, assuming 3p
2024/01/08 20:54:19	stdout	[lp-1  ] WARN 2024/01/08 20:54:19 poll mode '{always 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
2024/01/08 20:54:19	stdout	[main  ] ERROR 2024/01/08 20:54:19 creating vehicle id3 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': Get "https://emea.bff.cariad.digital/user-login/v1/authorize?client_id=a24fba63-34b3-4d43-b181-942111e6bda8%40apps_vw-dilab_com&code_challenge=Zw3yI6cakGwiY7j88BaJmgKS7A1c7xYc8d5CTYKBqZ0&code_challenge_method=S256&redirect_uri=weconnect%3A%2F%2Fauthenticated&response_type=code+id_token+token&scope=openid+profile+badge+cars+vin&nonce=poTnEfgAuOZNojyHCtKjFkZzaHSMKeUskCxoBGHADYI&state=c1961b2f-749b-44fe-aa53-34b41eefe44e": dial tcp: lookup emea.bff.cariad.digital: i/o timeout
2024/01/08 20:54:13	stdout	[mqtt  ] DEBUG 2024/01/08 20:54:13 tcp://192.168.1.35:1883 connected
2024/01/08 20:54:13	stdout	[mqtt  ] INFO 2024/01/08 20:54:13 connecting evcc at tcp://192.168.1.35:1883
2024/01/08 20:54:13	stdout	[main  ] INFO 2024/01/08 20:54:13 starting ui and api at :7070
2024/01/08 20:54:13	stdout	[main  ] INFO 2024/01/08 20:54:13 using config file: /etc/evcc.yaml
2024/01/08 20:54:13	stdout	[main  ] INFO 2024/01/08 20:54:13 evcc 0.123.7
```

### What type of operating system are you running?

Docker container

### Version

0.123.7

f-lehmann-gmx-de, 2024-01-08:
Ergänzung. Ich hatte auch in myVolkswagen login geguckt ob es neue AGB oder ähnliches gibt und habe mich dort eingeloggt und alle sachen nochmal approved und danach den docker neu gestartet. Halt leider nichts geholfen. Hatte ich probiert, da ich mehrfach in alten threads gelesen habe das diese geholfen hätte. Mich irritiert das im log immer i/o timeout steht. 
ich hatte auch beim vehicle beim template (id) versucht wie ich es zuvor hatte. die MEldung bleibt die gleiche mit dem I/o timeout

0x3d13f, 2024-01-08:
Sowohl für grünstromindex als auch für das Auto klappt die DNS-Auflösung nicht:
`dial tcp: lookup api.corrently.io: i/o timeout`
`dial tcp: lookup emea.bff.cariad.digital: i/o timeout`
Da wird wohl eher bei Dir lokal etwas in der Richtung kaputt sein.

0x3d13f, 2024-01-08:
Für alles andere hast Du ja direkt IPs in der Config, da schlägt der Fehler nicht zu.

f-lehmann-gmx-de, 2024-01-08:
wenn ich mich auf den docker mit der shell verbinde und dort ping emea.bff.cariad.digital bekomme ich die IP adresse. wenn es was mit meinem dns wäre, dann müsste ich dort auch ein timeout bekommen. oder übersehe ich etwas?

f-lehmann-gmx-de, 2024-01-08:
welcher Wert beim template beim vehicle ist richtig?

Wie in der docu beschrieben "vw" oder wie es zuvor erfolgreich in meiner yaml datei hatte mit "id"?

0x3d13f, 2024-01-08:
Es gibt nur ein Template `vw`, wenn ich den Code richtig verstehe, kann `id` als Alias verwendet werden.
Ich habe evcc in einer "klassischen" VM laufen, daher kann ich Dir bei dem DNS Problem im Container nicht so gut weiter helfen.

0x3d13f, 2024-01-08:
Vielleicht einfach mal den ganzen Container wegwerfen und neu bauen? Die evcc.yaml und die evcc.db sollten ja eh außerhalb liegen und erst beim Start reingemountet werden.

f-lehmann-gmx-de, 2024-01-08:
Das habe ich schon probiert. config yaml und .evcc folder für die sqlite db liegen außerhalb und werden reingemouted. habe akutell 5 verschiedne container versucht. Die version die zuvor mal ging, und dann latest, 0.123.7 was gleich ist wie aktuell latest und auch eine 0.123.5. Überall das gleiche. Auch eine ältere version 0.121.1 Die beschwert sich nur über ein setting beim grid den es in der Version noch nicht gab. Aber danke für den Tip. Ein Versuch war es wert. Leider ohne Erfolg

f-lehmann-gmx-de, 2024-01-08:
der hint mit der dns werde ich trotzdem mal nachgehen. Eigentlich dürfte dann in der console im container ein ping auch nicht funktionieren. aber es ist in der tat auffällig das im log eigentlich überall wo die ziel-adresse als name drin steht ein timeout steht. werde mir mal die nächsten tage meine firewall von ubiquity ansehen ob da etwas zu sehen ist oder geblockt wird

andig, 2024-01-08:
Bitte ohne Docker reproduzieren wie im Issue-Template beschrieben. Danke!

closed 2024-01-08 by andig

TobiasHuber1980, 2024-01-09:
also wenn du einen ID hast, dann ist auch das "ID" das richtige template....

f-lehmann-gmx-de, 2024-01-10:
Finales update. Ich bin dem punkt mit dem DNS gefolgt und das scheint wirklich die Ursache gewesen zu sein. Aus irgendeinem Grund musste ich die Konfiguration von meinem Synology NAS nochmal neu setzen. Danach ging es und die ganzen Meldungen mit I/O timeout sind verschwunden und es ging wieder
@0x3d13f Vielen Dank für deine Konstruktiven Nachfragen. Dadurch bin ich auf das Problem gestoßen und konnte es lösen.