evcc-io/evcc #11573: Tooltip bei "Min. Ladung" in "Ladeplanung" --> "Ankunft" ergibt bei Einstellung von "---" wenig Sinn

coderph0x, 2024-01-08:
### Describe the bug

Wenn man in der Ladeplanung im Reiter "Ankunft" die Mindestladung deaktiviert (also "---" auswählt), sagt der Tooltip:
```
Fahrzeug wird im PV-Modus „schnell“ auf x% geladen. Danach weiter mit PV-Überschuss. Nützlich, um auch an dunklen Tagen eine Mindestreichweite zu gewährleisten.
```
Hier sollte womöglich eher ein Hinweis auf die Deaktivierung der Mindestladung stehen, zumindest aber das `x%` geändert werden.

### Steps to reproduce

1. Auf Modus "PV" schalten
2. Auf Plan klicken
3. Reiter "Ankunft wählen
4. Min. Ladung auf "---" stellen

### Configuration details

```yaml
- type: hyundai
  title: Hyundai Ioniq 5
  user: ***
  password: ***
  vin: ***
  capacity: 72.4
  phases: 3
  cache: 10m  
  onIdentify:
    mode: pv
    priority: 1
  name: hyundai
```

### Log details

```text
imho hinfällig, da nicht Teil der Programmlogik
```

### What type of operating system are you running?

Linux

### Version

0.123.7

andig, 2024-01-09:
@coderph0x mir erschein "---" für "tut nix" klar. Einen dynamischen Tooltip fände ich eher verwirrend.
@naltatis evtl. "inaktiv" statt "---"? Sonst können wir das auch gerne lassen wie es ist.

coderph0x, 2024-01-09:
@andig da hast du mich offenbar missverstanden. Der Report bezog sich nicht auf die Auswahlmöglichkeit „---“ für „tut nix“, sondern auf die Erklärung rechts daneben, die bei einer Einstellung „tut nix“ so keinen Sinn ergibt. Es wird ja eben nicht schnell auf x% geladen, sondern nichts getan.

naltatis, 2024-01-09:
Ja, `x` in diesem Satz Platzhalter für den auszuwählenden Wert. Hast du einen konkreten Vorschlag für eine bessere Formulierung in diesem Zustand?

coderph0x, 2024-01-09:
x  erscheint ja nur, wenn man „---“ auswählt, ansonsten steht der korrekte Wert dort.
Aber ich verstehe langsam, weshalb ihr auf diese Idee kamt. Wenn man erstmalig den Bereich betritt, ist es sinnvoll, dass der Hilfstext dem Nutzer erklärt, was hier eingestellt werden kann. Wenn man jedoch bereits einen Wert eingestellt hatte und dann auf „---“ zurückstellt, verwirrt dieser generische Hinweis dann doch eher.
Mein Vorschlag wäre, bei Auswahl von „---“ den Hilfstext durch „Min. Ladung deaktiviert“ (o.ä.) zu ersetzen bzw. ergänzen.

andig, 2024-01-13:
> Mein Vorschlag wäre, bei Auswahl von „---“ den Hilfstext um „Min. Ladung deaktiviert“ (o.ä.) zu ergänzen.

Das werden wir nicht tun. Der Hilfetext soll statisch beschreiben wofür das Setting da ist. Wenn keine gute Idee für eine Formulierung gibt würde ich hier zu machen?

coderph0x, 2024-01-13:
Der Hilfstext ist aber nicht statisch, da immer der aktuell eingestellte Wert angezeigt wird. Wie dem auch sei, wenn man es zweimal genutzt hat, irritiert es nicht mehr. UX ist halt auch, was der unbedarfte Nutzer draus macht.

andig, 2024-01-13:
Dann mache ich hier zu- wenns einen konkreten Vorschlag gibt gerne kommentieren.

closed 2024-01-13 by andig