evcc-io/evcc #11575: Batterie wird durch Strom vom Netz geladen

netguru76, 2024-01-09:
### Describe the bug

Seit der neuen Version (vorher 123.5) lädt der "normale" Strom aus dem Netz meine Batterie auf. EVCC zeigt dies als "Erzeugung" an ....

### Steps to reproduce

1.simple start evcc
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 573a08db636491298c96b423afd20888ddf4455baa815addda579ada2522108b

interval: 15s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.39  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.39  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.39  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3
- type: template
  template: sungrow-hybrid 
  id: 2  
  host: 192.168.178.38  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv4

chargers:
- type: template
  template: elliconnect 
  ski: 3FE1 17F1 B5E4 A052 8F9C 1E8B E9CC A352 D326 4CF2  
  ip: 192.168.1.117 
  name: wallbox6

vehicles:
- type: template
  template: cupra 
  title: Cupra DN  
  user: dnink@web.de  
  password: cupra9999  
  vin: VSSZZZK1ZNP013084  
  capacity: 58  
  name: ev5

loadpoints:
- title: Eli
  charger: wallbox6
  vehicle: ev5
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  #resetOnDisconnect: true

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
    - pv4
    battery:
    - battery3

eebus:
  certificate:
```

### Log details

```text
funktioniert leider nicht, rennt auf einer synology ohne Docker
```

### What type of operating system are you running?

Docker container

### Version

123.7

premultiply, 2024-01-09:
Log fehlt.

closed 2024-01-09 by premultiply

evilmon, 2024-01-25:
Du hast mit der Config hier dein Passwort zu deinem Cupra-Account gepostet. Ändere das Passwort mal in deinem eigenen Interesse ;-)