evcc-io/evcc #11576: planenergy in the API should support relative value in % and not only absolute values in kWh

andromedavpn, 2024-01-09:
After the breaking change in API for the planner the targetSoC cannot be set anymore as a percent value. "planenergy" requires now an absolute value in kWh, e.g. /`api/loadpoints/1/plan/energy/22/2024-01-10T12:20:00Z` whereby 22 stands for 22kWh targetSoC

Preferably it should be possible to set  the targetSoC either as absolute or relative value (similar to the presentation in the UI). 
One possibility will be to consider all value smaller than 1 as fractions of 100% and all values above 1 as absolute kWh values. Other possibility will be to have something like
```
/api/loadpoints/1/plan/energy/absolute/22/2024-01-10T12:20:00Z 
/api/loadpoints/1/plan/energy/relative/45/2024-01-10T12:20:00Z 
```
Other approaches will be also fine.

(At least for my use cases I do not need absolute values at all.)

andig, 2024-01-09:
> After the breaking change in API for the planner the targetSoC cannot be set anymore as a percent value.

That's not true. The change is that target soc has become a vehicle instead of a loadpoint property, hence it must be set on vehicles.

closed 2024-01-09 by andig

andromedavpn, 2024-01-09:
Thanks for the quick reply... Then I am obviously confused by the meaning of the value in `/api/loadpoints/1/plan/energy/**value**/2024-01-10T12:20:00Z`. 

At least in my case (version 0.123.7) this value sets the charging goal (as shown in UI) in the charging plan of the load point. So at least for me it is unclear what will be the behaviour if the charge goal is 33kWh and the targetSoC 80% (=35kWh).

andig, 2024-01-09:
Es gibt nur entweder oder. Siehe https://github.com/evcc-io/evcc/pull/10335

andig, 2024-01-09:
...und https://github.com/evcc-io/evcc/issues/10330 für das zugehörige Design.