evcc-io/evcc #11577: M-Tec Energy Butler Gen 3 can't handle Battery and PV data via serial/modbus connection

ElHuscho, 2024-01-09:
### Describe the bug

We (watt analytics) found a problem when connecting to M-Tec Energy Butler GEN3 via rs485serial adapter. We wanted to have both PV and Battery Data delivered, but it seems that only one Data Source via serial/modbus is supported. We'll get the following error for the battery, when we configure both (PV and Battery) in the evcc.yaml:

Jan 09 09:37:11 iWattControllerV3 evcc[13474]: [site  ] ERROR 2024/01/09 09:37:11 battery 1 power: read failed: serial: timeout
Jan 09 09:37:16 iWattControllerV3 evcc[13474]: [site  ] ERROR 2024/01/09 09:37:16 battery 1 soc: read failed: serial: timeout

### Steps to reproduce

configure MTec GEN3 Pv and Battery with RS485 via adapter

### Configuration details

```yaml
---
network:
  schema: "http"
  host: "0.0.0.0"
  port: 7070
interval: "2s"
sponsortoken: "**********"
log: "info"
meters:
- name: "enerix Ennstal"
  type: "custom"
  power:
    source: "mqtt"
    topic: "wa-pi/grid"
  currents:
  - source: "mqtt"
    topic: "wa-pi/grid/p1"
  - source: "mqtt"
    topic: "wa-pi/grid/p2"
  - source: "mqtt"
    topic: "wa-pi/grid/p3"
- name: "Huawei"
  type: "template"
  template: "huawei-dongle"
  usage: "pv"
  host: "192.168.0.21"
  port: 502
  id: 1
  modbus: "tcpip"
- name: "Solinteg"
  type: "template"
  template: "mtec-eb-gen3"
  usage: "pv"
  id: 1
  modbus: "rs485serial"
  baudrate: 9600
  device: "/dev/ttyUSB0"
  comset: "8N1"
- name: "Solinteg Batterie"
  type: "template"
  template: "mtec-eb-gen3"
  usage: "battery"
  id: 1
  modbus: "rs485serial"
  baudrate: 9600
  device: "/dev/ttyUSB0"
  comset: "8N1"
chargers:
- name: "Go-E-Simulator"
  type: "template"
  template: "go-e-v3"
  host: "localhost:8091"
site:
  title: "enerix Ennstal"
  meters:
    grid: "enerix Ennstal"
[... 157 lines cut ...]
Jan 09 09:39:57 iWattControllerV3 evcc[13537]: [site  ] ERROR 2024/01/09 09:39:57 battery 1 power: read failed: serial: timeout
Jan 09 09:39:57 iWattControllerV3 evcc[13537]: [modbus] TRACE 2024/01/09 09:39:57 modbus: send 01 03 80 e8 00 01 2d fe
Jan 09 09:39:58 iWattControllerV3 evcc[13537]: [mqtt  ] TRACE 2024/01/09 09:39:58 recv wa-pi/grid: '1259.21'
Jan 09 09:39:58 iWattControllerV3 evcc[13537]: [mqtt  ] TRACE 2024/01/09 09:39:58 recv wa-pi/grid/p1: '0.737722'
Jan 09 09:39:58 iWattControllerV3 evcc[13537]: [mqtt  ] TRACE 2024/01/09 09:39:58 recv wa-pi/grid/p2: '4.1134605'
Jan 09 09:39:58 iWattControllerV3 evcc[13537]: [mqtt  ] TRACE 2024/01/09 09:39:58 recv wa-pi/grid/p3: '0.8651903'
Jan 09 09:39:59 iWattControllerV3 evcc[13537]: [mqtt  ] TRACE 2024/01/09 09:39:59 recv wa-pi/grid: '1277.2101'
Jan 09 09:39:59 iWattControllerV3 evcc[13537]: [mqtt  ] TRACE 2024/01/09 09:39:59 recv wa-pi/grid/p1: '0.7277093'
Jan 09 09:39:59 iWattControllerV3 evcc[13537]: [mqtt  ] TRACE 2024/01/09 09:39:59 recv wa-pi/grid/p2: '4.102718'
Jan 09 09:39:59 iWattControllerV3 evcc[13537]: [mqtt  ] TRACE 2024/01/09 09:39:59 recv wa-pi/grid/p3: '0.9655593'
Jan 09 09:40:02 iWattControllerV3 evcc[13537]: [modbus] TRACE 2024/01/09 09:40:02 modbus: recv
Jan 09 09:40:02 iWattControllerV3 evcc[13537]: [site  ] ERROR 2024/01/09 09:40:02 battery 1 soc: read failed: serial: timeout
Jan 09 09:40:02 iWattControllerV3 evcc[13537]: [site  ] DEBUG 2024/01/09 09:40:02 battery soc: 0%
Jan 09 09:40:02 iWattControllerV3 evcc[13537]: [site  ] DEBUG 2024/01/09 09:40:02 battery power: 0W
Jan 09 09:40:02 iWattControllerV3 evcc[13537]: [cache ] TRACE 2024/01/09 09:40:02 batteryCapacity: 0
Jan 09 09:40:02 iWattControllerV3 evcc[13537]: [cache ] TRACE 2024/01/09 09:40:02 batterySoc: 0
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.123.7

premultiply, 2024-01-09:
`interval: "2s"` ...

ElHuscho, 2024-01-11:
Problem was that M-Tec Energy Butler GEN3 requires id 247 instead of 1.  Works now.

closed 2024-01-11 by ElHuscho