evcc-io/evcc #11579: cannot create vehicle type 'porsche': missing credentials

ReLo99, 2024-01-09:
### Describe the bug

Das "Vehicle" konnte bis vor wenigen Wochen fehlerfrei über die yaml angelegt werden.
Ohne die yaml zu verändern erhalte ich jetzt folgenden Fehler: 
"creating vehicle Taycan4s failed: cannot create vehicle type 'template': cannot create vehicle type 'porsche': missing credentials"
Die Porsche-App funktioniert mit den identischen Login-Daten weiterhin.

### Steps to reproduce

1. sudo systemctl start evcc
2. fatal error.....cannot create vehicle type 'template': cannot create vehicle type 'porsche': missing credentials"
3.
...

### Configuration details

```yaml
network:
  #    schema is the HTTP schema
  #    setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  #    host is the hostname or IP address
  #    if the host name contains a `.local` suffix, the name will be announced on MDNS
  #    docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.108.11
  #    port is the listening port for UI and api
  #    evcc will listen on all available interfaces
  port: 7070

interval: 15s # control cycle interval

log: error

meters:
  - name: grid1
    type: movingaverage
    decay: 1.0
    meter:
      type: template
      template: vzlogger
      usage: grid1
      host: 192.168.108.11
      port: 8081
      uuid: 6836dd20-00d5-11e0-bab1-856ed5f959bd       # Hauptzaehler Einspeisung (W mit negativem Vorzeichen)
#    name: grid1
# PV Dach Einspeisung kumuliert
#  uuid: 6836dd20-00d5-11e0-bab1-856ed5f959af
# PV Carport Leistung
#  uuid: 6836dd20-00d5-11e0-bab1-856ed5f959ad

chargers:
  - name: KebaP30c
    type: template
    template: keba
    host: 192.168.108.188:7090      # IP-Adresse oder Hostname
    rfid: #
    serial: #

vehicles:
  - name: Taycan4s
    type: template
    template: porsche
    title: Taycan 4s
    user: #
    password: #
    vin: #
    capacity: 83.7
    mode: pv
loadpoints:
  - title: Carport
    charger: KebaP30c
    vehicles: Taycan4s
    mode: pv
    phases: 1
    mincurrent: 6
    maxcurrent: 32
    enable:               # Laden starten
      threshold: -1000    # Schwellwert für Netzexport -1000 Watt export
      delay: 3m
    disable:              # Laden beenden
      threshold: -999     # Schwellwert für Netzexport -999 Watt export
      delay: 30m

site:
  - title: Carport GT
    meters:
      grid: grid1
#  residualPower: 1250 # max. Netzbezug bei PV-Laden

mqtt:
  broker: localhost:1883
  topic: home-assistant/evcc
```

### Log details

```text
[main  ] INFO 2024/01/09 14:36:30 evcc 0.123.7
[main  ] INFO 2024/01/09 14:36:30 using config file: /etc/evcc.yaml
[main  ] INFO 2024/01/09 14:36:30 starting ui and api at :7070
[db    ] INFO 2024/01/09 14:36:30 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2024/01/09 14:36:30 connecting evcc-1768325745 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/01/09 14:36:30 tcp://localhost:1883 connected
[main  ] ERROR 2024/01/09 14:36:30 creating vehicle Taycan4s failed: cannot create vehicle type 'template': cannot create vehicle type 'porsche': missing credentials
[lp-1  ] DEBUG 2024/01/09 14:36:30 charge total import: 6618.867kWh
[site  ] INFO 2024/01/09 14:36:30 site config:
[site  ] INFO 2024/01/09 14:36:30   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2024/01/09 14:36:30     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/09 14:36:30   vehicles:
[site  ] INFO 2024/01/09 14:36:30     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/01/09 14:36:30 loadpoint 1:
[lp-1  ] INFO 2024/01/09 14:36:30   mode:        pv
[lp-1  ] INFO 2024/01/09 14:36:30   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/01/09 14:36:30   meters:      charge ✓
[lp-1  ] INFO 2024/01/09 14:36:30     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/01/09 14:36:30 phase timer inactive
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.123.7

andig, 2024-01-09:
> missing credentials

Zugangsdaten fehlen in Deiner Config

closed 2024-01-09 by andig

ReLo99, 2024-01-10:
Ich habe die Zugangsdaten:
user, password und VIN aus Datenschutzgründen mit einem # versehen.
Ansonsten stehen dort die richtigen Login-Daten, die auch mit der Porsche-App einwandfrei funktionieren und
auch zuvor viele Monate funktioniert haben.
Kann es sein, dass Porsche die API für meinen User gesperrt hat?

andig, 2024-01-10:
Nein, der Fehler kommt von evcc und er kommt nur wenn User/Pw fehlen.

ReLo99, 2024-01-10:
Anbei findest du meine eingetragenen Credentials (z.T. geschwärzt). 
Ich kann den Fehler nicht erkennen...
[image]

andig, 2024-01-10:
Vllt wird eine ganz andere Config verwendet…

ReLo99, 2024-01-10:
Wie kann ich feststellen, welche Config aktuell verwendet wird?

ReLo99, 2024-01-10:
gelöst! Ich habe das Password in der Porsche-App neu vergeben und zusätzlich das PW in einfache Anführungszeichen gesetzt.

andig, 2024-01-10:
> Das "Vehicle" konnte bis vor wenigen Wochen fehlerfrei über die yaml angelegt werden.

...das kann dann aber nicht stimmen. Jdfls gut, dass es gelöst ist.

ReLo99, 2024-01-13:
Nachtrag:
Unter LOADPOINTS hatte bei mir bisher der Eintrag VEHICLES fehlerfrei funktioniert.
Richtig ist aber die Einzahl VEHICLE. Anscheinend führt der alte Eintrage (VEHICLES) nach den letzten Updates zu einem Fehler in der Konfiguration. EVCC kann jetzt wieder die Fahrzeugdaten auslesen.
Durch weiteres Testen kann ich das PW-Problem (wie oben beschrieben) als Fehler ausschließen.