evcc-io/evcc #11581: Vestel EVC04 Charger

m0nn3, 2024-01-09:
### Describe the bug

Vestel EVC04   failsafe timeout.

Die Wallbox läuft schon eine weile, hatte vorher version 0.123.2 drauf und durch diese Fehlermeldung mal ein Update gemacht.
Was leider nichts brachte.

Habe auch versucht mit evcc configure eine neue evcc.yaml zu erstellen aber dort bekomme ich auch einen TimeOut.

Die Weboberfläche der Vestel Box kann ich erreichen und dort ist auch Modbus TCP noch aktiv.

HMI Softwarestand :
	v3.43.0
Leistungsplatine Softwarestand :
	1.0.88.0
Dauer nach dem Einschalten :
	2234 : 42 : 57 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.339 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.082 # [currency]/kWh

meters:
- type: template
  template: solax 
  id: 1  
  host: 192.168.178.201  
  port: 502  
  usage: grid  
  modbus: rs485tcpip  
  name: grid1
- type: template
  template: solax 
  id: 1  
  host: 192.168.178.201  
  port: 502  
  usage: pv  
  modbus: rs485tcpip  
  name: pv2
- type: template
  template: solax 
  id: 1  
  host: 192.168.178.201  
  port: 502  
  usage: battery  
  modbus: rs485tcpip  
  name: battery3

chargers:
- type: template
  template: vestel 
  id: 255  
  host: 192.168.178.63  
  port: 502  
  modbus: tcpip  
  name: wallbox5

vehicles:
- type: template
  template: renault 
  title: Twingo  
  user: *****
  password: *****
  vin: *****
  capacity: 22  
  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  vehicle: ev4
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 32

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
[main  ] INFO 2024/01/09 16:36:15 evcc 0.123.7
[main  ] INFO 2024/01/09 16:36:15 using config file: /etc/evcc.yaml
[main  ] INFO 2024/01/09 16:36:15 starting ui and api at :7070
[db    ] INFO 2024/01/09 16:36:15 using sqlite database: /home/m0nn3/.evcc/evcc.db
[main  ] FATAL 2024/01/09 16:36:25 cannot create charger 'wallbox5': cannot create charger type 'template': cannot create charger type 'vestel': failsafe timeout: dial tcp 192.168.178.63:502: i/o timeout
[main  ] FATAL 2024/01/09 16:36:25 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.123.7

andig, 2024-01-09:
i/o timeout -> die Box will nicht mit uns reden, warum auch immer. Entweder längerer Timeout oder Support muss die Ursache abstellen.

closed 2024-01-09 by andig

m0nn3, 2024-01-09:
Hallo,

Wie kann ich jetzt fortfahren um mein Fehler zu beheben?
Über die IP kann ich die Box erreichen.
Für mich scheint sie zu funktionieren.
Sollte ich jetzt den Hersteller kontaktieren?
Es wurde ja nichts am System getan, die Wallbox läuft ja schon ein paar Monate ohne Probleme.

Gruß Monne

premultiply, 2024-01-09:
Vielleicht andere Instanz bereits vorhanden und verbunden. Wer weiss. Jedenfalls keine Problem vom evcc.

m0nn3, 2024-01-11:
Hallo, vielen Dank.
Ich hatte auf meinen Rechner noch einen Container mit evcc. Dieser wurde gestartet.
Nach dem stoppen und löschen läuft wieder alles.

Vielen Dank für die Hilfe!