evcc-io/evcc #11582: Falscher PV Wert über MQTT - SMA Hybrid mit docker 0.123.7

q2doc, 2024-01-09:
### Describe the bug

Hi,

Seit 2 Tagen wird über den MQTT Adapter bei mir die falsche PV-Leistung übermittelt (bei 0.123.5 meine ich hat es noch richtig funktioniert, kann aber wegen watchtower nicht sagen wann das update erfolgt ist).

Influx Daten:

[image]

Die blaue Linie sollte die Summe der orangen und pinken sein.

Influx Daten (heute):

[image]

SMA Sunny Daten (heute):

[image]

### Steps to reproduce

1. immer, werde aber sobald Sonne ist einen Rollback auf die 0123.5 testen und berichten

### Configuration details

```yaml
meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.xxx.xx
  name: Netz
- type: template
  template: sma-hybrid
  id: 3
  host: 192.168.xxx.xx
  port: 502
  usage: pv
  modbus: tcpip
  name: PV
- type: template
  template: sma-hybrid
  id: 3
  host: 192.168.xxx.xx
  port: 502
  usage: battery
  modbus: tcpip
  capacity: 10.2
  name: battery2
```

### Log details

```text
folgt bei Sonne...
```

### What type of operating system are you running?

Docker container

### Version

0.123.7

StefanSchoof, 2024-01-09:
Ist das gleiche Problem wie #11562?

q2doc, 2024-01-09:
sorry ja - hatte nur die offenen issues geschaut. Installiere mal den developer release.

closed 2024-01-09 by q2doc