evcc-io/evcc #11596: Laden des Autos automatisch an die Stadtwerke überweisen

unl0ck, 2024-01-10:
**Is your feature request related to a problem? Please describe.**
Das Problem ist ja, wenn man zuhause sein Auto "auftankt" wird das oft durch die abschlagszahlung beglichen, es gibt keine möglichkeit nach einer Ladung die Summe an den Stromanbieter zu überweisen

**Describe the solution you'd like**
Eine möglichkeit, Montalich Täglich, pro Ladung die kosten an ein Konto zu überweisen beispielsweise Stromlieferant.
Man hinterlegt seine Kontologins, und führt dann automatisiert eine überweisung aus.

**Describe alternatives you've considered**
Paypal überweisungen

**Additional context**
Hab ich leider nicht. sorry.

mdkeil, 2024-01-10:
> 

> Das Problem ist ja, wenn man zuhause sein Auto "auftankt" wird das oft durch die abschlagszahlung beglichen, es gibt keine möglichkeit nach einer Ladung die Summe an den Stromanbieter zu überweisen

Was ist denn das für ein Use-Case? Davon habe ich ja noch nie was gehört. Und welcher Stromanbieter sollte sowas überhaupt wollen und anbieten?-- Ist doch ein Mehraufwand ohne jeglichen Nutzen..

unl0ck, 2024-01-10:
naja aktuell ist es ja so, ich geh an meine Wallbox lade mein Auto und am ende des Jahres zahl ich mehr weil die abschlagszahlung nicht dem entspricht was errechnet wurde.

wenn ich jetzt nach jeder ladung einfach das geld was ich sowieso dem Stromanbieter in rachen werfen muss schon direkt mache, ist das wie an der Tank wo ich mein Benzin direkt bezahle und nicht erst am ende des Jahres wo es gegen meine Abschlagszahlung geht.

VolkerK62, 2024-01-10:
Stell dir ne Spardose neben die Wallbox  🤣

menkej, 2024-01-10:
Seltsamer Usecase, da sich die Autonutzung ja irgendwann aufs Jahr einpendelt und damit die Abschläge passen sollten... Ansonsten: nimm dir doch einfach am Ende des Montags das Lade-Logbuch und mach dann eine Überweisung @unl0ck ...

TobiasHuber1980, 2024-01-10:
Oder passe die monatlichen Abschläge an deinen Anbieter nach oben an…. Hat ja nix mit evcc zu tun… 🧐

mdkeil, 2024-01-10:
> wenn ich jetzt nach jeder ladung einfach das geld was ich sowieso dem Stromanbieter in rachen werfen muss schon direkt mache, ist das wie an der Tank wo ich mein Benzin direkt bezahle und nicht erst am ende des Jahres wo es gegen meine Abschlagszahlung geht.

Glaube nicht, dass dies bei irgend ein Anbieter möglich ist.. Neben den bereits aufgezeigten Lösungsmöglichkeiten einfach den Mehrverbrauch durch die E-Karre überschlagen und den Abschlag anpassen.

naltatis, 2024-01-10:
Ja, definitiv out-of-scope für evcc. Aber wir stellen alle APIs bereit, um an die relevanten Daten zu kommen.

closed 2024-01-10 by naltatis