evcc-io/evcc #11611: EVCC stoppt Ladung bei 100% obwohl das Fahrzeug nicht voll ist

TweetsOfNiklas, 2024-01-11:
### Describe the bug

Wenn ich meinen Nissan Leaf 2012/24kWh anstecke habe ich eigentlich fast immer das Problem, dass er nicht voll lädt. Das Problem liegt wahrscheinlich daran, dass die Carwings API keine genauen Werte kennt, sondern nur Werte im Abstand von ca. 15%. Wenn also zuerst der Ladestand abgefragt wird, muss das nicht zwingend der exakte sein. Das ist im normalen Umfeld nicht problematisch. Wenn EVCC jetzt aber hochzählt oder die Kapazität der konfigurierten Batterie nicht exakt zur Degradation passt (in meinem Fall 66% SOH), dann wird zu früh 100% erreicht und EVCC erklärt den Ladevorgang für beendet.
Hier würde eine andere Logik mehr Sinn machen: Bei 100% solange laden bis das Fahrzeug die Ladung beendet.

### Steps to reproduce

1. Nissan Leaf Fahrzeug anlegen
2. Anstecken und auf Modus PV laden
3. Keinen Plan und kein Limit einstellen
4. Bei abgeschlossener Ladung auf Gastfahrzeug ändern
5. Auto lädt mit der vollen Leistung eine weitere Stunde

### Configuration details

```yaml
- type: template
  template: carwings 
  title: Nora  
  user: myuser  
  password: mypassword  
  capacity: 15  
  name: ev2
  phases: 1
```

### Log details

```text
Kann ich nachreichen wenn notwendig. Auto ist jetzt voll.
```

### What type of operating system are you running?

Docker container

andig, 2024-01-11:
Version unklar, Log fehlt.

closed 2024-01-11 by andig

VolkerK62, 2024-01-11:
dazu gab es doch in 0.123.3  https://github.com/evcc-io/evcc/pull/11348

andig, 2024-01-11:
Genau. Falls das immer nich auftritt muss es eine andere Ursache haben. Wir brauchen ein Logfile…