evcc-io/evcc #11614: Nach update (0.123.7): * '' has invalid keys: uri

mpucherSmIT, 2024-01-11:
### Describe the bug

Hallo, nach dem Update von 0.122.1 auf 0.123.7 kommt folgender Fehler. In meiner evcc.yaml ist die Uri aber in vielen Abschnitten enthalten. Gibt es die Möglichkeit, im Log die Zeilennummer mit auszugeben, damit ich weiß, welchen Teil ich anpassen muss?

`[main  ] FATAL 2024/01/11 12:17:38 failed parsing config file: 1 error(s) decoding:

* '' has invalid keys: uri`

### Steps to reproduce

1. Update
2. try to restart
3. read the log
...

### Configuration details

```yaml
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: trace
levels:
  core: trace #debug
  lp-1: trace #debug

# influx database
influx:
  url: http://influxdb:8086
  database: telegraf
  # user:
  # password:

# meter definitions

meters:
- name: pv-dc-kostal
  type: custom
  power:
    source: calc
    add: # The add plugin sums up both string values
    - source: modbus
      model: sunspec
      value: 160:1:DCW # string 1
      uri: 192.168.2.128:1502
      id: 71 # Configured Modbus Device ID 
    - source: modbus
      model: sunspec
      value: 160:2:DCW # string 2
      uri: 192.168.2.128:1502
      id: 71 # Configured Modbus Device ID 
    - source: modbus
      model: sunspec
      value: 160:1:DCW # WR1 (6a) String 1
      uri: 192.168.2.180:502
      id: 1 # Configured Modbus Device ID
    - source: modbus
      model: sunspec
      value: 160:2:DCW # WR1 (6A) String 2
      uri: 192.168.2.180:502
      id: 1 # Configured Modbus Device ID
    - source: modbus
      model: sunspec
      value: 160:1:DCW # WR2 (6) String 1
      uri: 192.168.2.181:502
      id: 2 # Configured Modbus Device ID
    - source: modbus
      model: sunspec
      value: 160:2:DCW # WR2 (6) String 2
      uri: 192.168.2.181:502
      id: 2 # Configured Modbus Device ID

- name: grid-kostal
  type: modbus
  model: ksem
  uri: 192.168.2.109:502
  id: 1 # Configured Modbus Device ID 
  power: Power
  energy: Export

- name: battery-kostal
  type: modbus
  power: 802:W
  uri: 192.168.2.128:1502
  id: 71 # Configured Modbus Device ID ("Gerät")
  soc: 802:SoC

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
chargers:
- name: walli-garage
#- name: phoenix-emcp
#  type: phoenix-emcp # ESL Walli charger Garage
  type: phoenix-em-eth
  #type: template
  #template: phoenix-em-eth
  ##modbus: tcpip
  #id: 255
  #host: 192.168.2.99
  #port: 502
  uri: 192.168.2.99:502 # ModBus address
  #meter:
  #  power: true
  #  energy: true
  #  currents: true

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
vehicles:
- name: kona
  type: hyundai
#  type: default
  title: Hyundai Kona
  capacity: 64 # kWh
#  charge: 
#    type: script
#    cmd: /home/mike/projects/energylog/bluelink/evcc_kona_charge.js
#    timeout: 20s
  user:  # user
  password:  # password
  cache: 5m

# site describes the EVU connection, PV and home battery
site:
  title: energylog EVCC # display name for UI
  #ResidualPower: 500
  meters:
    grid: grid-kostal # grid meter
    pv: pv-dc-kostal
    battery: battery-kostal

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Garage # display name for UI
  charger: walli-garage

#  charger: phoenix-emcp # charger
  #meters:
  #  charge: phoenix-garage-meter # charge meter
  vehicle: kona
  mode: minpv
#  targetSoC: 100 # charge to 100%
  soc:
#    alwaysUpdate: false # set true to update vehicle soc even when disconnected
#    levels: # target soc levels for UI
#    - 30
#    - 50
#    - 80
#    - 85
#    - 90
#    - 95
#    - 100
#  onDisconnect: # set defaults when vehicle disconnects
#    mode: minpv # switch back to pv mode
#    targetSoC: 85 # charge to 100%
  phases: 3 # ev phases (default 3)
  #sensitivity: 1 # current raise/lower step size (default 10A)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.2496 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.1559 # [currency]/kWh
```

### Log details

```text
[main  ] INFO 2024/01/11 12:22:34 evcc 0.123.7
[main  ] INFO 2024/01/11 12:22:34 using config file: /etc/evcc.yaml
[main  ] WARN 2024/01/11 12:22:34 `uri` is deprecated and will be ignored. Use `network` instead.
[main  ] INFO 2024/01/11 12:22:34 starting ui and api at :7070
[main  ] FATAL 2024/01/11 12:22:34 failed parsing config file: 1 error(s) decoding:

* '' has invalid keys: uri
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.123.7

mdkeil, 2024-01-11:
#10335 

**BC Breaks**
Global:
`uri` removed, use `network` instead

closed 2024-01-11 by andig

mpucherSmIT, 2024-01-11:
Danke, geht wieder