evcc-io/evcc #11623: Evcc startet nicht mehr

Dosepower, 2024-01-11:
### Describe the bug

Lief immer ohne Probleme. Aufeinmal startet es nicht mehr. habe nicht an der konfig verändert

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: debug  
levels:  
  cache: error  
  
# unique installation id  
plant: 070854873f368f9d78ffd29433203c242610b3ced2123f8e68169744ff86a8bd
  
interval: 10s # control cycle interval  
  
meters:
- type: template  
  template: varta   
  usage: grid    
  host: 192.168.178.10    
  port: 502    
  name: grid1  
- type: template  
  template: kostal-piko-legacy   
  usage: pv    
  host: 192.168.178.32    
  user: pvserver
  password: XXXXXX  
  name: pv2
- type: template
  template: shelly-1pm 
  usage: pv  
  host: 192.168.178.105  
  name: pv3
- type: template  
  template: varta   
  usage: battery    
  host: 192.168.178.10    
  port: 502    
  name: battery4  
    
    
chargers:  
- type: template  
  template: tinkerforge-warp   
  host: 192.168.178.68    
  port: 1883    
  topic: Dosepower    
  user: Dosepower
  password: XXXX
  name: wallbox3  
  
vehicles:  
- type: template
  template: offline 
  title: Kona  
  capacity: 39  
  name: ev3
  phases: 1
  mode: pv
  mincurrent: 6
  maxcurrent: 32
- type: template  
  template: id   
  title: ID 4    
  user: n.dose@swn.net
  password: XXXXX
  vin: WVGZZZE2ZNP057760
  capacity: 77    
  name: ev2
  phases: 3
  mode: pv
  mincurrent: 10
  maxcurrent: 16
  
  
loadpoints:  
- title: Carport  
  charger: wallbox3  
  vehicle: ev3  
  mode: pv  
  phases: 3  
  mincurrent: 6  
  maxcurrent: 32  
  resetOnDisconnect: true  
  
site:  
  title: Zuhause  
  meters:  
    grid: grid1  
    pv:  
    - pv2
    - pv3
    battery:  
    - battery4 
  residualPower: 300
  prioritySoc: 50
  bufferSoc: 80
  bufferStartSoc: 90
  
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.39 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.24 # [currency]/kWh
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/01/11 17:34:44 evcc 0.123.7
[main  ] INFO 2024/01/11 17:34:44 using config file: /config/evcc.yaml
[main  ] INFO 2024/01/11 17:34:44 starting ui and api at :7070
[db    ] INFO 2024/01/11 17:34:44 using sqlite database: /data/evcc.db
[warp  ] INFO 2024/01/11 17:34:45 connecting evcc-661355970 at tcp://192.168.178.68:1883
[warp  ] DEBUG 2024/01/11 17:34:45 tcp://192.168.178.68:1883 connected
[warp  ] DEBUG 2024/01/11 17:34:45 tcp://192.168.178.68:1883 subscribe Dosepower/evse/low_level_state
[main  ] FATAL 2024/01/11 17:34:51 failed configuring loadpoints: failed configuring loadpoint: 1 error(s) decoding:

* '' has invalid keys: resetondisconnect
[main  ] FATAL 2024/01/11 17:34:51 will attempt restart in: 5m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

VolkerK62, 2024-01-11:
>habe nicht an der konfig verändert

das ist aber notwendig. Mit 0.123.0 sind einige breaking changes reingekommen.
siehe https://github.com/evcc-io/evcc/discussions/10978

Die Fehlermeldung nennt einen Punkt der raus muss. Die anderen sind die Batterie Einstellungen in der site.

Dosepower, 2024-01-11:
prioritySoc: 50
  bufferSoc: 80
  bufferStartSoc: 90

habe die 3 sachen rausgelöscht geht immer noc nicht

0x3d13f, 2024-01-11:
`resetOnDisconnect: true` - diese Zeile muss auch noch raus, das sagt doch Deine Fehlermeldung.

Dosepower, 2024-01-11:
wo ist diese zeile?

0x3d13f, 2024-01-11:
letzte Zeile bei `loadpoints:`

Dosepower, 2024-01-11:
oh man war ich blind danke

closed 2024-01-11 by Dosepower