evcc-io/evcc #11624: Als Tool möchte ich automatisch erkennen wenn sich was am Datenmodell geändert hat und die Daten automatisch migrieren

xerion3800, 2024-01-11:
Wenn ich aktuell die Issues auf gitHub verfolge, muss ich mit erschrecken feststellen wie oft immer wieder die gleiche Fragen kommen wenn das aktuell Update eingespielt wird und die dadurch entstehenden Änderungen aus der evcc.yaml. Das ist einerseits nachvollziehbar andersrum gibt es natürlich eine Doku aber wie man sieht muss man die leider immer wieder "vorlesen" oder bzw. darauf hinweisen

Ich kenne diese Herausforderung auch aus meinen Job heraus wenn sich was von einer Version zu der anderen Version am Datenmodel geändert hat.
Wir haben es so gelöst, das wir die Daten automatisch migrieren (also ersetzen oder falls veraltet einfach mit einem Hinweis auskommentieren) und dann den User beim Start eine kleine Übersicht anzeigen was in den Daten geändert wurde.

Vielleicht würde so ein Feature die EVCC Entwickler entlasten, denn das Tool weiß m.E. am besten was die richtigen Daten sind und wenn diese migriert sind sollte es zu deutlich weniger Rückfragen kommen.

andig, 2024-01-11:
Gute Idee- beim nächsten Update werden wir sowas auch machen, siehe https://github.com/evcc-io/evcc/pull/11571. Es ist halt auch immer eine Kosten/Nutzen Frage…

StefanSchoof, 2024-01-11:
Wenn die evcc.yaml automatisch angepasst wird, hoffen wir das nicht Leute die als read only im docker gemountet haben oder andere Wege gefunden haben einen Schreibzugriff zu verhindern...

andig, 2024-01-11:
Der Punkt für mich ist- wie bei allen guten Ideen: wer macht es? Wie macht man es konkret? Und ist es wichtiger das zu machen oder etwas Anderes?

xerion3800, 2024-01-12:
> Gute Idee- beim nächsten Update werden wir sowas auch machen, siehe #11571. Es ist halt auch immer eine Kosten/Nutzen Frage…
> 
> Wenn du eine Idee hast wie das konkret aussehen soll freuen wir uns über PRs!

Mit eine PR kann ich leider dienen, da mir dafür die Programmierkenntnisse fehlen.
Ja Kosten/Nutzen Frage stellt sich immer aber wenn man mal die Anfragen zählt die seit Veröffentlichung der neuen Version kommen die  mit "Mein EVCC startet nicht mehr usw." zählt denke ich schon das der Nutzen überwiegt und gerade das Core Team entlasten würde und sich dadurch  dem Kernaufgabe wieder widmen kann und und nicht immer wieder die gleichen "First Level Support Fragen" zu beantworten.

xerion3800, 2024-01-12:
> Wenn die evcc.yaml automatisch angepasst wird, hoffen wir das nicht Leute die als read only im docker gemountet haben oder andere Wege gefunden haben einen Schreibzugriff zu verhindern...
 8
Ja mit Sicherheit wird es Ausnahmen geben aber wenn man mit dem Feature über 80% erreicht hat man m.E. auch schon  was gewonnen.

xerion3800, 2024-01-12:
> Der Punkt für mich ist- wie bei allen guten Ideen: wer macht es? Wie macht man es konkret? Und ist es wichtiger das zu machen oder etwas Anderes?

Das kann ich dir leider nicht beantworten aber wie schon oben geschrieben wenn es das Core Team entlastet um wieder zum Kerngeschäft kommen zu können, würde ich das von der Prio schon höher sehen.

Man könnte es ja auch in zwei oder drei Stufen umsetzen.
Die Einfache Variante wäre erstmal einfach die Attributen in der evcc.yaml auszukommentieren ( mit einem Hinweis das es deprecated ist) Somit würde EVCC erstmal nach einem Update starten und der einfache User wäre nicht damit überfordert alles selber zu machen. 
Es steht ja eigentlich schon mehr oder wenig eindeutig dar, warum EVCC nicht startet aber was die Anfragen in github zeigen, das die User sich doch immer wieder sehr schwer damit tun diese Hinweise zu verstehen und geschweige denn ohne fremde Hilfe/Gedankenstoße diese umzusetzen-

Im zweiten Schritt könnte man dann migrieren sofern das erforderlich ist, das man die Daten die in der evcc.yaml nichts mehr zu suchen haben in die Datenbank oder wo es erforderlich ist, migriert werden.

Und zum Schluss dann auch noch eine schönen Hinweis in der UI, das veraltete Daten erkannt wurden und diese automatisch durch EVCC angepasst wurden.

andig, 2024-01-12:
> Und zum Schluss dann auch noch eine schönen Hinweis in der UI, das veraltete Daten erkannt wurden und diese automatisch durch EVCC angepasst wurden.

Wir haben tw. >12 Monate entsprechende Meldungen gezeigt- genutzt hat es überhaupt nichts.

Lass uns hier schließen- die Anregung ist verstanden.

closed 2024-01-12 by andig

xerion3800, 2024-01-12:
> Wir haben tw. >12 Monate entsprechende Meldungen gezeigt- genutzt hat es überhaupt nichts.

Genau das ist ja das Problem, die Meldungen werden erst geklickt und dann gelesen.  ;-) Darum war ja die Idee  der automatischen Migration damit der User nichts dazu tun muss damit EVCC lauffähig bleibt.

Die Meldung die ich meine ist eher eine Summary am Ende was EVCC gemacht hat um die Daten zu migrieren damit der "erfahrene" User sich später nicht wundert warum die yaml verändert wurde.