evcc-io/evcc #11628: Switch from 1p for PV to 3p for planned charging but keep using 1p for PV charging

Tigraine, 2024-01-11:
**Is your feature request related to a problem? Please describe.**
I have a Renault Zoe which is terrible at switching phases a lot when using the PV excess for charging (connected through a Go-E wallbox).
Another thread here mentioned that doing a 1p charge is better with the Zoe - especially now in the winter where power is varying quite a bit even on sunny days (1kw to max 6 kw excess depending on heat pump and clouds etc..)
So limiting the car to 1p works quite well to fully consume all PV energy.
But when I want to do a planned charge the manual 1p setting is still in effect and I am not optimally using my Awattar variable tariff with the cheapest load logic.
It would be very convenient if there were two settings for the phases - one for PV assisted / battery assisted charging and one for "grid consumption".

**Describe the solution you'd like**
Not yet 100% sure how this would be best modeled - I honestly haven't looked at the code yet. Maybe it just makes sense to add a phase selector to the charge plan dialog so I can easily switch this for the upcoming plan.

andig, 2024-01-13:
Not sure how this should work. Planner and PV slots may alternate. Should we switch all the time?

> Maybe it just makes sense to add a phase selector to the charge plan dialog so I can easily switch this for the upcoming plan.

It seems this would increase overall costs since more expensive slots are needed. It does make sense in light of https://github.com/evcc-io/evcc/issues/11324 though.

Tigraine, 2024-01-14:
True if the tariffs also charge more for higher consumption that's added complexity.
(But that might just be an extension to the tariff cost function).
In my case my tariff is only cost based, so it's more efficient to charge for 1 hour at the cheapest point than to wait 3 hours at 1p.

andig, 2024-01-14:
Then I don‘t understand this request. Planner already switches to 3p as necessary.

andig, 2024-01-23:
No additional insights. Closing.

closed 2024-01-23 by andig

michaelheiml, 2024-11-19:
I am having the same issue and I think it is not solved yet. Please let me try to explain it again:
So I have a Zoe + go-e + this for me optimal config for PV charging:
minCurrent: 7A
maxCurrent: 25A
phases: 1
This allows me to plugin the Zoe and whenever there is PV available, it charges 1p (!) with ~1,6kW up to 5,7kW
When there is a lot of sunlight, I manually change from 1p to 3p to charge with ~4,8kW up to 17,5kW
This all works well and I do not want to change the values. Since it is PV energy I do not mind about a bit energy loss because of slower charging with 1p.

But now I have occasions where I want to use planned charging. I want planned charging to charge with full power (= 3p = 17,5kW) so that charging time is as short as possible and as little energy loss as possible. But this is not whats happening in evcc - with the settings above, planned charging charges with 1p (!) and therefore with max 5,7kW.

So I think this is what Tigraines' issue is about - to have the possibility to configure evcc to allow exactly the latter situation.

andig, 2024-11-19:
> When there is a lot of sunlight, I manually change from 1p to 3p

None of this is necessary. Evcc will already do that if you just let it…

michaelheiml, 2024-11-21:
> > When there is a lot of sunlight, I manually change from 1p to 3p
> 
> None of this is necessary. Evcc will already do that if you just let it…

I have had this discussion in the past, actually it makes sense to manually switch. As this deviates from the topic that this issue is about, I will not deepen that discussion.

Anyways. The issue described is a different one. It is about *always* using 3p when Planned Charging, no matter what is setup for PV/fast charging. Could this be set/forced by the config file?

andig, 2024-11-21:
> So limiting the car to 1p works quite well to fully consume all PV energy.

> actually it makes sense to manually switch

It doesn‘t for this pupose.

Thats the intended behavior. If a planner slot is considered cheap by any measure it makes sense to use it as much as possible, hence 3p. We do not intend to change this. You can still use the apis to modify the phases setting.

michaelheiml, 2024-11-21:
I have the feeling we are not talking about the same thing here..

What this is **not** about: how to handle 1p/3p when PV charging with the Zoe.. as the Zoe (and maybe other cars) needs different settings for 1p (starting with 7A) and for 3p (starting with 12A) it makes sense to use mostly 1p.. e.g. our typically-sized PV (12kw) on average produces 5-8kw => 1p with 25A covers that perfectly with some rest for the house. Only in some cases (midday, cloudfree) production is higher than 12A*230*3p = 8kw. If auto switching is on, this would lead to a lot of switching back and forth between 3a and 1a.. together with the 10s wait time between switching this really becomes annoying and a waste of energy. Its just much better to have set to 1p manually and only if its really a "good" day switch to 3p (PV+min).

But as written. The issue discussed here is not about auto or manual switching in PV charging mode. It is about planned charging.

As it is right now, evcc shows a wrong behaviour. **Look at the 2 screenshots**. I have manually set 3p (!) and the planned charging screen calculates the duration and charge power with 1p (!) - now, when I start planned charging, evcc actually charges with 3p with in turn means that the charging goal is reached long before what evcc calculates. I typically charge the car in the morning just before I leave. With 1p (5,8kw) planned charging starts at around 3am so car is ready at 7 when I leave. Fine. But now I manually set charging to 3p. Evcc still starts charging at 3am but is ready at around 4:30, way too early (battery cools down again before I drive, which is bad).

I do not understand why it would be problematic to add a parameter to planned charging which tells evcc if 1p or 3p should be used (besides fixing the issue shown in the screenshot that the calculation is wrong).
[image]
[image]

michaelheiml, 2024-11-21:
> > I have manually set 3p (!) and the planned charging screen calculates the duration and charge power with the maximum 5,8kW of 1p (!)
> 
> You're hijacking this issue with a new topic. Please open your own issue and provide a logfile. Thanks.

Sorry to say this but I don't think I am hijacking this thread as the original posting was: "add a phase selector to the charge plan dialog so I can easily switch this for the upcoming plan" and title "Planner: switch from 1p for PV to 3p for planned charging but keep using 1p for PV charging"

I think @Tigraine message is about the same thing than mine, except the additional issue with the dialogue window showing 1p planned charging although 3p is set.