evcc-io/evcc #11635: `loadpoint0#limitSoC` is reset to 100% during restart of evcc

MikeTheTux, 2024-01-12:
### Describe the bug

While updating from [0.123.7](https://github.com/evcc-io/evcc/releases/tag/0.123.7) to [0.123.8](https://github.com/evcc-io/evcc/releases/tag/0.123.8) `loadpoint0#limitSoC` was reset from the previous `40%` to `100%`.

The `loadpoint0#mode` was kept over the reset.

For whatever reason, the setting of `loadpoint1#limitSoC` which is a heating system remains on the `40%` (=`40°C`) that have been set. I'm happy that it doesn't jump to `100%` which would mean `100°C`.

### Steps to reproduce

1. Set a `loadpoint0#limitSoC` != to 100%
2. restart evcc
3. check for `loadpoint0#limitSoC` - will be 100% instead of the previously set value

### Configuration details

```yaml
/app # evcc dump --cfg
[main  ] INFO 2024/01/12 10:19:56 evcc 0.123.8
[main  ] INFO 2024/01/12 10:19:56 using config file: /etc/evcc.yaml
[db    ] INFO 2024/01/12 10:19:57 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/01/12 10:19:57 connecting evcc-1038806283 at tcp://192.168.69.68:1883
[mqtt  ] DEBUG 2024/01/12 10:19:57 tcp://192.168.69.68:1883 connected

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: nas
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

# log settings
log: debug
levels:
  cache: info
  audi: info
  site: info
  lp-1: info
  script: info
  push: info
  lp-2: info
  main: info
  telegram: info

interval: 10s # control cycle interval

chargers:
  - name: keba
    type: template
    template: keba-modbus
    host: 192.168.69.108
#    serial: *****
    id: 255
    port: 502
  - name: heatpump_switch
    type: template
    template: shelly
    host: 192.168.69.152
    standbypower: -1
    integrateddevice: true
    heating: true
    icon: waterheater

loadpoints:
- title: Wallbox
  charger: keba
#  mode: off                 # set default charge mode, use "off" to disable by default if charger is publicly available, now, off, minpv, pv
#  resetOnDisconnect: true   # set defaults when vehicle disconnects
  phases: 3                 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
  enable:                   # pv mode enable behavior
    delay: 1m               # threshold must be exceeded for this long
    threshold: 0            # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable:                  # pv mode disable behavior
    delay: 3m               # threshold must be exceeded for this long
    threshold: 0            # maximum import power (W)
  guardDuration: 5m         # switch charger contactor not more often than this (default 5m)
  minCurrent: 6             # minimum charge current (default 6A)
  maxCurrent: 32            # maximum charge current (default 16A)
  vehicle: Q5
  priority: 10
- title: Heatpump
  charger: heatpump_switch
  meter: heatpump_power
  vehicle: heatpump
  mode: pv
  phases: 1
  mincurrent: 2             # 2A * 230V = 460W
  maxcurrent: 3             # 3A * 230V = 690W
  enable:
    threshold: 0
    delay: 1m
  disable:
    threshold: 0
    delay: 15m              # Laufzeit mindestens 15 Minuten
  guardduration: 15m        # Pause mindestens 15 Minunten
  priority: 10
#  soc:
#    estimate: true
#    poll:
#      mode: connected
#      interval: 1m

meters:
- name: gridmeter
  type: template
  template: fronius-gen24
  usage: grid
  host: 192.168.69.144 # IP-Adresse oder Hostname
  port: 502 # Port (Optional)
- name: pvmeter
  type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.69.144 # IP-Adresse oder Hostname
  port: 502 # Port (Optional)
- name: batterymeter
  type: template
  template: fronius-gen24
  usage: battery
  host: 192.168.69.144 # IP-Adresse oder Hostname
  port: 502 # Port (Optional)
  capacity: 5.12
- name: heatpump_power
  type: custom
  power:
    source: mqtt
    topic: heatpump/power
#    timeout: 15m

site:
  title: home
  meters:
    grid: gridmeter
    pv:
      - pvmeter
    battery:
      - batterymeter
#    aux:
#      - heatpump_aux
#  prioritySoc:    15/80  # Hausbatterie bekommt bis zum SOC 15% Priorität beim Laden
#  bufferStartSoc: 95  # Hat die Hausbatterie SOC 90% erreicht startet der Ladevorgang gestützt durch die Battery, auch wenn die Solar-Leistung alleine nicht ausreicht
#  bufferSoc:      85  # Hausbatterie wird oberhalb SOC 70% als Stütze genutzt - Annahme - mit den verbl. 75% kommt man über die Nacht
  residualPower:   100  # 50% 1 Phase 6A 230V = -1380 W, davon 50%: -690 W
#  batteryDischargeControl: true

# tariffs are the fixed or variable tariffs
# cheap (tibber/awattar) can be used to define a tariff rate considered cheap enough for charging
tariffs:
  currency: EUR           # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price
    type: fixed
    price: 0.4868          # EUR/kWh

    # # or variable via tibber
    # type: tibber
    # cheap: 0.2 # EUR/kWh
    # token: *****
    # homeid: "cc83e83e-8cbf-4595-9bf7-c3cf192f7d9c" # optional if multiple homes associated to account

    # # or variable via awattar
    # type: awattar
    # cheap: 0.2 # EUR/kWh
    # region: de # optional, choose at for Austria
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082           # EUR/kWh

# mqtt message broker
mqtt:
  broker: 192.168.69.68:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: *****
  password: *****

# influx database
influx:
  url: http://192.168.69.68:8086
  database: evcc_db
  token: *****
  org: c3po

# eebus credentials
eebus:
  # uri: # :4712
  # interfaces: # limit eebus to specific network interfaces
  # - en0
  # certificate: # local signed certificate, required, can be generated via `evcc eebus-cert`
  #   public: # public key
  #   private: # private key

# push messages
messaging:
  events:
    start:                                                      # charge start event
      title: Charge started
      msg: ${title} started charging ${vehicleTitle} in ${mode} mode
    stop:                                                       # charge stop event
      title: Charge finished
      msg: ${title} finished charging ${vehicleTitle} with ${chargedEnergy:%.1fk}kWh in ${chargeDuration}
    connect:                                                    # vehicle connect event
      title: Car connected
      msg: "${vehicleTitle} connected on ${title} at ${pvPower:%.1fk}kW PV"
    disconnect:                                                 # vehicle connected event
      title: Car disconnected
      msg: ${vehicleTitle} disconnected from ${title} after ${connectedDuration}
    soc:                                                        # vehicle soc update event
      title: SoC updated
      msg: Battery of ${vehicleTitle} charged to ${vehicleSoC:%.0f}%
    guest:                                                      # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
  # - type: pushover
  #   app: *****
  #   recipients: *****
  #   - # list of recipient ids
  - type: telegram
    token: *****
    chats: *****
    - 30190795
  # - type: email
  #   uri: smtp://<user: *****

vehicles:
  - name: Q5
    type: template
    template: audi
    title: Audi Q5
    phases: 2
#    mode: now                     # now, off, minpv, pv
    user: *****
    password: *****
    vin: *****
    capacity: 14.1                # Akku-Kapazit  in kWh # Optional
    identifiers:
    - 0000000000000000
    icon: car
#    targetsoc: 100                  # onIdentify
  - name: heatpump
    type: custom
    title: Heatpump
    icon: waterheater
    phases: 1
    onIdentify:
      mode: pv
#      minsoc: 40
#      targetsoc: 65     # max 66.5
    soc:
      source: mqtt
      topic: heatpump/temperature
#      timeout: 15m

# token
sponsortoken: *****

Version: `0.123.8`
```

### Log details

```text
.
```

### What type of operating system are you running?

Docker container

### Version

0.123.8

andig, 2024-01-12:
Thank you for the report. It seems the setting actually _is_ restored but not published and hence not visible for the ui. Fix to follow.

closed 2024-01-13 by andig

noottzz, 2024-01-27:
ich weiß nicht, ob das hier hin gehört: seit dem Update auf v1.23.9 behält der Schieber beim Ladelimit seine Position nicht mehr, sondern geht beim Anstecken des Autos auf 100%. vorher v1.22x stand der Schieber fest, bis ich ihn geändert habe. dazu habe ich keine Lösung gefunden.

andig, 2024-01-27:
Default Limits stellst Du unter "Plan->Ankunft" ein, sonst gilt das nur für genaue eine Ladesession.