evcc-io/evcc #11636: Min+PV mode ignores min charge current set in UI

davidgiga1993, 2024-01-12:
### Describe the bug

When using the `Min+PV` charge mode, the minimum current set the by UI seems to be ignored.
[image]
[image]

### Steps to reproduce

1. Enable Min+PV mode
2. Set min charge to a higher value
3. See in log that evcc still uses the min charge reported by the charger - this can be identified as the yaml defines 2A which is lower than the 2.4A reported by the charger. Thus it can be assumed that the value reported by the charger overrides the value set by the UI

### Configuration details

```yaml
See #11478
```

### Log details

```text
Jan 12 14:38:51 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:38:51 set min current: 4
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [site  ] DEBUG 2024/01/12 15:39:00 ----
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:00 charge power: 1471W
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [site  ] DEBUG 2024/01/12 15:39:00 pv power: 50W
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [site  ] DEBUG 2024/01/12 15:39:00 grid power: 2912W
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [site  ] DEBUG 2024/01/12 15:39:00 grid powers: [970 767 1176]W
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [site  ] DEBUG 2024/01/12 15:39:00 grid currents: [4.97 3.91 5.33]A
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [site  ] DEBUG 2024/01/12 15:39:00 site power: 2522W
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:00 charge currents: [2.4 2.4 2.4]A
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:00 detected active phases: 3p
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:00 charger status: C
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:00 soc estimated: 40.10% (vehicle: 39.00%)
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:00 vehicle soc: 40%
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:00 set taycan plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
Jan 12 14:39:00 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:00 pv charge current: 0A = 2.2A + -3.66A (2522W @ 3p)
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [site  ] DEBUG 2024/01/12 15:39:30 ----
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:30 charge power: 1472W
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [site  ] DEBUG 2024/01/12 15:39:30 pv power: 50W
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [site  ] DEBUG 2024/01/12 15:39:30 grid power: 2898W
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [site  ] DEBUG 2024/01/12 15:39:30 grid powers: [961 763 1175]W
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [site  ] DEBUG 2024/01/12 15:39:30 grid currents: [4.94 3.9 5.33]A
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [site  ] DEBUG 2024/01/12 15:39:30 site power: 2508W
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:30 charge currents: [2.4 2.4 2.4]A
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:30 detected active phases: 3p
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:30 charger status: C
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:30 soc estimated: 40.13% (vehicle: 39.00%)
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:30 vehicle soc: 40%
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:30 set taycan plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
Jan 12 14:39:30 pi4 docker[9339]: evcc  | [lp-1  ] DEBUG 2024/01/12 15:39:30 pv charge current: 0A = 2.2A + -3.64A (2508W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

0.123.7

andig, 2024-01-12:
Related discussion in https://github.com/evcc-io/evcc/issues/11280#issuecomment-1872559005

davidgiga1993, 2024-01-12:
The question is: Does this make sense? Naively I would assume (and I think I said that already in the past):
- The min current is the maximum of all min currents provided (loadpoint, charger, vehicle, user config)
- The max current is the minimum of all max currents provided (loadpoint, charger, vehicle, user config)

This way the min/max is always correctly guarded and you can still make changes without the need of logical prioritization of certain values

andig, 2024-01-12:
Exactly- except for this one exception. Seems we'll need to revert it.

davidgiga1993, 2024-01-12:
Zu meinen Verständnis: Woher kommt der default? Könnte man nicht annehmen dass der Wert welcher von der UI geändert werden kann den 6A default hat und alle anderen 0? Dann wäre es weiterhin modifizierbar und 6A der Standard sofern der Nutzer es nicht ändert

docolli, 2024-01-12:
Der loadpoint hat mincurrent=2A, 
für das vehicle ist übers GUI ein mincurrent=4A definiert worden 
und evcc arbeitet nun mit 2,4A?

Wie kann das sein, wenn die Logik `minCurrent: Use the max of loadpoint, charger and vehicle minCurrent` implementiert ist?
Siehe https://github.com/evcc-io/evcc/pull/11311

davidgiga1993, 2024-01-12:
> Der loadpoint hat über die evcc.yaml ein mincurrent=2A, für das vehicle ist übers GUI ein mincurrent=4A definiert worden und evcc arbeitet nun mit 2,4A?
> 
> Wie kann das sein, wenn die Logik `minCurrent: Use the max of loadpoint, charger and vehicle minCurrent` implementiert ist? Siehe #11311

Der Charger meldet die 2,4A, im Code gibt es ja den Sonderfall dass der min Wert vom Charger genommen wird sofern einer existiert

VolkerK62, 2024-01-12:
>we need a default minimum (6A)

>im Code gibt es ja den Sonderfall dass der min Wert vom Charger genommen wird sofern einer existiert

Ich vermute mal, nicht jeden Charger liefert einen Min-Wert? 
Wäre es möglich für Charger, die keinen liefern, diesen auf 6A zu setzen?
Dann hätte man einen default und für die anderen, gilt der gelieferte Wert.
Dann könnte man doch die "max()" Regel ohne Ausnahmen umsetzen.

docolli, 2024-01-12:
>Ich vermute mal, nicht jeden Charger liefert einen Min-Wert?

Ja, meiner hier ist so einer. Darum musste ich auch in der evcc.yaml beim loadpoint ´mincurrent: 6` definieren. Man könnte vermutlich prüfen ob am loadpoint ein ´mincurrent´ definiert ist, wenn nicht, dann könnte man 6A als default annehmen, ABER das würde mit EEBUS Walboxen kollidieren. Entweder diese müssen mit ´mincurrent: 0` definiert werden, oder (was logischer wäre) man führt eine config Syntax wie `mincurrent: auto` ein, um festzulegen, dass der loadpoint das ´mincurrent´ selber liefert.

davidgiga1993, 2024-01-12:
Ich hab mein ein Draft erzeugt. Im MR ist ein Flussdiagram wie das minCurrent berechnet wird, tests sind auch angepasst. Feedback willkommen

andig, 2024-01-13:
@naltatis müssen wir durchsprechen.

DerAndereAndi, 2024-01-13:
Der Min-Wert vom Charger ist entweder durch das verwendete Protokol IEC61851 definiert, dort ist er dann immer 6A. Bei Wallboxen welche über ISO kommunizieren können und ein Auto angeschlossen haben das über die AC Verbindung auch ISO unterstützt, wird der min-Wert aus dieser Kommunikation ermittelt. Und wenn bei diesen Wallboxen ein EV mit IEC61851 kommunizert, ist er 6A.

hidminds, 2024-02-05:
Siehe auch #7241 .

hidminds, 2024-02-14:
Wenn der minCurrent Wert vom Loadpoint genommen würde, wäre das für mich vermutlich ein Workaround dafür, dass der am vehicle eingestellte minCurrent derzeit auch nicht funktioniert.

Das sieht dann übrigens so aus..  Gesamt Out ist bei mir höher als IN. Das wäre schön! ;-)
Die Werte beim "Ausgang" sind falsch berechnet. Tatsächlich gelangt wirklich nichts ins Fahrzeug weil der Zoe mindestens 3p mit 8A geladen werden muss, und der Hausverbrauch ist in etwa bei 0,8 kWh.
[image]

TAEmty, 2024-02-17:
Hi zusammen,
ich kann das von hidminds gezeigte Verhalten reproduzieren, bei mir macht es allerdings das EVCC aufgrund der äußeren Umstände komplett unbrauchbar: Der Zoe meiner Frau fängt bei 8A an zu laden, seit etwa 0.123.0 (hat ne weile gedauert, bis ich raushatte, was los war, deshalb bin ich unsicher) bekomme ich aber in einem Modus < "schnell" immer nur 6A Ladestromvorgabe, was dazu führt, dass der Zoe gar nicht lädt. Ich habe eine Elli Wallbox über EEBUS am EVCC.

hidminds, 2024-02-17:
Hi @TAEmty !

> edit: Ok, das ist EXAKT, was er geschrieben hat. Ich kanns also 1:1 reproduzieren.

Ja richtig, wir haben denke ich exakt dasselbe Problem. Ich habe auch einen EEBus Elli Charger und vor 0.123 hatte ich die minimale Stromstärke auf 8A am Zoe vehicle einstellen können und das hatte auch funktioniert.
Seit 0.123 ist nun leider mit evcc nur schnelles Laden ( bei mir mit 11kW Leistung ) oder batterieunterstütztes Laden bei Sonnenschein möglich wenn evcc mal über 8A/5,5kW am Charger einstellt.
Damit beim Renault Zoe überhaupt Strom in die Autobatterie gelangt, benötigt man leider eine effektive minimale Stromstärke Einstellung von 8A bei 3 Phasen = 5,5 kW Leistung. Vor 0.123 hatte die mincurrent Einstellung am vehicle noch funktioniert. Jetzt wird am Charger warum auch immer die Stromstärke unter 8A eingestellt/geregelt. Das ist per api Aufruf auch ersichtlich.
Unter 5,5kW steht in der UI, dass der laden würde und es wird dann auch bereits der neue Akku Füllstand erhöht aber tatsächlich geht unter 8A/5,5kW gar nichts in den Akku von diesem speziellen EAuto.

davidgiga1993, 2024-02-17:
Nur als kleine Anmerkung: Das Problem wurde in meinem MR bereits behoben. Wer will kann mein fork bauen, da drin geht alles. Kann die Tage auch noch ein Release veröffentlichen

hidminds, 2024-02-17:
> Nur als kleine Anmerkung: Das Problem wurde in meinem MR bereits behoben. Wer will kann mein fork bauen, da drin geht alles. Kann die Tage auch noch ein Release veröffentlichen

@davidgiga1993 danke, deine Changes würde ich gerne mal testen. Hab evcc noch nie gebaut. Könnte/Sollte das aber mal machen. Dann könnte ich evtl selbst auch mal forken und PRs zur Diskussion erstellen... ;-)

Ich habe mir deine Änderungen angeschaut.
Ich denke, dass es für mein Problem idealer wäre, wenn ich wieder am vehicle einen absoluten mincurrent einstellen könnte. D.h. nur wenn die minCurrent Einstellung am vehicle gesetzt ist, und dieses an einem Loadpoint aktiv ist, soll effektiv der Loadpoint minCurrent auf das am vehicle definierte minCurrent eingestellt werden. Der current am Loadpoint darf dann nie darunter geregelt werden.

Ich werde bspw ab Sommer noch ein zweites EAuto (Tesla) dort öfter mal laden. Wenn der Tesla an dem Loadpoint angesteckt ist, soll der Elli Charger die Lade Stromstärke bitte NICHT auf 8A Minimum einstellen. Bei dem Tesla sollte das Laden bereits ab 6A funktionieren.
Beim Zoe soll evcc wieder das Minimum auf 8A einstellen. Eben je nach erkanntem vehicle.

Ramada2020, 2024-03-09:
Hallo. Gibt ein schon ein Idee wann dieser Fix in das offizielle Release eingebaut wird. Habe mit meinem Twingo das gleiche Problem. Danke. #12770

TAEmty, 2024-03-10:
> > Nur als kleine Anmerkung: Das Problem wurde in meinem MR bereits behoben. Wer will kann mein fork bauen, da drin geht alles. Kann die Tage auch noch ein Release veröffentlichen

Hi davidgiga1993. Danke für die Info und die Mühe!. Ich bin unsicher, ob ich das von meiner Qualifikation her hinbekomme und dann auch irgendwann den Weg zurück. Ich warte, wenn ich auf mich allein gestellt bin, lieber auf ein release.
  Aber: EEBUS und die Elli Wallbox ist ne ziemliche Katastrophe, das hab ich mittlerweile mitbekommen. Buggy und instabil, zum Teil dämlich gelöst. Wenn es der Allgemeinheit hilft, schick ich gerne Logs, mach Versuche mit ID.4 <-> und Zoe an der Elli. Wenn ich mit was konretem helfen kann, werde ich das gerne machen.

andig, 2024-03-11:
Dieser Fehler ist schon viel zu lange offen. Siehe https://github.com/evcc-io/evcc/pull/12866 für einen Vorschlag:

- Charger kann den Minimalstrom des Ladepunkts nach unten und oben überschreiben
- Vehicle kann den Minimalstrom des Ladepunkts nach unten und oben überschreiben. Nach *unten*- um Fehlkonfigurationen zu vermeiden- aber nur dann, wenn der Charger auch dynamische Ströme (`GetMinMaxCurrent()`) unterstützt.

closed 2024-03-11 by andig

hidminds, 2024-03-12:
@andig Deine Änderungen im commit sehen auf den ersten Blick schon mal vielversprechend aus. Vielen Dank! :+1: 
Ich habe mir gerade den aktuellen master gepullt, cross compiled und auf meinem raspi eingespielt. Mein Zoe ist aktuell zu 100% geladen und es ist dunkel. Ist also gerade schlecht mit testen im Real-World-Szenario :wink: . Ich werde deinen Fix die nächsten Tage mit allen Lademodi mal testen und werde Bescheid geben.

hidminds, 2024-03-14:
@andig Habs heute Nachmittag mal testen können.

Ich hatte mit einer "sauberen" (ohne irgendwelche lokale Änderungen) 0.124.9~next Version von vorgestern mit deinen Änderungen getestet.

Funktioniert wunderbar. Test mit meinem Renault Zoe an meiner Elli Charger Pro Wallbox bei etwas Sonnenschein.
Es folgen JSON api/state Abzüge vom Test:

1. Laden startet im PV Mode.
2. Umgeschaltet in den Min+PV Mode.
3. Umgeschaltet in den Now (Schnell) Mode. Huch! Wie geil ist das denn? Die neue Huawei Battery Control aus #12302 entlädt nun bei Schnell meine Hausbatterie nicht mehr! Danke! 👍 
4. Zurück in den PV Mode

[1-pv-0.124.9-next.json](https://github.com/evcc-io/evcc/files/14606730/1-pv-0.124.9-next.json)
[2-minpv-0.124.9-next.json](https://github.com/evcc-io/evcc/files/14606731/2-minpv-0.124.9-next.json)
[3-now-0.124.9-next.json](https://github.com/evcc-io/evcc/files/14606732/3-now-0.124.9-next.json)
[4-pv-0.124.9-next.json](https://github.com/evcc-io/evcc/files/14606733/4-pv-0.124.9-next.json)

Der Zoe wurde durchgängig in allen Modes effektiv mit mindestens ~5,5 kW geladen. Die effective MinCurrent wurde wie am vehicle definiert immer auf minimal 8A eingestellt.

Herzlichen Dank! 🍻

andig, 2024-03-14:
Na gottseidank 😂. Danke fürs Feedback und sorry, dass das so lange gedauert hat. Da hab ich den Wals vor lauter Bäumen nicht gesehen…

andig, 2024-07-03:
@davidgiga1993 hatte der Fix für Dich den Zweck erreicht? https://github.com/evcc-io/evcc/issues/14418 sagt mir, dass nicht alle Fälle abgedeckt sind :(