evcc-io/evcc #11638: Vehicle API extension to request current values from vehicle

MarkusGH, 2024-01-12:
Recently the Porsche API has developed a new "behavioural disorder". Requests are answered by the API server, but i.E. the SOC is not updated for a long time (I have seen 1.5 hrs and more). A refresh using the Porsche APP or [pyporscheconnectapi](https://github.com/CJNE/pyporscheconnectapi), as well as a restart of evcc updates the values, afterwards the server returns the new values.

It would be good, if the verhicle API differentiated between "current" and "cached" values.

When "cached" values are requested the API server decides, if the EV is requested or if cached values are returned.
When "current" values are requested the API server is instructed to contact the EV to fetch current values.

At least the Porsche API has this distinction (in some calls, but any call to the EV will update the cached values anyways).

My further suggestions would be to fetch "current" values while charging and "cached" values while not charging.

andig, 2024-01-12:
> It would be good, if the evcc vehicle API differentiated between "current" and "cached" values.

How?

MarkusGH, 2024-01-12:
> > It would be good, if the evcc vehicle API differentiated between "current" and "cached" values.
> 
> How?

Additional parameter in the API calls?

i.E:

`func (v *Provider) Soc(requestCurrentValue bool) (float64, error) {`

> 
> > My further suggestions would be to fetch "current" values while charging and "cached" values when not charging.
> 
> That would kill the climater detection.

Why?
The suggested change ist to add an option to actively request current values - that should not break anything (apart from waking up the car more often, which is no problem while it is charging).

andig, 2024-01-13:
Wir haben schon ein funktionierendes Caching. Dein Problem hier ist doch, dass der Porsche Server alte Daten liefert. Es geht also nicht um evcc Caching sondern warum, woran man diese alten Daten a) erkennt und dann b) refreshed. Hast Du dazu eine Idee?

MarkusGH, 2024-01-13:
Es geht in der Tat nicht um das evcc Caching (das hatte ich auch nicht geschrieben).
Es geht um das Caching im des Porsche API Servers.

Das Problem ist in der Tat, dass der Porsche Server alte Daten liefert. Macht Sinn, da er wohl nicht für jeden Request das Auto aufwecken will. Eventuell ist das auch so geregelt, dass das Auto Updates aktiv an den Server sendet wenn es wach ist und wenn was passiert. Fakt ist, dass das wohl aktuell nicht richtig funktioniert.

evcc verwendet nach meinem Verständnis Requests, bei denen kein refresh angefordert wird.
Auch das macht Sinn, denn damit könnte man auch Poll Mode "always" benutzen, ohne dass man das Auto permanent weckt.

Wie dem auch sei, man möchte den SOC nicht nur alle 1,5 Stunden aktualisiert haben, denn so gut ist der Estimator auch nicht.

Für Abhilfe gäbe es zwei Möglichkeiten:

1) während des Ladens immer refresh anfordern (die WakeUp Methode macht das).
2) Alte Werte erkennen und bei Bedarf refreshen.

Zu 2) habe ich keine Idee, 1) kann ich zumindest beschreiben...

andig, 2024-01-13:
Den Abruf steuert der Loadpoint, so funktioniert evcc. Mangels 2 ist auch nicht klar, wann 1 notwendig wäre. Daher erstmal wontfix.

closed 2024-01-13 by andig

MarkusGH, 2024-01-13:
> Den Abruf steuert der Loadpoint, so funktioniert evcc. Mangels 2 ist auch nicht klar, wann 1 notwendig wäre. Daher erstmal wontfix.

Deswegen mein Vorschlag, dass der Loadpoint die Fähigkeit bekommt, abgestuft abzurufen:

a) "gib mir was Du hast" (wenn nicht geladen wird)
b) "frag das Auto nach seinem aktuellen Zustand" (solange geladen wird)

andig, 2024-01-13:
Hab ich verstanden, wir haben aber grad keine Kapa eine Sonderlösung für Porsche zu bauen.

MarkusGH, 2024-01-13:
OK - das sehe ich ein.
Ich würde es ja machen, aber es macht keinen Sinn wenn ich anfange API Schnittstellen anzufassen.
Hätte man die Möglichkeit dem Aufruf des vehicle API einen "Wunsch" des Loadpoints oder meinetwegen auch etwas, was den Charger State widerspiegelt, mitzugeben, wäre die eigentliche Implementation ein Klacks.