evcc-io/evcc #11641: NaN value

andig, 2024-01-12:
Ich hab noch einen gefunden:

[image]

```json
{
  "result": {
    "auth": {
      "vehicles": {}
    },
    "battery": [
      {
        "power": -750,
        "controllable": false
      }
    ],
    "batteryCapacity": 0,
    "batteryConfigured": true,
    "batteryDischargeControl": true,
    "batteryEnergy": 0,
    "batteryMode": "unknown",
    "batteryPower": -750,
    "batterySoc": 0,
    "bufferSoc": 0,
    "bufferStartSoc": 0,
    "currency": "EUR",
    "gridConfigured": true,
    "gridCurrents": [
      3,
      2,
      1
    ],
    "gridPower": -1725.774966337804,
    "homePower": 2524.225033662196,
    "loadpoints": [
      {
        "chargePower": 0,
        "chargerFeatureHeating": false,
        "chargerFeatureIntegratedDevice": false,
        "chargerIcon": "",
        "disableThreshold": 0,
        "effectiveLimitSoc": 100,
        "effectiveMaxCurrent": 20,
        "effectiveMinCurrent": 6,
        "effectivePlanSoc": 0,
        "effectivePlanTime": "0001-01-01T00:00:00Z",
        "effectivePriority": 0,
        "enableThreshold": 0,
        "guardAction": "inactive",
        "guardRemaining": 0,
        "limitEnergy": 0,
        "limitSoc": 0,
        "maxCurrent": 20,
        "minCurrent": 6,
        "mode": "now",
        "phaseAction": "inactive",
        "phaseRemaining": 0,
        "phasesActive": 3,
        "phasesConfigured": null,
        "phasesEnabled": 3,
        "planEnergy": 0,
        "planTime": "0001-01-01T00:00:00Z",
        "priority": 0,
        "pvAction": "inactive",
        "pvRemaining": 0,
        "title": "Garage",
        "vehicleCapacity": 0,
        "vehicleClimaterActive": null,
        "vehicleDetectionActive": false,
        "vehicleFeatureOffline": false,
        "vehicleIcon": "car",
        "vehicleName": "audi",
        "vehicleOdometer": 0,
        "vehiclePresent": true,
        "vehicleRange": 0,
        "vehicleSoc": 0,
        "vehicleTargetSoc": 0,
        "vehicleTitle": "Audi A3"
      }
    ],
    "prioritySoc": 0,
    "pv": [
      {
        "power": 5000
      }
    ],
    "pvConfigured": true,
    "pvEnergy": 0,
    "pvPower": 5000,
    "residualPower": 0,
    "siteTitle": "Zuhause",
    "smartCostActive": false,
    "smartCostLimit": 0,
    "smartCostType": "co2",
    "vehicles": {
      "audi": {
        "title": "Audi A3"
      },
      "vehicle_1": {
        "title": "blauer e-Golf"
      },
      "vehicle_2": {
        "title": "weißes Model 3"
      },
      "vehicle_7": {
        "title": "offline"
      }
    }
  }
}
```

naltatis, 2024-01-12:
Ich brauch mehr Kontext. Wann passiert das?

andig, 2024-01-12:
Evtl. wieder ein nicht publizierter Initialwert. Ich fänds gut wenn wir im UI dafür gegen 0 initialisieren könnten?

andig, 2024-01-12:
Hier noch einer aus Demo, siehe Vorzeichen (eigentlich separates Issue):

[image]

naltatis, 2024-01-12:
> Hier noch einer aus Demo, siehe Vorzeichen (eigentlich separates Issue):

Jo, das ist was anderes. Hier bräuchte ich den `sessionCo2PerKWh` Wert.

andig, 2024-01-12:
S.o.- es gibt keinen.

naltatis, 2024-01-12:
> S.o.- es gibt keinen.

Hm, das kann ich mir noch nicht wirklich erklären. Ich hab jetzt doch den 0-default Weg gewählt. Siehe PR.

PR #11642 (2024-01-13): Avoid NaN in UI if values is missing
fixes #11641

Use 0-defaults for price, co2 and solar if no data is available yet.

[image]
closed 2024-01-13 by andig