evcc-io/evcc #11645: cannot create vehicle type 'template':

abc2006, 2024-01-12:
### Describe the bug

bekomme in letzter zeit vermehrt solche Meldungen im Log, auch über mehrer evcc-versionen hinweg. Daraufhin die Anzeige "Turtle (offline)" im Webfrontend. Nach mehreren restarts von evcc läufts dann wieder.

cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: Get "https://owner-api.teslamotors.com/api/1/vehicles": dial tcp: lookup owner-api.teslamotors.com: i/o timeout

Leider sind selbst mit loglevel trace nicht mehr infos im log zu erhalten. 

### Steps to reproduce

1. restart evcc
2.
3.
...

### Configuration details

```yaml
vehicles:
- type: template
  template: tesla 
  title: Turtle  
  accessToken: xxx
  refreshToken:yyy
  vin: LRW3xxxyyy  
  capacity: 82  
  name: turtle
  ##targetSoc: 90

- type: template
  template: offline
  title: M3Flitzer
  capacity: 82  
  name: flitzer 

- type: template
  template: offline
  title: Sabines Maus
  name: SMaus
  capacity: 42
```

### Log details

```text
Jan 12 18:33:02 evcc evcc[670]: [main  ] ERROR 2024/01/12 18:33:02 creating vehicle turtle failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: Get "https://owner-api.teslamotors.com/api/1/vehicles": dial tcp: lookup owner-api.teslamotors.com: i/o timeout
```

### What type of operating system are you running?

Linux

### Version

123.8

davidgiga1993, 2024-01-13:
Von der Fehlermeldung her kann es ja nur ein Netzwerkproblem bei dir oder Tesla sein? Besonders da es über mehrere Versionen auftritt.

andig, 2024-01-13:
> i/o timeout

Tesla antwortet nicht.

closed 2024-01-13 by andig

abc2006, 2024-01-13:
Danke für die Info. Tritt das auch bei anderen Nutzern auf? 
Warum kann er kein Fahrzeug erzeugen, wenn Tesla nicht antwortet? Er erzeugt doch dann "Turtle (offline)"?

Richy0007, 2024-02-02:
Ich hab lange gesucht und die Lösung gefunden

Das API Token und das Template wurde geändert.
Mit neuen Token ist evcc nun auch auf der Tesla Webseite einsehbar.

ein neues API Token bekommt man hier:
https://tesla.evcc.io/

evcc.ymal muss template geändert werden !

vehicles:
  - name: my_car
    type: template
    template: tesla-command
    title: Tesla Modell 3
    accessToken:   xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    refreshToken:  xxxxxxxxxxxxxxxxxxxxxxxxx