evcc-io/evcc #11648: EEbus: SoC from vehicle not correct - estimate overrides actual value

davidgiga1993, 2024-01-12:
### Describe the bug

When starting evcc and connecting the vehicle, the correct SoC is displayed. After ~1 minute it jumps to 99% (which is incorrect).

### Steps to reproduce

1. Connect vehicle or restart evcc
2. Wait

It's not clear to me why this wasn't happening in the past. It just randomly started happening. I've tested this with 2 different evcc versions (see below)

### Configuration details

```yaml
See #11478
```

### Log details

```text
Jan 12 18:43:28 pi4 docker[25246]: evcc  | [lp-1  ] INFO 2024/01/12 19:43:28 car connected
Jan 12 18:43:28 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:43:28 pv timer elapse
Jan 12 18:43:28 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:43:28 pv timer inactive
Jan 12 18:43:28 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:43:28 charger: guard elapse
Jan 12 18:43:28 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:43:28 guard timer inactive
Jan 12 18:43:28 pi4 docker[25246]: evcc  | [lp-1  ] ERROR 2024/01/12 19:43:28 charger vehicle id: must retry
Jan 12 18:43:28 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:43:28 vehicle soc: 64%
Jan 12 18:43:28 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:43:28 set taycan plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:43:58 ----
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:43:58 charge power: 0W
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:43:58 pv power: 0W
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:43:58 grid power: 2329W
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:43:58 grid powers: [385 327 1618]W
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:43:58 grid currents: [2.39 1.7 6.98]A
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:43:58 site power: 1939W
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:43:58 charge currents: [0 0 0]A
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:43:58 charger status: B
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [lp-1  ] ERROR 2024/01/12 19:43:58 charger vehicle id: must retry
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:43:58 soc estimated: 99.07% (vehicle: 64.00%)
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:43:58 vehicle soc: 99%
Jan 12 18:43:58 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:43:58 set taycan plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:44:28 ----
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:44:28 charge power: 0W
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:44:28 pv power: 0W
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:44:28 grid power: 2267W
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:44:28 grid powers: [384 312 1572]W
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:44:28 grid currents: [2.38 1.64 6.79]A
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:44:28 site power: 1877W
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:44:28 charge currents: [0 0 0]A
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:44:28 charger status: B
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [lp-1  ] ERROR 2024/01/12 19:44:28 charger vehicle id: must retry
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:44:28 soc estimated: 99.07% (vehicle: 64.00%)
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:44:28 vehicle soc: 99%
Jan 12 18:44:28 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:44:28 set taycan plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
Jan 12 18:44:58 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:44:58 ----
Jan 12 18:44:58 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:44:58 charge power: 0W
Jan 12 18:44:58 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:44:58 pv power: 0W
Jan 12 18:44:58 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:44:58 grid power: 2287W
Jan 12 18:44:58 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:44:58 grid powers: [383 326 1578]W
Jan 12 18:44:58 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:44:58 grid currents: [2.39 1.69 6.82]A
Jan 12 18:44:58 pi4 docker[25246]: evcc  | [site  ] DEBUG 2024/01/12 19:44:58 site power: 1897W
Jan 12 18:44:58 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:44:58 charge currents: [0 0 0]A
Jan 12 18:44:58 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:44:58 charger status: B
Jan 12 18:44:58 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:44:58 soc estimated: 99.07% (vehicle: 64.00%)
Jan 12 18:44:58 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:44:58 vehicle soc: 99%
Jan 12 18:44:58 pi4 docker[25246]: evcc  | [lp-1  ] DEBUG 2024/01/12 19:44:58 set taycan plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
```

### What type of operating system are you running?

Linux

### Version

Tested in 0.123.8 and 0.123.7

andig, 2024-01-13:
@premultiply hier kommt der Soc vom Charger- könntest Du schauen was der Estimator damit macht?
@davidgiga1993 `estimate: false` als Workaround.

andig, 2024-01-13:
@davidgiga1993 the full log would probably help. Seems the estimator sees a jump and extrapolates that. That jump is missing from your log.

andig, 2024-01-30:
ping @premultiply @davidgiga1993

premultiply, 2024-01-30:
Der berechnete SoC kann nur springen wenn die Energie einen Sprung macht.
Das ist hier ursächlich.

Vielleicht falsche Einheit bei der Messung oder statt geladener Energie wird hier Energiegehalt des Fahrzeugakkus übermittelt?

andig, 2024-02-03:
Looks like we cannot figure out why energy and hence soc jumped. Closing.