evcc-io/evcc #11654: Planner enhancements

andig, 2024-01-13:
This is a tracking issue for various requests that have been made for enhancing the planner:

1. always charge during last hour to pre-heat battery/vehicle (https://github.com/evcc-io/evcc/issues/10950)
2. spread charging as much as possible to avoid peak loads (https://github.com/evcc-io/evcc/issues/11324)
3. integrate PV forecast (https://github.com/evcc-io/evcc/pull/9454)

seki09, 2024-03-03:
Hi. 

Zum Thema Ladeplanung:

Ähnliche Anforderungen kommen auch bald bei uns in Österreich.
Hier erfolgt bald eine Umstellung, wonach Netzgebühren nach der Peak Leistung gerechnet werden, diese kann man sich dann entsprechend zukaufen.

Ich möchte daher anregen ob es nicht besser wäre ein zusätzliches config property einzufügen, welches dem heutigen Max current entspricht, aber nur gilt, wenn nicht per Überschuss geladen wird.

Beispiel: 16A Max current bei pv Modus
9A Max current wenn per Planer oder smarten laden in der Nacht geladen wird.

Somit kann jeder User steuern, wie viel Peak shaving er betreiben will (auch je nachdem wie viel Peak Leistung er erworben hat beim Netzbetreiber) und es wird nicht automatisch auf das Minimum reduziert.

michiiee93, 2024-04-29:
Hier noch zwei Ideen die mir helfen würden:

Idee 1:
Es wäre super, wenn man anstatt einer einzelnen Ladeplanung, mehrere Ladeplanungen einstellen könnte.
Aktuell kann man nur eine einzelne Ladeplanung für wenige Tage im voraus aktivieren.
Man müsste im Ladeplanungs-Dialog einen "Hinzufügen" Button einbauen. Wenn man darauf klickt, taucht eine neue weitere Zeile auf wo man eine weitere Ladeplanung einstellen kann.

Idee 2:
Oder noch besser, wie in Outlook bei einem Serientermin, eine Serienmäßige Ladeplanung.
Use case: Alle zwei Wochen wird Freitags 100% Ladung benötigt. Das würde ich gerne einstellen können.

passuff, 2024-05-16:
Mir würde es tatsächlich vorerst ausreichen, dass ich einen Wochenplan erstelle (wie jetzt auch schon) der sich mit einem gesetzten Haken "Wiederholung" zukünftig wiederholt. Ausnahmen bekomme ich auch anders geregelt..

vkr2791, 2024-05-16:
Das wäre in der Tat eine sehr einfache Erleichtertung, erstmal einfach ein Häkchen "täglich wiederholen" oder so ähnlich, dann könnte man dieses setzten und man wüsste, dass das Auto jeden Morgen einfach z.B. um 06:00 Uhr mit 85% geladen ist.

Dennis1000, 2024-08-16:
Wenn die Ladeplanung als Plugin ausgeführt werden könnte, könnte jeder sich seine eigene Ladeplanung basteln - dann könnte diese in der UI neben PV, PV+Min, Schnell, LadeplanungPlugin:Title zur Auswahl stehen.

andig, 2024-08-16:
Verstehe ich nicht. Wenn du deine eigene Planung machst- wofür brauchts dann ein Plugin? Du kannst evcc ja eh steuern wie du willst?

Dennis1000, 2024-08-17:
Es würden eben alle Inputs bequem zur Verfügung stehen und eine sehr schöne und praktische Oberfläche gibts ja noch dazu. War ja auch nur ne Idee um "usergewünschte" Planungen einfacher zu ermöglichen :-)

andig, 2024-08-17:
Würde? Auch das geht doch schon. Du musst einfach einen "Tarif" anlegen mit dem Du dem Planner Deiner gewünschte Kostenfunktion unterschiebst.

Dennis1000, 2024-08-17:
Wenn ich damit "Spread charging" und sowas machen kann, dann hab ich das Tarifzeugs einfach nicht richtig verstanden, da werde ich wohl noch etwas code-einlesen machen müssen, danke für den Hinweis

Coernel82, 2024-11-19:
Ich habe eine **Ladeplanung mit Solcast** geschrieben, die ich die nächsten Tage bis Wochen für Betatesting öffentlich mache. Was mir aber aktuell leider fehlt ist ein Api-Zugraff für die Hausbatterie für ``batteryGridChargeLimit`` - das was ich genutzt habe, nämlich ``/api/smartcostlimit/{upper_limit_price_battery}`` gilt ja für alles andere, aber nicht die Hausbatterie. @andig gibt es da was? Wäre es aufwändig für dich, das zu implementieren? (Bitte, bitte!)

Hier mal ein paar Infos:

# SmartCharge - Intelligente Optimierung der Ladung von E-Fahrzeugen und Hausbatterien 🚗🔋🌞

SmartCharge ist ein fortschrittliches Python-basiertes Programm, das entwickelt wurde, um den Ladevorgang für Elektrofahrzeuge (EVs) und Hausbatterien intelligent zu optimieren. Durch die Integration von Echtzeitdaten und intelligenten Prognosen aus verschiedenen Quellen wie Solarwettervorhersagen und dynamischen Strompreisen ermöglicht SmartCharge eine effiziente und kostengünstige Energienutzung in Ihrem Zuhause.

## Hauptfunktionen 🌟

- **Intelligente EV-Ladungsoptimierung**: Berechnet den erforderlichen Ladezustand (SoC) für jedes EV basierend auf detaillierten Nutzungsplänen, die sowohl regelmäßige als auch einmalige Fahrten umfassen, sowie präzisen Wettervorhersagen, welche benötigt werden, um den Energieverbrauch des Autos anzupassen.
  
- **Fortschrittliches Hausbatterie-Management**: Verarbeitet umfassende Daten von Hausbatterien, um die Batterie prognosebasiert mit günstigem Netzstrom zu laden, um mögliche Unterdeckungen auszugleichen. Hierbei wird der prognostizierte Verbrauch (Heizung, Grundlast) berücksichtigt.

- **Heizungsverbrauch** wird an Hand des Energieausweises ermittelt und indirekte solare Strahlung als proportionaler Faktor zur kWP-Leistung der Solaranlage mit berücksichtigt.
  
- **Nutzung von Energieüberschüssen aus Solarstrom**: Identifiziert und nutzt überschüssige Energie aus der Solaranlage effizient zur Ladung von EVs und Hausbatterien. Die überschüssige Energie wird mithilfe der **Solcast API** präzise prognostiziert, um eine maximale Nutzung der selbst erzeugten Energie zu gewährleisten.
  
- **Kosteneffizienz durch dynamische Preisoptimierung**: Analysiert aktuelle und prognostizierte Strompreise, um die günstigsten Zeitfenster für das Laden zu identifizieren. Setzt obere Preisgrenzen basierend auf minimalen Strompreisen, um die Hausbatterie so zu laden, dass am nächsten Tag keine Unterdeckung entsteht. _**--> Hier fehlt mir eine API POST-Möglichkeit!**_
  

- **Prognosebasierte Energieverwaltung**: Nutzt fortschrittliche Algorithmen und historische Daten, um zukünftige Energiebedarfe und -angebote vorherzusagen, wodurch eine proaktive und optimierte Energieverteilung möglich wird.

- **automatische Präzisierung der Prognose:** die Heizenergie wird durch einen Korrekturfaktor jeden Tag genauer prognostiziert - so zumindest die Idee. Ob das klappt, wird sich herausstellen. Es wird berücksichtigt, ob es Sommer oder Winter ist, welches automatisch an Hand einer Temperaturschwelle erkannt wird.
  
- **Umfassende Protokollierung und Berichterstattung**: Führt detaillierte Logs über alle Prozesse und Berechnungen, die eine transparente Nachverfolgung und Analyse der Energieflüsse ermöglichen.

 🚗🔋🌞

andig, 2024-11-19:
Das hat nicht wirklich was mit dem Planner zu tun, OT? Welches Api Dir da fehlt (Doku gesichtet?) verstehe ich leider nicht.

Coernel82, 2024-11-19:
In der Doku https://docs.evcc.io/docs/integrations/rest-api gibt es nichts entsprechendes, kann mir aber vorstellen, dass es da noch was gibt, was experimentell ist. Würde mir sehr helfen, also API um das Kostenlimit für Netzladen für die Hausbatterie zu setzen.
Bezüglich dem Planer: doch, ich bin mir ziemlich sicher, dass es etwas damit zu tun hat. Ich habe die Logik entwickelt, PV Prognosen (Stichwort Solcast!), Heizenergie, Grundlast pro Stunde zu berechnen und den Überschuss dann für die Ladeplanung zu berücksichtigen.
Z.B. braucht der Opel noch 30 kW bis zur Abfahrt, PV liefert bis dahin 25, Haus benötigt aber 5: 10 kW werden mit günstigstem Tarif geladen.
So ähnlich mit der Hausbatterie.

VolkerK62, 2024-11-20:
> kann mir aber vorstellen, dass es da noch was gibt, was experimentell ist.

schau mal hier https://github.com/evcc-io/evcc/blob/master/server/http.go

Coernel82, 2024-11-20:
> > kann mir aber vorstellen, dass es da noch was gibt, was experimentell ist.
> 
> schau mal hier https://github.com/evcc-io/evcc/blob/master/server/http.go
> 

POST /api/batterygridchargelimit/{value:-?[0-9.]+} to set the battery grid charge limit

Klingt gut für mich. Ist es schon implementiert? Dann probiere ich es morgen mal aus, dann paar Bugfixes und dann public Beta! Und vielleicht könnt ihr was von der Logik nutzen!

andig, 2024-11-20:
Mach bitte einen neuen Post dafür, hier OT.  Ich ziehe die Doku fürs api nach.

andig, 2024-11-24:
Fixed in https://github.com/evcc-io/docs/pull/677