evcc-io/evcc #11666: Fahrzeug wird nicht mehr erkannt

cburghardt, 2024-01-13:
### Describe the bug

Ich habe soeben von 0.123.5 auf 0.123.8 aktualisiert.
Seit dem wird keines der Fahrzeuge mehr erkannt und es wird kein Fahrzeug als verbunden angezeigt.

### Steps to reproduce

Auto angesteckt, keinerlei Logausgabe

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 2dd502a030abe216cc6cdde4984bc60a198e279b6e5a52e250212eea441cb430

interval: 30s # control cycle interval

meters:
  - name: pv
    type: custom
    power:
      source: mqtt
      topic: homeassistant/sensor/power_solar_generation/state
      timeout: 24h
  - name: grid      
    type: custom
    power:  
      source: mqtt
      topic: homeassistant/sensor/solaredge_modbus_m1_ac_power/state
      scale: -1
      timeout: 10m
    currents:
      - source: mqtt    
        topic: homeassistant/sensor/solaredge_modbus_m1_ac_current_a/state
        timeout: 8h
      - source: mqtt    
        topic: homeassistant/sensor/solaredge_modbus_m1_ac_current_b/state
        timeout: 8h
      - source: mqtt    
        topic: homeassistant/sensor/solaredge_modbus_m1_ac_current_c/state
        timeout: 8h
  - name: battery
    type: custom
    power:
      source: mqtt
      topic: homeassistant/sensor/solaredge_modbus_b1_dc_power/state
      scale: -1
      timeout: 24h 
    soc:
      source: mqtt
      topic: homeassistant/sensor/solaredge_modbus_b1_state_of_energy/state
      timeout: 24h

chargers:
  - name: wallbox
    type: template
    template: go-e-v3
    host: go-echarger-217527

site:
  title: Home
  residualPower: 100
  maxGridSupplyWhileBatteryCharging: 50
  meters:
    pv: pv
    grid: grid
    battery: battery  

loadpoints:
  - title: Garage
    charger: wallbox
    mincurrent: 6
    maxcurrent: 32

mqtt:
  broker: 192.168.178.109:1883
  topic: evcc 
  clientid: evcc

vehicles:
  - name: disco
    type: template
    title: Discovery Sport
    template: jaguar-landrover
    phases: 1  
    maxcurrent: 16
  - name: etron
    type: template
    title: ETron
    template: audi
    phases: 3  
    maxcurrent: 20
```

### Log details

```text
Jan 13 14:21:00 evcc evcc[15123]: [main  ] INFO 2024/01/13 14:21:00 evcc 0.123.8
Jan 13 14:21:00 evcc evcc[15123]: [main  ] INFO 2024/01/13 14:21:00 using config file: /etc/evcc.yaml
Jan 13 14:21:00 evcc evcc[15123]: [main  ] INFO 2024/01/13 14:21:00 starting ui and api at :7070
Jan 13 14:21:00 evcc evcc[15123]: [db    ] INFO 2024/01/13 14:21:00 using sqlite database: /var/lib/evcc/evcc.db
Jan 13 14:21:00 evcc evcc[15123]: [mqtt  ] INFO 2024/01/13 14:21:00 connecting evcc at tcp://192.168.178.109:1883
Jan 13 14:21:00 evcc evcc[15123]: [mqtt  ] DEBUG 2024/01/13 14:21:00 tcp://192.168.178.109:1883 connected
Jan 13 14:21:00 evcc evcc[15123]: [jlr   ] WARN 2024/01/13 14:21:00 new device id generated, add `deviceid` to config: 9b5816cc-374a-4ca3-96b2-9f0e88a3d07e
Jan 13 14:21:02 evcc evcc[15123]: [lp-1  ] DEBUG 2024/01/13 14:21:02 charge total import: 1402.946kWh
Jan 13 14:21:02 evcc evcc[15123]: [site  ] INFO 2024/01/13 14:21:02 site config:
Jan 13 14:21:02 evcc evcc[15123]: [site  ] INFO 2024/01/13 14:21:02   meters:      grid ✓ pv ✓ battery ✓
Jan 13 14:21:02 evcc evcc[15123]: [site  ] INFO 2024/01/13 14:21:02     grid:      power ✓ energy ✗ currents ✓
Jan 13 14:21:02 evcc evcc[15123]: [site  ] INFO 2024/01/13 14:21:02     pv 1:      power ✓ energy ✗ currents ✗
Jan 13 14:21:02 evcc evcc[15123]: [site  ] INFO 2024/01/13 14:21:02     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
Jan 13 14:21:02 evcc evcc[15123]: [site  ] INFO 2024/01/13 14:21:02   vehicles:
Jan 13 14:21:02 evcc evcc[15123]: [site  ] INFO 2024/01/13 14:21:02     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
Jan 13 14:21:02 evcc evcc[15123]: [site  ] INFO 2024/01/13 14:21:02     vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
Jan 13 14:21:02 evcc evcc[15123]: [lp-1  ] INFO 2024/01/13 14:21:02 loadpoint 1:
Jan 13 14:21:02 evcc evcc[15123]: [lp-1  ] INFO 2024/01/13 14:21:02   mode:        now
Jan 13 14:21:02 evcc evcc[15123]: [lp-1  ] INFO 2024/01/13 14:21:02   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
Jan 13 14:21:02 evcc evcc[15123]: [lp-1  ] INFO 2024/01/13 14:21:02   meters:      charge ✓
Jan 13 14:21:02 evcc evcc[15123]: [lp-1  ] INFO 2024/01/13 14:21:02     charge:    power ✓ energy ✓ currents ✓
Jan 13 14:21:02 evcc evcc[15123]: [lp-1  ] DEBUG 2024/01/13 14:21:02 phase timer inactive
Jan 13 14:21:02 evcc evcc[15123]: [lp-1  ] DEBUG 2024/01/13 14:21:02 pv timer inactive
Jan 13 14:21:02 evcc evcc[15123]: [lp-1  ] DEBUG 2024/01/13 14:21:02 guard timer inactive
Jan 13 14:21:02 evcc evcc[15123]: [lp-1  ] DEBUG 2024/01/13 14:21:02 max charge current: 6A
Jan 13 14:21:02 evcc evcc[15123]: [site  ] DEBUG 2024/01/13 14:21:02 ----
Jan 13 14:21:02 evcc evcc[15123]: [lp-1  ] DEBUG 2024/01/13 14:21:02 charge power: 3515W
Jan 13 14:21:02 evcc evcc[15123]: [site  ] DEBUG 2024/01/13 14:21:02 pv power: 115W
```

### What type of operating system are you running?

Linux

VolkerK62, 2024-01-13:
>Seit dem wird keines der Fahrzeuge mehr erkannt und es wird kein Fahrzeug als verbunden angezeigt.

das sind aber 2 verschiedene Sachverhalte.
Die Anzeige "verbunden" kommt von der Wallbox, nicht vom Auto.
Was sagen denn `evcc charger` und `evcc vehicle`?

VolkerK62, 2024-01-13:
`Jan 13 14:21:00 evcc evcc[15123]: [jlr   ] WARN 2024/01/13 14:21:00 new device id generated, add `deviceid` to config: 9b5816cc-374a-4ca3-96b2-9f0e88a3d07e`
Die Warnung für den Jaguar hast du bemerkt?

cburghardt, 2024-01-13:
evcc charger:
```
[main  ] INFO 2024/01/13 19:09:09 evcc 0.123.8
[main  ] INFO 2024/01/13 19:09:09 using config file: /home/carsten/evcc.yaml
[db    ] INFO 2024/01/13 19:09:09 using sqlite database: /home/carsten/.evcc/evcc.db
[mqtt  ] INFO 2024/01/13 19:09:09 connecting evcc at tcp://192.168.178.109:1883
Power:          0W
Energy:         1406.3kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 233V 228V 231V
Charge status:  B
Enabled:        true
Charged:        3.4kWh
Identifier:     <none>
```

evcc vehicle
```
[main  ] INFO 2024/01/13 19:09:54 evcc 0.123.8
[main  ] INFO 2024/01/13 19:09:54 using config file: /home/carsten/evcc.yaml
[db    ] INFO 2024/01/13 19:09:54 using sqlite database: /home/carsten/.evcc/evcc.db
[mqtt  ] INFO 2024/01/13 19:09:54 connecting evcc at tcp://192.168.178.109:1883
[jlr   ] WARN 2024/01/13 19:09:54 new device id generated, add `deviceid` to config: 323e96e8-70a0-4317-9746-3385bf140849
Soc:           100%
Capacity:      0.0kWh
Charge status: A
Range:         33km
Odometer:      22447km
Finish time:   not available
Position:      48.30609166666667,10.965939722222222
Features:      []
```
Die Warnung vom Landrover war schon immer da, allerdings wird auch kein key "deviceid" in der config akzeptiert. Zumindest hatte ich das mal vor Monaten probiert.

Tihmann, 2024-01-13:
Das gleiche Verhalten hier auch nach Update auf  0.123.8 . Kein Erkennen von Fahrzeugen in der GUI, keine Steuerung in der GUI mehr möglich. evcc charger & evcc vehicle  zeigen Werte an

cburghardt, 2024-01-13:
Was auffällig ist: normal kommen alle paar Sekunden debug Meldungen von den mqtt Werten rein. Das hat komplett aufgehört. Es passiert schlichtweg nix im log.

Tihmann, 2024-01-13:
Verbrauchswerte werden bei mir auch nicht mehr angezeigt, kommen über mqtt

andig, 2024-01-14:
> Kein Erkennen von Fahrzeugen in der GUI, keine Steuerung in der GUI mehr möglich. 

Was soll das heissen?

andig, 2024-01-14:
> Wird ein Auto angesteckt, wird es nicht erkannt, das Log zeigt trotzdem debug level keinerlei Ausgabe.

Wo sehe ich das in Deinem Log? Timestamp?

cburghardt, 2024-01-14:
> > Kein Erkennen von Fahrzeugen in der GUI, keine Steuerung in der GUI mehr möglich.
> 
> Was soll das heissen?

Aus meiner Perspektive: in der GUI werden keinerlei Verbrauchswerte angezeigt und kein aktives Fahrzeug.

cburghardt, 2024-01-14:
So sehen die log Ausgaben mit der Version 0.123.5 aus, während es bei der Version .8 still ist:

```
Jan 14 09:57:45 evcc evcc[15446]: [site  ] DEBUG 2024/01/14 09:57:45 ----
Jan 14 09:57:45 evcc evcc[15446]: [lp-1  ] DEBUG 2024/01/14 09:57:45 charge power: 0W
Jan 14 09:57:45 evcc evcc[15446]: [site  ] DEBUG 2024/01/14 09:57:45 pv power: 246W
Jan 14 09:57:45 evcc evcc[15446]: [site  ] DEBUG 2024/01/14 09:57:45 battery soc: 9%
Jan 14 09:57:45 evcc evcc[15446]: [site  ] DEBUG 2024/01/14 09:57:45 battery power: 0W
Jan 14 09:57:45 evcc evcc[15446]: [site  ] DEBUG 2024/01/14 09:57:45 grid power: 2096W
Jan 14 09:57:45 evcc evcc[15446]: [site  ] DEBUG 2024/01/14 09:57:45 grid currents: [4.6 3.4 3.8]A
Jan 14 09:57:45 evcc evcc[15446]: [site  ] DEBUG 2024/01/14 09:57:45 site power: 2196W
Jan 14 09:57:45 evcc evcc[15446]: [lp-1  ] DEBUG 2024/01/14 09:57:45 charge currents: [0 0 0]A
Jan 14 09:57:45 evcc evcc[15446]: [lp-1  ] DEBUG 2024/01/14 09:57:45 charge total import: 1406.258kWh
Jan 14 09:57:45 evcc evcc[15446]: [lp-1  ] DEBUG 2024/01/14 09:57:45 charger status: B
```

andig, 2024-01-14:
Dazwischen sind noch 2 Versionen. Mit welcher kommt das Problem?

cburghardt, 2024-01-14:
> Dazwischen sind noch 2 Versionen. Mit welcher kommt das Problem?

Die Version .7 geht noch, das Problem kommt also mit der .8 Version. Und ist wohl irgendwo beim mqtt zu suchen.

naltatis, 2024-01-14:
Wir scheinen hier generell ein Asynchronitätsproblem zu haben. Ich hab nen PR angelegt, in dem man das gut sehen kann. Allerdings noch kein Fix für das Problem.

andig, 2024-01-14:
> Das würde ich die Entwickler dann vermutlich auch fragen.

Würde ich mit gerne anschauen falls Du mir Deine vollständige Config an info@evcc.io schicken würdest.

PR #11675 (2024-01-15): Fix inconsistant vehicle publishing
fixes: #11666

I made a few simplifications to the code in order to make get to the root cause.

- 🗑️ removed `vehiclePresent` since its redudant to the existance of `vehicleName`
- 🗑️ removed `vehicleTitle` since it should be lookedup in `vehicles` via `vehicleName`
- 🪲 some minor UI code cleanups and error handlings

The issue seems to be out-of-sync publishing. The `.publish()` calls are executed in the right order but are sometimes processed out of sync.

**Reproduce:**
```
go run main.go --log TRACE` (using `demo.yaml`)
```

In one time of three the err [... cut]