evcc-io/evcc #11680: go-E charger : Änderung und Anzeige des EVCC Lademodus über Button und LED Farbe

flushbug, 2024-01-14:
Auf Basis der Diskussion im Slack hier der entsprechende Feature Request. [link](https://evccgroup.slack.com/archives/C01321PUJAD/p1704582769700279)

**Feature 1 : Anzeige des aktuellen EVCC Lademodus über die LED Farbe am go-E Charger.** 

Die Farbe kann über die API Keys "cid" (Farbe im Idle Mode) sowie "cch" (Farbe im Lademodus) über RGB Hex Werte gesetzt werden.

_Programmablauf :_ 
```
Bei Änderung des EVCC Lademodus :

Wenn EVCC Lademodus = now -> "cch" und "cid" = "#0000FF" 
' LED Farbe blau 
Wenn EVCC Lademodus = pv -> "cch" und "cid" = "#00FF20" 
' LED Farbe grün
Wenn EVCC Lademodus = minpv -> "cch" und "cid" = "#FFA400" 
' LED Farbe orange
Wenn EVCC Lademodus = off -> "cch" und "cid" = "#FF0000" 
' LED Farbe rot
```

(Die Farben sind natürlich nur ein Vorschlag und könnten z.B. über die config anpassbar gemacht werden)

**Feature 2 : Wechseln des EVCC Lademodus über den Hardware Button am go-E** 
Der Go-E Button wird üblicherweise zum durchschalten vordefinierter maximaler Ladeströme verwendet. Zusätzlich ist ein Druck auf den Button aber auch über die API abfragbar. Sobald der Button gedrückt wurde ist unter dem Key "lbp" ein Zeitstempel (Systemzeit) des letzten Knopfdrucks auslesbar. Der Zeitstempel ist unwichtig aber die Änderung des Werts kann als trigger benutzt werden.

Bei jedem Knopfdruck soll der EVCC Lademodus in definierter Reihenfolge weiter geschaltet werden. Zur Vereinfachung würde ich nur die beiden häufigen Modi "PV" und "now" anwählbar machen. Ließe sich aber z.B. auch konfigurierbar machen. Ist einer der anderen Modi "off" oder "minpv" aktiv wird beim Knopfdruck auf "now" geschaltet.

Um den Button ohne Seiteneffekte verwenden zu können müssen die internen Strompegel über die API jeweils auf 0A gesetzt werden damit das Feature deaktiviert wird. Tut man dies nicht schaltet der Charger die internen Limits mit jedem Kopfdruck weiter und begrenzt zusätzlich zu EVCC den Ladestrom auf das jeweils zufällig gesetzte Limit. 

EVCC kann diese Konfiguration bei der Initialisierung über den apikey clp = [0,0,0,0,0] durchführen (setzen aller 5 Ladestufen auf 0A deaktiviert das feature)
Alternativ gibt es über die API die Möglichkeit den Wechsel des Ladestroms per Knopfdruck über den key "bac" = 0 zu deaktivieren. Ich habe dies getestet aber es führt beim Kopfdruck zu einer "Fehlermeldung" in Form von 5 roten LEDs was den Anwender sicher verwirrt. Die sinnvollere Lösung ist das setzen der Ladestufen auf 0A.

_Programmablauf :_ 
```
Bei Initialisierung von EVCC 
"clp" = [0,0,0,0,0] 
'deaktivierung der Strombegrenzung über Knopfdrücke

Bei Änderung von "lbp" 
' knopf wurde gedrückt

wenn EVCC mode = pv -> evcc mode = now
' vorheriger mode pv -> wechsel auf now
wenn EVCC mode = now -> evcc mode = pv
' vorheriger mode now -> wechsel auf pv
default -> evcc mode = now
'  vorheriger mode off oder minpv -> wechsel auf now
```

**Feature 3 : Anzeige des freigegebenen Stroms über Anzahl leuchtender LEDs am charger (aktuell noch nicht umsetzbar)** 

Schick wäre es noch den aktuell über EVCC freigegebenen Strom über mehrere oder weniger leuchtende LEDs am charger zu signalisieren. Dies ist aber über die API aktuell aber noch nicht steuerbar. Ich habe einen entsprechenden Feature Request im Go-E Github gestellt.
Der go-E hat dieses Feature selbst eingebaut wenn man die eingebauten Stromstufen verwendet (z.B. Level 16A bei einem 32A charger bedeutet die hälfte der LEDs leuchten usw. Dies jetzt aber über die aktuelle API durch permanentes Anpassen der Stromstufen "clp" umzusetzen wäre sehr aufwändig und könnte auch zu Konflikten mit der EVCC Regelung führen wenn es nicht 100% richtig gebaut ist.

andig, 2024-01-14:
Schön wäre, cch/cid vllt. Nach Helligkeit zu unterscheiden (in der Annahme dass „idle“ beim Go-E nicht freigegebene Ladung bedeutet)? Knopf kann ich mir nicht vorstellen, da wir den dafür dauernd pollen müssten- leider meldet sich der Knopf ja nicht bei uns.

flushbug, 2024-01-14:
Die Helligkeit kann man nur global verstellen und nicht individuell. Idle bedeutet freigegeben aber kein Fahrzeug angesteckt. Es gibt die 4 LED zustände Idle, waitcar, Charging und finished. Vermutlich müsste man dann auch noch waitcar mit der richtigen Farbe beschreiben. Wobei für mich der Hauptusecase eigentlich idle ist. Man geht zum Charger und möchte das Auto anstecken, dabei wissen welcher Mode gerade aktiv ist und diesen ggfs. über den knopfdruck in den gewünschten Modus bringen. Gerade jetzt im Winter wo meist keine PV Leistung hat und man das Auto einfach nur voll machen möchte ohne groß aufs Handy zu schauen etc. 

So 100% zufrieden bin ich mit den Farben alleine auch noch nicht. 4 Modi von EVCC über Farben zu unterscheiden ist ohne weitere Erklärung nicht easy. Zusätzlich die Anzahl der LEDs zu verwenden wäre natürlich schicker (wenige LEDs an -> PV mode und wenig Power / alle LEDs an = Full power freigegeben)

Ich verstehe das Problem mit dem Button und notwendigem pollen. Das ist unschön. Ich habe das ganze Feature bei mir gerade über MQTT und nicht per Rest-API in Home Assistant eingebunden. Da bekomme ich dann einen Trigger weil der go-E aktiv eine Message mit dem neuen Zeitstempel an den MQTT Broker schickt sobald der Knopf gedrückt wurde.

andig, 2024-01-14:
Wir nutzen das HTTP API, neu implementieren möchte ich es nicht.

> Die Helligkeit kann man nur global verstellen und nicht individuell. 

ok

> Es gibt die 4 LED zustände Idle, waitcar, Charging und finished.

Je nach Modus (außer off) könnte man damit auch noch zeigen, ob evcc gerade lädt (also Charging in anderem Farbton). Dann könnte man außer Charging eine weniger gesättigte Farbe nehmen? Wäre hilfreich, nochmal die komplette Tabelle aufzumalen.

naltatis, 2024-01-15:
Ich werd' hier in ein paar Wochen nen go-E Charger in Betrieb nehmen. Kann das dann gerne am lebenden Objekt entwickeln und testen. Aber falls jemand das vorher schon bauen mag, freu ich mich natürlich noch mehr 😂

flushbug, 2024-01-18:
Kleines update zu meinen Experimenten zu dem Thema. Scheinbar gibt es MQTT erst ab Go-E Hardware Version v3 d.h. Um kompatibel zu älteren Chargern zu bleiben müssten man ohnehin die HTTP API verwenden. 

Wie häufig fragt ihr die HTTP API denn aktuell ab ? Im Prinzip könnte man den Button auch jedes mal mit auslesen aber es würde eben einfach länger dauern bis die Änderung wirksam wird. Da beim Buttonpress ein Zeitstempel hinterlegt d.h. „Hochgezählt“ wird bekommt man den Knopfdruck auch durch Vergleich von „neuer != alter Zeitstempel“ mit wenn seltener abgefragt wird. Die User Experience wäre aber vermutlich nicht die beste wenn nicht direkt beim Knopfdruck etwas passiert. Das Abfrageintervall sollte daher schon im einstelligen Sekundenbereich liegen.

menkej, 2024-02-12:
@flushbug die Einschränkung, dass man die Leistung nicht mehr über den Button einstellen kann, gilt doch nur, wenn "im Netz" und mit EVCC verbunden. Sobald man mobil unterwegs ist, funkt ja EVCC nicht mehr dazwischen, richtig?

flushbug, 2024-02-12:
@menkej nein, die durch den go-e durchschaltbaren leistungsstufen müssen über die goe App deaktiviert werden, damit sich diese nicht mit von Evcc gesteuerten maximalströmen überlagern. Diese Einstellung bleibt bestehen auch wenn Evcc nicht mehr verfügbar ist und müsste manuell wieder per App aktiviert werden.

motze92, 2024-02-12:
> > Scheinbar gibt es MQTT erst ab Go-E Hardware Version v3 d.h. Um kompatibel zu älteren Chargern zu bleiben müssten man ohnehin die HTTP API verwenden.
> 
> Bei der EVCC Kofiguration wird eh zwischen den älteren und V3 unterschieden. Wenn die älteren da was nicht unterstützen, dann ist es natürlich schade, aber man kann es sauber zumindest für die Neuen umsetzen. Ich bin mir aber ziemlich sicher, dass bei meinem alten Go-e aus 2020 (glaube der müsste V2 sein) auch eine MQTT Option auswählbar war. Müsste ich mir aber ggfs. nochmal ansehen. Haber aber auch noch 2 unbenutzte Gemini (1x fix und 1x CEE 32A rot) da. Falls es bei der Entwicklung helfen würde, könnte ich auch welche von denen verleihen oder verkaufen.
> 
> Als Kompromiss für das Polling würde ich evtl. vorschlagen, dass nur Knopfdrücke in einer bestimmten Zeit nach dem Anstecken berücksichtigt werden. Und bei den Farben könnte man ja auch mit wechselnden arbeiten. z.B. ein mehr oder weniger schnelles Blinken für sofortiges Laden oder so.

Was spricht eigentlich gegen das Polling? Frage bei einem anderem Projekt den Status des Buttons auch alle 2 Sekunden ab. Laut Go-E Support ist das für sie auch kein Problem.

fuxae, 2024-02-18:
> @menkej nein, die durch den go-e durchschaltbaren leistungsstufen müssen über die goe App deaktiviert werden, damit sich diese nicht mit von Evcc gesteuerten maximalströmen überlagern. Diese [...]

Da geht es aber ja nur um die vordefinierten Ampere. In der App hat man trotzdem noch den Schieberegler, den man bedienen kann. Dann hat man halt keinen blauen Kreis mehr worauf man drücken kann, sondern den Schieberegler. Kein großer Unterschied.

Obwohl man da ja sowieso nicht mehr rumspielen sollte, wenn evcc die Steuerung übernimmt. Aber wenn man aus welchem Grund auch immer, evcc mal kurz deaktiviert, kann man das easy darüber einstellen, ohne dass man die vordefinierten Stärken wieder aktivieren müsste. 

Ich habe die Funktion mit dem Knopf bereits über HA umgesetzt, auf den Knopf drücken stellt bei mir von PV auf sofort um. Funktioniert wunderbar.

fuxae, 2024-03-07:
> 
> wie hast du das mit dem Knopf in HA realisiert? wäre perfekt für mich.

Das ist relativ einfach.
Im go-eCharger aktivierst du mqtt (das funktioniert natürlich auch parallel zur "normalen" Anbindung in evcc) und in HA gibt es eine custom Integration für den go-eCharger basierend auf MQTT. Natürlich kannst du die topics aber auch manuell abfragen.

Dadurch bekommst du einen sensor "last Button pressed" (mqtt topic lbp) und du erstellst eine Automatisierung, sobald sich dieser Status verändert schickst du per API einen Befehl an evcc

domii666, 2024-03-07:
> Das ist relativ einfach. Im go-eCharger aktivierst du mqtt (das funktioniert natürlich auch parallel zur "normalen" Anbindung in evcc) und in HA gibt es eine custom Integration für den go-eCharger [...]

ok ist soweit klar, nur wie schicke ich per API?

fuxae, 2024-03-07:
> 
> ok ist soweit klar, nur wie schicke ich per API?

Schau mal in die evcc Doku. Da gibt's die Referenz auf die API. 
Ich habe es ebenfalls mit MQTT gelöst. Sprich, in der HA Automatisierung dann den Dienst mqtt.publish ausführen mit Topic und Payload für die Änderung des Lademodus.

flushbug, 2024-03-07:
Anbei mein Code. Hoffe es ist klar und funktioniert so.

HA MQTT Sensordefinition -> configuration.yaml

```
mqtt:
  - sensor:
      name: "go-E Buttonpress"
      state_topic: "go-eCharger/lbp"

  - sensor:
      name: "Lademodus Wallbox"
      state_topic: "evcc/loadpoints/1/mode"
      json_attributes_topic: "evcc/loadpoints/1/mode"
      json_attributes_template: >
        {"chargemode":"{{value}}"}
```

HA Automation Code/YAML für den Mode Switcher

```
alias: Go-E Button EVCC Mode switcher
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.go_e_buttonpress
    alias: Capture go-e Charger buttonpress
condition: []
action:
  - choose:
      - conditions:
          - condition: state
            entity_id: sensor.lademodus_wallbox
            attribute: chargemode
            state: now
        sequence:
          - service: mqtt.publish
            metadata: {}
            data:
              qos: "0"
              topic: evcc/loadpoints/1/mode/set
              retain: true
              payload: pv
            enabled: true
            alias: Set chargemode to pv
        alias: Changing chargemode from now to pv
      - conditions:
          - condition: state
            entity_id: sensor.lademodus_wallbox
            attribute: chargemode
            state: pv
        sequence:
          - service: mqtt.publish
            metadata: {}
            data:
              qos: "0"
              topic: evcc/loadpoints/1/mode/set
              retain: true
              payload: now
            enabled: true
            alias: Set chargemode to now
        alias: Changing chargemode from pv to now
    default:
      - service: mqtt.publish
        metadata: {}
        data:
          qos: "0"
          topic: evcc/loadpoints/1/mode/set
          retain: true
          payload: now
        enabled: true
        alias: Set chargemode to now (previous state was off or minpv)
mode: single
```

Code für den Farbwechsel

```
alias: EVCC charging mode indicator -> go-E
description: EVCC charging mode indicator -> go-E
trigger:
  - alias: Capture EVCC mode change
    platform: state
    entity_id:
      - sensor.lademodus_wallbox
condition: []
action:
  - choose:
      - conditions:
          - condition: state
            entity_id: sensor.lademodus_wallbox
            attribute: chargemode
            state: pv
        sequence:
          - alias: Set Go-E Charger idle colour to green
            service: mqtt.publish
            metadata: {}
            data:
              qos: "0"
              retain: true
              topic: go-eCharger/cid/set
              payload: "\"#00FF20\""
          - alias: Set Go-E Charger charging colour to green
            service: mqtt.publish
            metadata: {}
            data:
              qos: "0"
              retain: true
              topic: go-eCharger/cch/set
              payload: "\"#00FF20\""
        alias: Chargemode pv
      - conditions:
          - condition: state
            entity_id: sensor.lademodus_wallbox
            attribute: chargemode
            state: now
        sequence:
          - alias: set go-e Charger idle colour to blue
            service: mqtt.publish
            metadata: {}
            data:
              qos: "0"
              retain: true
              topic: go-eCharger/cid/set
              payload: "\"#004FFF\""
          - alias: set go-e Charger charging colour to blue
            service: mqtt.publish
            metadata: {}
            data:
              qos: "0"
              retain: true
              topic: go-eCharger/cch/set
              payload: "\"#004FFF\""
        alias: Chargemode now
      - conditions:
          - condition: state
            entity_id: sensor.lademodus_wallbox
            attribute: chargemode
            state: "off"
            alias: Chargemode off
        sequence:
          - alias: set go-e Charger idle colour red
            service: mqtt.publish
            metadata: {}
            data:
              qos: "0"
              retain: true
              topic: go-eCharger/cid/set
              payload: "\"#FF0000\""
          - alias: set go-e Charger charging colour to red
            service: mqtt.publish
            metadata: {}
            data:
              qos: "0"
              retain: true
              topic: go-eCharger/cch/set
              payload: "\"#FF0000\""
        alias: Chargemode off
      - conditions:
          - alias: Chargemode minpv
            condition: state
            entity_id: sensor.lademodus_wallbox
            attribute: chargemode
            state: minpv
        sequence:
          - alias: set go-e Charger idle colour orange
            service: mqtt.publish
            metadata: {}
            data:
              qos: "0"
              retain: true
              topic: go-eCharger/cid/set
              payload: "\"#FFA400\""
          - alias: set go-e Charger charging colour to orange
            service: mqtt.publish
            metadata: {}
            data:
              qos: "0"
              retain: true
              topic: go-eCharger/cch/set
              payload: "\"#FFA400\""
        alias: Chargemode minpv
    default: []
mode: single
``

flushbug, 2024-03-08:
Die HACS Integration verwendet ich nicht und diese wird auch nicht benötigt. EVCC ist bei mir der Master über den Charger. 
Wenn der Go-Charger bei dir die Seriennummer im Topic enthält musst du es im Script anpassen. Oder vielleicht auch die Seriennummer aus dem Topic rausnehmen falls du z.B. nur einen im Netzwerk hast. Dies kann man in der Go-E App einstellen. Wie das genau aussieht am besten mit MQTT Explorer überprüfen. Wichtig zu wissen : Beim  MQTT schreiben/publish auf ein Topic hat Home Assistant garkeine Ahnung dass es dieses überhaupt gibt. Nur für den Automation Trigger muss das Topic mit dem Buttonpress als Sensor in der HA configuration hinterlegt werden damit HA dieses zyklisch ausliest und auf Änderungen reagieren kann.

Die Loadpoints werden von EVCC durchnummeriert so wie du sie in der evcc.yaml angelegt hast. Wenn du mehrere hast bekommt der erste die 1, der zweite die 2 usw. Und dementsprechend musst du natürlich die richtige ID im Script einfügen. Auch hier am besten mal mit dem MQTT Explorer schauen was wo ist.

domii666, 2024-03-08:
versteh ich, EVCC soll bei mir auch der Master sein. Habe 2 go E, daher benötige ich schon eine Zuordnung. als Prefix steht bei mir in der goE App "go-ECharger/22XXXX/" 

im MQTT Explorer sehe ich das cid oder cch kommt ja von der Automatisierung, das schaut ja schon mal gut aus, nur halt nicht unter der Seriennummer = Skript anpassen.

Loadpoints hab ich verstanden.

flushbug, 2024-03-08:
Korrekt, Lademodus müsste in mqtt bei Änderung über die EVCC gui auch geändert werden.

domii666, 2024-03-08:
so jetzt sollte es gehen, MQTT war noch in evcc ausgeklammert. 

danke dir!

andig, 2024-06-02:
Sehr viele OT Kommentare. Wenn ich es richtig verstehe, dann
- können alle GoE jetzt mqtt
- ist der Buttonstatus über mqtt sofort verfügbar?

Dann müsste eine neue GoE über MQTT Implementierung erstellt werden? Freiwillige vor- seitens Kernteam wollen wir das derzeit nicht angeben. Beispiele gibt es genug, siehe z.B. Warp oder OpenWB.

flushbug, 2024-06-09:
Soweit ich es verstanden habe können alle Go-E ab v3 MQTT d.h. alle die Phasenumschaltung bieten und hier unterscheidet EVCC ohnehin schon. D.h. man müsste die "Charger Gemini, HOME+, HOMEfix (V3)" integration alternativ über MQTT anbieten.

Der Buttonstatus ist über MQTT sofort sichtbar und alle über die reguläre API verfügbaren Keys sind genauso über MQTT les und schreibbar.

Ich sehe mir die anderen MQTT Integrationen mal an aber meine letzten ernsthafen Programmiertätigkeiten sind viele Jahre her. Mit Go selbst habe ich noch nicht gearbeitet. Immerhin kann ich EVCC schon bauen ;-) Möchte aber nichts versprechen.

andig, 2024-06-09:
mce/mcu/mtp verraten MQTT aktiv/ Server/ Topic. Damit wäre schon klar wo auf den Knopf zu lauschen wäre. Siehe Warp oder OpenWB für Beispiele. Modusumschaltung funktioniert dann über `api.LoadpointController`, siehe `updateSmartCharging` bei der Easee.

andig, 2024-06-29:
Wollte das eigentlich jemand versuchen? Sonst könnten wir hier auch langsam zu machen da keine lebenswichtige Funktion.

closed 2024-07-02 by andig

derlangemarkus, 2026-08-19:
Danke für die Inspiration. Ich nutze Home Assistant, EVCC und NodeRed. Sollte noch jemand anderes diese Kombination im Einsatz haben, dann wäre das hier die Kopiervorlage:

[image]

`[{"id":"c165fc28758d25f6","type":"server-state-changed","z":"d8dfed954be252bb","name":"Wallbox-Knopf gedrückt","server":"ad5078d7.503bb8","version":6,"outputs":1,"exposeAsEntityConfig":"","entities":{"entity":["sensor.go_e_buttonpress"],"substring":[],"regex":[]},"outputInitially":false,"stateType":"str","ifState":"","ifStateType":"str","ifStateOperator":"is","outputOnlyOnStateChange":true,"for":"0","forType":"num","forUnits":"minutes","ignorePrevStateNull":false,"ignorePrevStateUnknown":false,"ignorePrevStateUnavailable":false,"ignoreCurrentStateUnknown":false,"ignoreCurrentStateUnavailable":false,"outputProperties":[{"property":"payload","propertyType":"msg","value":"string","valueType":"entityState"},{"property":"data","propertyType":"msg","value":"","valueType":"eventData"},{"property":"topic","propertyType":"msg","value":"","valueType":"triggerId"}],"x":150,"y":80,"wires":[["5d32f881bd09c795"]]},{"id":"5d32f881bd09c795","type":"api-call-service","z":"d8dfed954be252bb","name":"Nächster Ladebodus","server":"ad5078d7.503bb8","version":7,"debugenabled":false,"action":"select.select_next","floorId":[],"areaId":[],"deviceId":[],"entityId":["select.evcc_wallbox_garage_mode"],"labelId":[],"data":"","dataType":"jsonata","mergeContext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","blockInputOverrides":true,"domain":"select","service":"select_next","x":380,"y":80,"wires":[[]]},{"id":"ad5078d7.503bb8","type":"server","name":"Home Assistant","addon":true},{"id":"33990e0fedfb835c","type":"global-config","env":[],"modules":{"node-red-contrib-home-assistant-websocket":"0.80.3"}}]`

flushbug, 2026-08-20:
Schön zu sehen dass meine Idee von vor 2 Jahren noch jemand ausgegraben hat.

Daher eine kleine Rückmeldung zur Langzeiterfahrung mit dem Go-E Button als Trigger für den Lademodus. Ich habe die Automation in Home Assistant, die auf den sich ändernden Button einen Moduswechsel auslöst vor einigen Monaten deaktiviert. 
Teilweise hat der Go-E wohl die Verbindung verloren oder evtl. Einen Reset durchgeführt und somit ist der Zeitstempel aus dem aus dem „ go-eCharger/lbp“ auf einen N/A wert in Home Assistant gewechselt. Das hat dann einen Moduswechsel von bei mir default „PV“ nach „schnell“ ausgelöst und ungewollt ein angeschlossenes Fahrzeug mit Max Power geladen.

Ggfs. werde ich das ganze nochmal genauer analysieren bzw. den Trigger etwas schlauer bauen und z.B. auf valide Zeitstempel filtern.

derlangemarkus, 2026-09-19:
Den versehentliche Modus-Wechsel kann ich bestätigen. Scheint daran zu liegen, dass die Integration neu lädt. Könnte man jetzt z.B. abfangen, indem man den Wechsel nur bei Doppelklick macht. Ich vermute, das dürfte helfen. 

Ich bin jetzt aber dazu übergegangen, mir mit einen d1-Mini und ESP-Home selbst ein kleines Kästchen zu bauen, dass drei Schalter hat, die den Modus anzeigen und umschalten. Und wer die geheime Schalterkombination drückt, kann sogar den Wagen auf- und zuschließen, wenn man mal schnell was raus holen möchte und den Schlüssel nicht dabei hat 😄

[image]