evcc-io/evcc #11683: Feed tariffs via HTTP/MQTT

odlg, 2024-01-14:
Allow feeding tariffs to evcc via HTTP and/or MQTT. 
An example could be to POST a json structure to /api/tariff/grid very similar to one returned on a GET request to /api/tariff/grid - 'result' should be 'feedin' or similar. When using evcc like this, it should be possible to disable evcc fetching tariffs. 
For simplicity, a feedin should just replace current tariff data, it should not be possible to replace the tariffs partly.
Motivated by https://github.com/evcc-io/evcc/issues/10585

andig, 2024-01-14:
Interesting idea. Would you have an example for such a tariff? Wondering how much use there is for a wider user base. Would be happy to look at a PR.

closed 2024-01-23 by andig

ExGarder, 2024-04-21:
It's sad to see that suggestions like this is just ditch'd by You.
I used evcc in the summer and it was realy good.
But in autumn and winter evcc is pretty useless in our area, without propper power prices.
I might come back to evcc when You implement something like suggested.

As a sample there is billigkwh.dk, i am sure there are other outwhere.
Here I can put in my price area, grid area and produkt, to get my specific power price hour for hour.
https://billigkwh.dk/api/Priser/HentPriser?sted=DK1&netselskab=dinel_c&produkt=clever

It should be possible to adjust this output with jq to something that suits evcc.

andig, 2024-04-21:
Sorry, not nothing has been ditched! I just didn't receive a response to the question of how you'd propose to do this. Without any idea there was nothing to do, hence closed. Meanwhile there was a good idea and it was implemented in https://github.com/evcc-io/evcc/pull/13451.

> It should be possible to adjust this output with jq to something that suits evcc.

That's exactly what was finally done.

andig, 2024-04-21:
Btw., in your example I don't see how one would do that with JQ. If you find one happy to take a PR for that specific tariff. Also see https://github.com/evcc-io/evcc/issues/13435.

ExGarder, 2024-04-21:
I don't know anything about JQ, i just saw it mensioned somethere.
If I knew the format i could ask them to add it, or I can make my own wrapper API.

I'm "just" a system developer in the world of Windows and .net, I'm afraid I can't help You with the coding.