evcc-io/evcc #11684: enphase Probleme in der Anzeige/Auswertung

speedy002, 2024-01-14:
### Discussed in https://github.com/evcc-io/evcc/discussions/11681

<div type='discussions-op-text'>

<sup>Originally posted by **speedy002** January 14, 2024</sup>
Hallo Zusammen,

die Anzeige der aktuellen Leistungen meiner enphase werden irgendwie nicht richtig angezeigt.
Gerade wird aus der Batterie ein Teil der notwendigen Leistung entnommen, aber in EVCC steht null.
Auch im Balken ist die Entnahme aus dem Akku nicht angezeigt.

Eigentlich sind die Daten abrufbar. Im Home Assistant sehe ich die Leistung in und aus dem Akku.

[image]
 
[image]

Meine Config ist.

[image]
[image]

</div>

andig, 2024-01-14:
Logfile/Config fehlen. Bitte keine Screenshots.

andig, 2024-01-14:
/cc @premultiply

closed 2024-01-14 by andig

premultiply, 2024-01-14:
https://docs.evcc.io/docs/devices/meters/#enphase-iq-envoy

> Enphase IQ Envoy
> 
> Als Batteriespeicher werden seitens der Enphase-API derzeit nur die Batterien vom Typ "AC Battery" unterstützt.

Das sollte die Frage beantworten, oder?
Ich gehe davon aus dass du keine Batterie vom Typ "AC Battery" hast und diesen Hinweis nicht gelesen oder ignoriert hast.