evcc-io/evcc #11687: Opel Corsa-e offline - Stellantis API - oauth2 fails

hlinden, 2024-01-14:
### Describe the bug

Currently, my Opel Corsa-e is marked as offline. It seems that something around the OAuth process fails on Stellantis' side.

### Steps to reproduce

1. own an Opel Corsa-e
2. configure it in evcc as documented
3. notice the error message on startup

### Configuration details

```yaml
[...]
vehicles:
- type: template
  template: opel 
  title: Corsa  
  user: heresmy@mailadress.tld  
  password: <removed>  
  vin: VXKUHZKXZL<shortened>  
  capacity: 42  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: pv
  minCurrent: 6  
  maxCurrent: 16  
  name: ev1
[...]
```

### Log details

```text
[main  ] INFO 2024/01/14 19:55:31 evcc 0.123.8
[main  ] INFO 2024/01/14 19:55:31 using config file: /config/evcc.yaml
[main  ] INFO 2024/01/14 19:55:31 starting ui and api at :7070
[db    ] INFO 2024/01/14 19:55:32 using sqlite database: /data/evcc.db
[main  ] ERROR 2024/01/14 19:55:32 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'opel': login failed: oauth2: "server_error" "Internal Server Error"
[site  ] WARN 2024/01/14 19:55:36 vehicle 'Corsa (offline)' does not support automatic detection
[...]
[site  ] INFO 2024/01/14 19:55:36   vehicles:
[site  ] INFO 2024/01/14 19:55:36     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/01/14 19:55:36     vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[...]
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.123.8

andig, 2024-01-14:
We cant help Stellantis ;)

closed 2024-01-14 by andig

hlinden, 2024-01-14:
I know, but is it possible that something about their API changed a while ago and EVCC kept using the old whateveritwas until it finally broke now?

flarno11, 2024-01-16:
Are there any more insights why Stellantis doesn't work and how it is different to the Opel template?

atze09, 2024-01-16:
@hlinden @flarno11 is it working for you?
For me it is not working in EVCC, but Flobz Server in Home Assistant and also Opel app is working again.

hlinden, 2024-01-17:
> @hlinden @flarno11 is it working for you? For me it is not working in EVCC, but Flobz Server in Home Assistant and also Opel app is working again.

Same. Opel app works again, EVCC still gives up with...

`[main  ] ERROR 2024/01/17 09:50:48 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'opel': login failed: oauth2: "server_error" "Internal Server Error"`

MMA150475, 2024-01-24:
Same here - Opel Mokka is not recognized, but accessible via the App.

andig, 2024-01-24:
Siehe https://github.com/evcc-io/evcc/issues/11735