evcc-io/evcc #11691: Add meter to myStrom WiFi Switch

naltatis, 2024-01-15:
**Is your feature request related to a problem? Please describe.**

Implement power and maybe energy for myStrom Wifi Switch.

Link to correct HTTP endpoint is already in the source code:
https://github.com/evcc-io/evcc/blob/a2e72e747e3e77396c86ea6bf212522b1dd72928/charger/mystrom.go#L14C4-L15C1

Power should be straight forward. Energy must be pieced together from deltas.

andig, 2024-01-15:
> Energy must be pieced together from deltas.

Don't. Loadpoint will already do this.

andig, 2024-01-15:
@naltatis Power ist im SwitchSocket schon drin! Anders gesagt: Problem ist nicht klar.

naltatis, 2024-01-15:
Ok, du sagst du müssten jetzt schon Power Werte erscheinen? Dann teste ich das mal. Bzw. lasse das testen. #fragFürNenFreund

closed 2024-01-15 by naltatis

andig, 2024-01-15:
Ja. Da ist ja heute schon die standby Power Magie drin.