evcc-io/evcc #11692: Docs: add meter label for chargers

naltatis, 2024-01-15:
**Is your feature request related to a problem? Please describe.**
Would be nice if the chargers overview would also indicate if the evcc implementation supports built-in meter reading. I should become a filter like 1p3p or RFID.
https://docs.evcc.io/docs/devices/chargers

Some chargers come in different variants (with or without meter), but this is something the user has to take care of when buying. From our side, I think it's sufficient to indicate that we can read it if it exists.

**Describe the solution you'd like**
Manually go through all the templates and add a `meter` tag to the devices `capabilities`.

**Additional context**
Helpful for making buying decisions based on supported feature set.

premultiply, 2024-01-15:
Wakeup fehlt noch, aber das weiß man meist einfach nicht.

andig, 2024-04-14:
> Finde ich gut

Habs mir anders überlegt- v.a. weil es diverse Boxen in unterschiedlichen Varianten gibt. Close?

naltatis, 2024-04-15:
> Some chargers come in different variants (with or without meter), but this is something the user has to take care of when buying. From our side, I think it's sufficient to indicate that we can read it if it exists.

Ich glaub weiterhin dass das ne gute Idee ist. Aktuell muss der User in den Code gucken um rauszufinden ob wir das unterstützen. Die Info hilft bei der Kaufentscheidung.

VolkerK62, 2024-07-06:
>Habs mir anders überlegt- v.a. weil es diverse Boxen in unterschiedlichen Varianten gibt. Close?

geht beim label auch mehr als nur ja/nein? Evtl noch ein "variable"

naltatis, 2024-07-06:
Meine Idee war erstmal alle Templates bei denen wir Meter Support haben mit `capabilities: ["meter"]` zu taggen und die Tatsache, ob es immer ein Meter hat oder je nach Modell ein Meter haben kann erstmal auszuklammern. Wenn wir die Info bekommen könnten wir das aber auch gleich mit abbilden. Also anstatt eines `meter` Tags könnte man diese beiden verwenden: `meter-always` und `meter-possible`. In der Doku und Config UI könnten wir das dann je nach Bedarf zusammenfassen oder ggf. nen "hängt vom Modell ab" Hinweis zusätzlich ausspielen.