evcc-io/evcc #11697: Möglichkeit des Speicherns des gesetzten target SOCs pro Vehicle

TheMr-S, 2024-01-15:
In den letzten Releases wurde ja die Einstellung "targetSOC" aus der yaml entfernt. Leider fehlt aus meiner Sicht noch ein richtiger Ersatz dafür:

Ich kann zwar über die UI und den grünen Schieber den target festlegen, aber ich vermisse ein automatisch gesetztes target, wenn ich das Auto verbinde. Aktuell wird immer automatisch 100% ausgewählt - wenn ich das nicht sehe bzw. vergesse, lädt das Auto ggf. ungewollt ganz voll.

Kurzum: Es wäre schon eine wirkliche Verbesserung, wenn der manuell eingestellte targetSOC pro Auto beim nächsten Verbinden des Autos wieder gesetzt wird oder wenn man pro Auto bei Bedarf wieder ein festes target einstellen kann.

Danke!

0x3d13f, 2024-01-15:
Das kannst Du wenn das Auto erkannt oder ausgewählt ist unter Plan -> Ankunft einstellen.

TheMr-S, 2024-01-15:
Danke! Das habe ich bis dato noch nicht gesehen/realisiert.

closed 2024-01-15 by TheMr-S