evcc-io/evcc #11698: Möglichkeit des "kurzzeitigen Anladens" bei Verbindung des Autos im PV-Modus

TheMr-S, 2024-01-15:
Hallo,
wir haben u.a. einen Dacia Spring. Dieser zeichnet sich beim Anschließen an eine deaktivierte Säule dadurch aus, dass er die Säule nicht erkennt, nicht verriegelt und dann einschläft. Über die Cloud kann EVCC dann auch das Auto nicht erkennen, weil das Auto ja nicht weiß, dass es an einer Säule hängt :). Auch WakeUp-Calls über die App etc. gehen dann logischerweise nicht.

Dieses Verhalten hat man also immer dann, wenn man im PV-Modus ohne Überschuss ansteckt. Bei mir fast der Standardfall. Wenn ich daran denke, umgehe ich dies indem ich direkt einmal den Min/PV kurz anschalte, damit kurz geladen wird, und dann wieder in den PV wechsle.

Mein Wunsch wäre nun also die Option (z.B. als Parameter in der yaml für die Loadpoints), dass die Säule im PV-Modus standardmäßig einen Ladevorgang (ggf. mit minimaler Leistung) inittiert und vielleicht nach 30 Sekunden wieder stoppt.

Vielen Dank schon einmal!

mdkeil, 2024-01-15:
Als Workaround einfach in der yaml unter dem vehicle `mode: minpv` setzen, dann würde der dacia  nach Anstecken den Ladevorgang starten. Nur dann musst Du daran denken, manuell wieder auf PV zu stellen.. das könnte man aber relativ einfach über eine eventgesteuerte externe Regelung machen.

Edit: geht ja auch nicht, da er dafür das Auto erstmal erkennen müsste-- ok, du könntest den loadpoint auf `mode: minpv` stellen..

mdkeil, 2024-01-15:
Ich würde mit der Problematik an der Hersteller herantreten, da wohl die Fahrzeugschnittstelle nicht konform arbeitet.. sofern Du bei Dir eine irgendwie geartete Smarthome-Lösung einsetzt, würde ich diesem Fall evcc extern über die API triggern. Ich kann mir nämlich nicht vorstellen, dass dieses "Feature" implementiert wird, da es keinem sinnvollen use-case entspricht.

TheMr-S, 2024-01-15:
Dacia-Support *lächel*.
Ich lasse das hier trotzdem mal drin. Ich weiß zwar nicht, ob / wieviele andere Autos ein ähnlich unglückliches Verhalten haben - aber einen sinnlosen use-case nenne ich es nicht, wenn man einer serienmäßigen Eigenart eines bestimmten Autos angemessen entgegenwirkt. Und das Auto ist halt so gebaut bzw. programmiert, dass es eine aktive Säule beim Anstecken verlangt...

Wie gesagt, diese Funktion kann ja -wenn dann- als standardmäßig deaktivierte Option implementiert werden, falls man so etwas benötigt. Und (man möge mich berichtigen) von der Implementation her erscheint es mir nicht als ein besonders umständliches / kompliziertes Feature.

mdkeil, 2024-01-15:
> Und das Auto ist halt so gebaut bzw. programmiert, dass es eine aktive Säule beim Anstecken verlangt...

evcc deaktiviert die Säule im eigentlichen ja auch nicht.. das Dacia da nicht viel machen wird, das sehe ich aber auch so, denn ich glaube nicht, dass in diesem Preissegment eine leistungsfähige Sw-Abteilung bezahlbar wäre :) --aber vielleicht nimmt sich ein Entwickler dem ja auch an-- Ich würde aber nicht so viel Hoffnung reinstecken.. wie gesagt, das kann man mit überschaubaren Mitteln auch extern über die evcc-API selbst umsetzen.

VolkerK62, 2024-01-15:
@TheMr-S  keine Ahnung wie "fit" du in Sachen IT bist. Ich hatte vor 2 Jahren davon keine Ahnung.
Mittlerweile habe ich mir einige "Nebenregelungen" in NodeRed gebaut um damit meine individuelle Wünsche abbilden zu können.
Wie von @mdkeil schon erwähnt: Das was du möchtest ist mit Hilfe der API kein Hexenwerk.

TheMr-S, 2024-01-15:
> [Verfolgungsbild][https://github.com/notifications/beacon/A3Z2PX6YS7QYUWHL2FFRND3YOUZCRA5CNFSM6AAAAABB3EZS3WWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTQZEUNS.gif]

Moin,
ich zähle mich was das reine selbst -Programmieren angeht eher zu denen, die keine Ahnung haben :)

Aber wenn das tatsächlich nicht als allgemein hilfreiches Feature gesehen wird (sehe ich halt anders), dann kann ich mich da sicher auch selbst reinfuchsen.
Beste Grüße,
Silas Rätzke von Stoyentin

Große Veränderungen fangen klein an - Ressourcen schonen und nicht jede E-Mail drucken
(gesendet von meinem Android-Smartphone mit FairEmail)

*Von: *VolkerK62 ***@***.***>
*An: *evcc-io/evcc ***@***.***>
*Kopie: *TheMr-S ***@***.***>; Mention ***@***.***>
*Datum: *15.01.2024 15:01:26
*Betreff: *Re: [evcc-io/evcc] Möglichkeit des "kurzzeitigen Anladens" bei Verbindung des Autos im PV-Modus (Issue #11698)

VolkerK62, 2024-01-15:
Das beschriebene Verhalten ist sonderbar. Was ist das für eine Wallbox?

TheMr-S, 2024-01-15:
Ein Goe-Charger V2.

VolkerK62, 2024-01-15:
> Dieser zeichnet sich beim Anschließen an eine deaktivierte Säule dadurch aus, dass er die Säule nicht erkennt, nicht verriegelt und dann einschläft.

Was du beschreibst, würde ja bedeuten, dass es an keine Wallbox funktioniert, die mit RFID gesichert und die beim Anstecken noch nicht freigegeben wurde.