evcc-io/evcc #11706: Battery control für Fronius GEN24

jobe39, 2024-01-15:
Battery control (sperren der Entladung bei smart charging) steht ja bereits für einige Wechselrichter zur Verfügung. Für den Fronius Gen24 habe ich hierzu ein interessantes Projekt gefunden:

[https://github.com/muexxl/batcontrol](url)

Hier wird das Fronius Solar API dazu genutzt, die Batterie zeitweise zu sperren (bzw. auch die Batterie zeitweise aus dem Netz zu beladen, aber das ist nochmals ein anderes Thema).

Im File [https://github.com/muexxl/batcontrol/blob/main/fronius/fronius.py](url) ist die Funktionalität in Python implementiert. 
Siehe die Methode set_mode_avoid_discharge:

`def set_mode_avoid_discharge(self):
        timeofuselist= [{'Active': True, 
            'Power': 0, 
            'ScheduleType': 'DISCHARGE_MAX', 
            'TimeTable': {'End': '23:59', 'Start': '00:00'},
            'Weekdays': {'Fri': True, 'Mon': True, 'Sat': True, 'Sun': True, 'Thu': True, 'Tue': True, 'Wed': True}
        }]
        self.set_allow_grid_charging(False)
        return self.set_time_of_use(timeofuselist)`

Bzw. dann in weiterer Folge: set_time_of_use
`    def set_time_of_use(self, timeofuselist):
        config={
            'timeofuse' : timeofuselist
        }
        payload=json.dumps(config)
        response= self.send_request('/config/timeofuse', method='POST',payload=payload,auth=True)
        response_dict=json.loads(response.text)
        expected_write_successes=['timeofuse']
        for expected_write_success in expected_write_successes:
            if not expected_write_success in response_dict['writeSuccess']:
                raise RuntimeError(f'failed to set {expected_write_success}')
        return response`

Das müsste doch auch in evcc umsetzbar sein, oder?

VelinesMO, 2024-01-15:
Das ganze müsste auch direkt per Modbus klappen.
Modbus-Register Tabelle gibt's unter http://www.fronius.com/QR-link/0024 , die relevanten Register scheinen zu sein:

40155: Maximum rate of energy transfer into the storage device.
40157: Maximum rate of energy transfer out of the storage device.

Ich hätte auch möglicherweise einen Gen24 Symo 6kW mit BYD-Speicher im Zugriff, um das ganze zu Testen, kann aber noch nicht versprechen, ob das klappt. Das ist die Anlage bei meinen Eltern, da bin ich nicht regelmäßig.

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/IDKFA/AppData/Local/Packages/oice_16_974fa576_32c1d314_161/AC/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/IDKFA/AppData/Local/Packages/oice_16_974fa576_32c1d314_161/AC/Temp/msohtmlclip1/01/clip_filelist.xml">
<title>Fronius Gen24 Primo/Symo Inverter Register Map with Float Inverter Model
and storage configured</title>
<style>

</style>
</head>

<body link=blue vlink=purple>

40157 | 40157 | 1 | R | 0x03 | MaxDisChaRte | Maximum rate of   energy transfer out of the storage device. | uint16 | W | MaxDisChaRte_SF |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

</body>

</html>

Velines, 2024-01-15:
Das ganze müsste auch direkt per Modbus klappen.
Modbus-Register Tabelle gibt's unter http://www.fronius.com/QR-link/0024 , die relevanten Register scheinen zu sein:

40155: Maximum rate of energy transfer into the storage device.
40157: Maximum rate of energy transfer out of the storage device.

Ich hätte auch möglicherweise einen Gen24 Symo 6kW mit BYD-Speicher im Zugriff, um das ganze zu Testen, kann aber noch nicht versprechen, ob das klappt. Das ist die Anlage bei meinen Eltern, da bin ich nicht regelmäßig.

jobe39, 2024-01-15:
Beim Testen kann ich gerne helfen. Für die Implementierung fehlt mir beruflich schlichtweg die Zeit, um mich da reinzuarbeiten.

andig, 2024-01-15:
> die relevanten Register scheinen zu sein:

Die sind Readonly...