evcc-io/evcc #11713: Vehicles offline due to API changes?

hyperbart, 2024-01-15:
### Describe the bug

All vehicles (bmw and polestar) configured in GUI are marked offline.

### Steps to reproduce

Configure bmw or polestar in GUI, validate, error exclamation mark shows up with notification about EVCC API problem code 400 for vehicle 1, when trying with other vehicle same but number 2.

### Configuration details

```yaml
Ubuntu 22.04 lts
Docker EVCC
```

### Log details

```text
Later
```

### What type of operating system are you running?

Linux

fuxae, 2024-01-15:
Also mein BMW wird ohne Probleme erkannt. An der allgemeinen BMW API dürfte es daher also nicht liegen.

VolkerK62, 2024-01-16:
Just to make sure, you configured the cars not within the yaml-file but by using the GUI configuration?
Might be a bug, because this is still highly experimental.

andig, 2024-01-16:
Missing config

hyperbart, 2024-01-16:
> Just to make sure, you configured the cars not within the yaml-file but by using the GUI configuration? Might be a bug, because this is still highly experimental.

Correct.
And indeed, I saw another thread with @andig about API changes and I think one is related to the other.

hyperbart, 2024-01-16:
> Missing config

Config has been added (was on mobile when creating issue)

hyperbart, 2024-01-16:
Validation of BMW:

[image]

When I go back to the main screen there is a red triangle:

[image]

Notifcations:

[image]

Then I go back to the vehicles and try to validate the Polestar:

[image]

Returning to main screen:

[image]

GrimmiMeloni, 2024-01-16:
Please check the evcc log, or post it here.

andig, 2024-01-16:
Sowohl mein Test- BMW als auch der Polestar funktionieren. Wie immer mal in App/Webseite einloggen, Credentials prüfen und trace log.

closed 2024-01-16 by andig

hyperbart, 2024-01-17:
> Please check the evcc log, or post it here.

@GrimmiMeloni and @andig : I have sent the logs to the emailaddress of EVCC.