evcc-io/evcc #11717: Solis-Hybrid: Batterie Laden und entladen vertauscht

premultiply, 2024-01-16:
### Discussed in https://github.com/evcc-io/evcc/discussions/11716

<div type='discussions-op-text'>

<sup>Originally posted by **dr-nitrox** January 16, 2024</sup>
Hallo zusammen,
vielleicht ist mein Hinweis im alten Post untergegangen. Ich konnte jetzt erfolgreich meinen Solis-Hybrid-Wechselrichter in evcc einbinden. Die Daten kommen alle, lediglich "Batterie Laden und Entladen" sind in evcc vertauscht. Wäre es bitte möglich, dieses von einem der Experten zu korrigieren oder könnte ich das eventuell auch selber machen? Für jeden Hinweis bin ich dankbar.
Vielen Dank und viele Grüße</div>

premultiply, 2024-01-16:
Bitte mal die Ausgabe von `evcc meter` wo man das nachvollziehen kann.

dr-nitrox, 2024-01-16:
@premultiply 

hilf mir bitte. Ist das die evcc.yaml Datei? 
Wenn Sie das nicht ist, wo finde ich die richtige Datei, die du brauchst?

premultiply, 2024-01-16:
Konsole: `evcc meter` eingeben. Enter.
Ausgabe als Code formatiert hier einfügen.
evcc Dienst ggf. vorher stoppen.

dr-nitrox, 2024-01-16:
@premultiply 
Ich hatte das im Ursprungspost noch präzisiert. Das Laden und Entladen der Batterie wird in evcc ohne Vorzeichen unter Batterie Laden gelistet.

premultiply, 2024-01-16:
Bitte `evcc meter`

dr-nitrox, 2024-01-16:
pi@raspberrypi:~ $ sudo systemctl stop evcc
pi@raspberrypi:~ $ evcc meter
[main  ] INFO 2024/01/16 16:11:54 evcc 0.123.8 (60faa202)
[main  ] INFO 2024/01/16 16:11:55 using config file: /etc/evcc.yaml
[db    ] INFO 2024/01/16 16:11:55 using sqlite database: /home/pi/.evcc/evcc.db
grid1
-----
Power:          11W
Energy:         3704.9kWh
Current L1..L3: 0.58A 0.57A 1.04A
Power L1..L3:   34W 91W -114W

pv2
---
Power:  15W
Energy: 4569.0kWh

battery3
--------
Power:  -299W
Energy: 1603.0kWh
Soc:    81%

premultiply, 2024-01-16:
Da wurde entladen, oder?

dr-nitrox, 2024-01-16:
Ja

dr-nitrox, 2024-01-16:
Kannst du mir bitte einen Tipp geben, was ich jetzt tun sollte?

premultiply, 2024-01-16:
Morgen neues Nightly Build / Unstable installieren.

dr-nitrox, 2024-01-16:
Vielen Dank.

dr-nitrox, 2024-01-17:
@premultiply 
Hallo,
nach dem Update steht jetzt der Wert beim Entladen der Batterie auch auf "Batterie entladen", allerdings steht er beim Laden jetzt auch bei "Batterie entladen". Es ist also nur komplett auf die andere Bezeichnung gewechselt. Das Laden bzw. Entladen der Batterie wird dadurch nicht richtig angezeigt. Hier die Daten beim Laden der Batterie.

[main  ] INFO 2024/01/17 11:00:12 evcc 0.123.9
[main  ] INFO 2024/01/17 11:00:12 using config file: /etc/evcc.yaml
[db    ] INFO 2024/01/17 11:00:12 using sqlite database: /home/pi/.evcc/evcc.db
grid1
-----
Power:          5W
Energy:         3715.2kWh
Current L1..L3: 0.58A 0.5A 0.93A
Power L1..L3:   34W 52W -79W

pv2
---
Power:  612W
Energy: 4570.0kWh

battery3
--------
Power:  378W
Energy: 1609.0kWh
Soc:    77%

dr-nitrox, 2024-01-17:
[image]

premultiply, 2024-01-17:
Na super.
Leistungswert ohne Vorzeichen.
Wie ich sowas liebe...

Ich schaue nochmal in der Doku ob ich etwas anderes finde.

dr-nitrox, 2024-01-17:
Ich denke, dass du das Modbus-Protokoll des Solis-Hybrid hast. Wenn nicht, kann ich es dir gerne schicken.

PR #11720 (2024-01-16): Solis-Hybrid: fix battery power sign
Fixes  #11717
PR #11842 (2024-01-25): Solis-Hybrid: Add external battery power sign
Hopefully fixes #11717

Signed battery power is missing the sign.
Ginlong documentation seems to be invalid.
closed 2024-01-25 by andig