evcc-io/evcc #11727: Fehlerhafte Berechnung Ersparnis bei neuem Strompreis

spirit0607, 2024-01-16:
### Describe the bug

Ich habe seit Jahreswechsel einen neuen Stromvertrag abgeschlossen. Neuer Preis 0,26 EUR / kWh (alter Preis 0,36 / kWh)
Ich habe den neuen Preis in der Config Datei entsprechend erfasst. Die Referenzdaten passen auch (siehe Screenshot).
Der Energiepreis stimmt schon (er sinkt jetzt durch den neuen Tarif ab Jahreswechsel). Die errechnete Ersparnis ist aber nicht richtig, bzw. irreführend, da für die komplette Zeitperiode mit dem erst seit 01.01. gültigen Strompreis referenziert wird. Hier müsste auch der bis zum 31.12.2023 gültige Strompreis mit einbezogen werden.

[image]
[image]

### Steps to reproduce

nicht notwendig

### Configuration details

```yaml
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.2679 # [currency]/kWh
```

### Log details

```text
nicht notwendig
```

### What type of operating system are you running?

Linux

andig, 2024-01-16:
@naltatis die Ersparnis bzw. die Preise müssten am Ladevorgang hängen damit das funktionieren kann? Falls das heute nicht der Fall sein sollte evtl. als 1x-Migration?

naltatis, 2024-01-16:
Ja, ist ein Duplikat zu https://github.com/evcc-io/evcc/issues/11160
Wir müssen die Ersparnis (Preis und co2) zukünftig am Ladevorgang berechnen und speichern.