evcc-io/evcc #11728: Nach Neustart von EVCC wird zuvor korrekt erkanntes Fahrzeug nur noch als Gast erkannt

nicx, 2024-01-17:
### Describe the bug

Ich mache einen nächtlichen Neustart meines Docker Containers EVCC. Dabei ist mir aufgefallen dass EVCC das vor dem Neustart angeschlossene und korrekt erkannte Fahrzeug hinterher nur noch als "Gastfahrzeug" erkennt. In meinem Fall handelt es sich um einen Tesla Model Y.

Wenn ich dann in diesem Zustand in der UI wieder den Tesla auswähle, wird er selbst dann nicht mehr erkannt. Ich muss noch zusätzlich das Fahrzeug aufwecken, indem ich die Tesla App auf dem Handy öffne. Dann dauert es eine Weile (10 Minuten?), dann wird in der UI auch der Ladestand wieder korrekt angezeigt.

### Steps to reproduce

1. Anschliessen des Tesla -> Fahrzeug wird richtig erkannt
2. Fahrzeug geht nach x Minuten in den Schlafmodus
3. Neustart des Docker Containers -> Fahrzeug wird nicht mehr erkannt (drehendes Rädchen) und springt dann auf Gastfahrzeug

### Configuration details

```yaml
network:
  schema: http
  host: xxx
  port: 7070

interval: 30s # control cycle interval

sponsortoken: xxx

log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:
  - name: grid
    type: custom
    power:
      source: mqtt
      broker: 192.168.0.1:1883
      topic: ha2evcc/sensor/power_meter_active_power
      scale: -1
      timeout: 30s
    energy:
      source: mqtt
      broker: 192.168.0.1:1883
      topic: ha2evcc/sensor/power_meter_consumption
      scale: 1
      timeout: 30s
  - name: pv
    type: custom
    power:
      source: mqtt
      broker: 192.168.0.1:1883
      topic: ha2evcc/sensor/inverter_input_power
      scale: 1
      timeout: 30s
    energy:
      source: mqtt
      broker: 192.168.0.1:1883
      topic: ha2evcc/sensor/power_meter_consumption
      scale: 1
      timeout: 30s
  - name: battery
    type: custom
    power:
      source: mqtt
      broker: 192.168.0.1:1883
      topic: ha2evcc/sensor/battery_charge_discharge_power
      timeout: 30s
      scale: -1
    soc:
[... 540 lines cut ...]
[site  ] DEBUG 2024/01/17 07:40:43 site power: 1492W
[lp-1  ] DEBUG 2024/01/17 07:40:43 charge currents: [0.01 0.01 0.012]A
[lp-1  ] DEBUG 2024/01/17 07:40:43 charge total import: 4169.327kWh
[lp-1  ] DEBUG 2024/01/17 07:40:43 charger status: B
[lp-1  ] DEBUG 2024/01/17 07:40:43 pv charge current: 0A = 0A + -6.49A (1492W @ 1p)
[site  ] DEBUG 2024/01/17 07:41:13 ----
[lp-1  ] DEBUG 2024/01/17 07:41:13 charge power: 0W
[lp-2  ] DEBUG 2024/01/17 07:41:13 charge power: 0W
[site  ] DEBUG 2024/01/17 07:41:13 pv power: 0W
[site  ] DEBUG 2024/01/17 07:41:13 battery soc: 5%
[site  ] DEBUG 2024/01/17 07:41:13 battery power: 0W
[site  ] DEBUG 2024/01/17 07:41:13 grid power: 1407W
[site  ] DEBUG 2024/01/17 07:41:13 site power: 1507W
[lp-2  ] DEBUG 2024/01/17 07:41:13 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/01/17 07:41:13 charge total import: 215.598kWh
[lp-2  ] DEBUG 2024/01/17 07:41:13 charger status: A
```

### What type of operating system are you running?

Docker container

### Version

0.123.9

andig, 2024-01-17:
Ich hab den Titel mal aussagekräftig geändert. Bitte Zugangsdaten an info@evcc.io.

nicx, 2024-01-17:
@andig der issue ist aber nicht für den Opel gedacht... sondern für die (Wieder)Erkennung des Tesla ;)

andig, 2024-01-17:
Dann brauchts ein passendes Logfile!

nicx, 2024-01-17:
@andig gerne... was genau könnte ich in der config anpassen um mehr Aussagekraft im log zum Thema zu bekommen? 

mit folgender config bekomme ich auch nicht wirklich mehr:

```
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
```

mdkeil, 2024-01-17:
> Dann brauchts ein passendes Logfile! Die evcc Glaskugel ist immer noch zur Reparatur.

Dann solltet du aber auch schreiben, was genau an Logs mehr benötigt wird.. aus dem Log ist z.B. klar ersichtlich, dass der charger alle 30s zwischen status A und B switcht (warum auch immer) und der anschließende vehicle.api refresh fehlschlägt.

```
[lp-1  ] DEBUG 2024/01/17 07:35:43 vehicle api refresh
[main  ] ERROR 2024/01/17 07:35:43 vehicle status: asleep
```

andig, 2024-01-17:
Es geht um einen Tesla Fehler? Dann kann ein trace log für Tesla helfen. Ein Log ohne jede Indikation des Fehlers hilft gar nicht.

naltatis, 2024-01-17:
Meine Vermutung ist, dass das der Effekt von hier ist: https://github.com/evcc-io/evcc/pull/11675 und das mit dem nächsten Release erledigt sein sollte.

nicx, 2024-01-17:
ich habe das Thema nochmal nachgestellt mit folgender angepassten log configuration:

```
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  Tesla_Model_Y: debug

```

@andig damit kommt keine Logzeile mehr ausser die beiden Zeilen von @mdkeil 

wie kann ich mehr zielführende logs generieren?

VolkerK62, 2024-01-17:
@mdkeil 
>aus dem Log ist z.B. klar ersichtlich, dass der charger alle 30s zwischen status A und B switcht (warum auch immer)

Achtung, es sind 2 Loadpoints ;-)

VolkerK62, 2024-01-17:
@nicx 
>  Tesla_Model_Y: debug

versuch mal `tesla: trace`

nicx, 2024-01-17:
@VolkerK62 ok, damit funktioniert es. da habe ich die doku evtl. missverstanden, dort ist nicht "tesla" als config option erwähnt, sondern "Fahrzeugname" :)

@andig hier nun das gewünschte trace log:

```
[main  ] INFO 2024/01/17 14:08:38 evcc 0.123.9
[main  ] INFO 2024/01/17 14:08:38 using config file: /etc/evcc.yaml
[main  ] INFO 2024/01/17 14:08:38 starting ui and api at :7070
[mqtt  ] INFO 2024/01/17 14:08:38 connecting evcc-1648033164 at tcp://192.168.0.1:1883
[tesla ] TRACE 2024/01/17 14:08:39 POST https://auth.tesla.com/oauth2/v3/token
[main  ] ERROR 2024/01/17 14:08:39 creating vehicle Opel_Mokka failed: cannot create vehicle type 'template': cannot create vehicle type 'opel': login failed: oauth2: "server_error" "Internal Server Error"
[tesla ] TRACE 2024/01/17 14:08:39 client_id=ownerapi&grant_type=refresh_token&refresh_token=***
--
{"access_token":"11111","expires_in":28800,"token_type":"Bearer"}
[tesla ] TRACE 2024/01/17 14:08:39 GET https://owner-api.teslamotors.com/api/1/vehicles
[tesla ] TRACE 2024/01/17 14:08:40 {"response":[{"id":1111,"vehicle_id":1111,"vin":"1111","color":null,"access_type":"OWNER","display_name":"Model Y","option_codes":null,"granular_access":{"hide_private":false},"tokens":["c101cc913de87b5d","b80b3323d438d11c"],"state":"asleep","in_service":false,"id_s":"11111","calendar_enabled":true,"api_version":71,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}],"count":1}
[lp-1  ] DEBUG 2024/01/17 14:08:40 charge total import: 4169.416kWh
[lp-2  ] DEBUG 2024/01/17 14:08:40 charge total import: 215.598kWh
[site  ] WARN 2024/01/17 14:08:40 vehicle 'Opel Mokka (offline)' does not support automatic detection
[site  ] INFO 2024/01/17 14:08:40 site config:
[site  ] INFO 2024/01/17 14:08:40   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/01/17 14:08:40     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/01/17 14:08:40     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/01/17 14:08:40     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/01/17 14:08:40   vehicles:
[site  ] INFO 2024/01/17 14:08:40     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/01/17 14:08:40     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/01/17 14:08:40 loadpoint 1:
[lp-1  ] INFO 2024/01/17 14:08:40   mode:        pv
[lp-1  ] INFO 2024/01/17 14:08:40   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/01/17 14:08:40   meters:      charge ✓
[lp-1  ] INFO 2024/01/17 14:08:40     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2024/01/17 14:08:40 loadpoint 2:
[lp-2  ] INFO 2024/01/17 14:08:40   mode:        pv
[lp-2  ] INFO 2024/01/17 14:08:40   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-2  ] INFO 2024/01/17 14:08:40   meters:      charge ✓
[lp-2  ] INFO 2024/01/17 14:08:40     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2024/01/17 14:08:40 set buffer soc: 100
[site  ] DEBUG 2024/01/17 14:08:40 set smart cost limit: 0.2
[site  ] DEBUG 2024/01/17 14:08:40 set priority soc: 100
[lp-1  ] DEBUG 2024/01/17 14:08:40 phase timer inactive
[lp-1  ] DEBUG 2024/01/17 14:08:40 pv timer inactive
[lp-1  ] DEBUG 2024/01/17 14:08:40 guard timer inactive
[lp-2  ] DEBUG 2024/01/17 14:08:40 phase timer inactive
[lp-2  ] DEBUG 2024/01/17 14:08:40 pv timer inactive
[lp-2  ] DEBUG 2024/01/17 14:08:40 guard timer inactive
[site  ] DEBUG 2024/01/17 14:08:40 ----
[lp-1  ] DEBUG 2024/01/17 14:08:40 charge power: 0W
[lp-2  ] DEBUG 2024/01/17 14:08:40 charge power: 0W
[site  ] DEBUG 2024/01/17 14:08:40 pv power: 1569W
[site  ] DEBUG 2024/01/17 14:08:40 battery soc: 7%
[site  ] DEBUG 2024/01/17 14:08:40 battery power: -688W
[site  ] DEBUG 2024/01/17 14:08:40 grid power: -0W
[site  ] DEBUG 2024/01/17 14:08:40 battery has priority at soc 7% (< 100%)
[site  ] DEBUG 2024/01/17 14:08:40 site power: 100W
[lp-1  ] DEBUG 2024/01/17 14:08:40 charge currents: [0.012 0.011 0.008]A
[lp-1  ] DEBUG 2024/01/17 14:08:40 charge total import: 4169.416kWh
[lp-1  ] DEBUG 2024/01/17 14:08:40 charger status: B
[lp-1  ] INFO 2024/01/17 14:08:40 car connected
[lp-1  ] DEBUG 2024/01/17 14:08:40 vehicle api refresh
[lp-1  ] DEBUG 2024/01/17 14:08:40 pv timer elapse
[lp-1  ] DEBUG 2024/01/17 14:08:40 pv timer inactive
[lp-1  ] DEBUG 2024/01/17 14:08:40 charger: guard elapse
[lp-1  ] DEBUG 2024/01/17 14:08:40 guard timer inactive
[lp-1  ] DEBUG 2024/01/17 14:08:40 charge total import: 4169.416kWh
[tesla ] TRACE 2024/01/17 14:08:40 GET https://owner-api.teslamotors.com/api/1/vehicles/11111/vehicle_data
[tesla ] TRACE 2024/01/17 14:08:40 {"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}
[main  ] ERROR 2024/01/17 14:08:40 vehicle status: asleep
[lp-1  ] DEBUG 2024/01/17 14:08:40 available power -100W < 4140W min 3p threshold
[lp-1  ] DEBUG 2024/01/17 14:08:40 phase scale1p in 0s
[... 15 lines cut ...]
[lp-2  ] DEBUG 2024/01/17 14:09:10 charge total import: 215.598kWh
[lp-2  ] DEBUG 2024/01/17 14:09:10 charger status: A
[lp-2  ] INFO 2024/01/17 14:09:10 car disconnected
[lp-2  ] DEBUG 2024/01/17 14:09:10 set session soc limit: 0
[lp-2  ] DEBUG 2024/01/17 14:09:10 set session energy limit: 0
[site  ] DEBUG 2024/01/17 14:09:40 ----
[lp-1  ] DEBUG 2024/01/17 14:09:40 charge power: 0W
[lp-2  ] DEBUG 2024/01/17 14:09:40 charge power: 0W
[site  ] DEBUG 2024/01/17 14:09:40 pv power: 1531W
[... cut ...]
[tesla ] TRACE 2024/01/17 14:09:40 GET https://owner-api.teslamotors.com/api/1/vehicles/1111/vehicle_data
[tesla ] TRACE 2024/01/17 14:09:40 {"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}
[main  ] ERROR 2024/01/17 14:09:40 vehicle status: asleep
[lp-1  ] DEBUG 2024/01/17 14:09:40 pv charge current: 0A = 0A + -0.426A (98W @ 1p)
```

sollte EVCC denn das Fahrzeug überhaupt grundsätzlich aufwecken? Oder ist das Verhalten evtl. so wie es ist erwünscht?

naltatis, 2024-01-17:
@nicx magst du mal auf Nightly wechseln und schauen ob der Fehler dort auch noch vorhanden ist?

VolkerK62, 2024-01-17:
das letzte Nightly ist 12 Std alt .... der Release nur 8 Std.

andig, 2024-01-17:
Bitte aktuelles Release testen. Kein nightly.

nicx, 2024-01-17:
@VolkerK62 ah verstehe... danke.
dann meinte @naltatis vielleicht irgend einen dev stand... nun gut. bis dahin kann als Antwort geben: mit der aktuellen nightly ist das problem noch vorhanden :)

GrimmiMeloni, 2024-01-20:
Ich kann hier erstmal noch keinen Fehler erkennen. Im zuletzt gezeigten Log ist kein PV Überschuss verfügbar, folglich wird auch nicht geregelt, und deshalb auch das Fahrzeug nicht aufgeweckt.
@nicx hast Du schon Fast Mode ausprobiert? Siehst Du das Problem dann auch? Wenn ja, bitte davon nochmal ein Log. Danke.

andig, 2024-01-20:
Ich mache mal zu bis es ein aussagekräftiges Logfile gibt das den Fehler überhaupt zeigt.

closed 2024-01-20 by andig