evcc-io/evcc #11730: Support configurable minimum target SOC in battery control

andreas-bulling, 2024-01-17:
The new battery control feature is pretty nice (prevents discharge in fast mode and planned charging). Works like a charm.

But one thing that I find rather annoying is that, by default, it resets the target SOC to 20% after charging is finished.

This is not only hard-coded and cannot be changed by the user afaik but it is also rather high. Many people use 10% as a safety margin below which they don't discharge the battery.

Would be nice if this minimum SOC was configurable

mdkeil, 2024-01-17:
For victron Inverters, you can set the minsoc and maxsoc via evcc.yaml. It is not documented at the moment.

https://github.com/evcc-io/evcc/blob/ffc3ccdc23c9f31c58f9ebf7c509fd43e2215e16/templates/definition/meter/victron-energy.yaml#L16-L22

mdkeil, 2024-01-17:
`I assume this goes into "meters" and there the entry that has "usage: battery"?`

yes!

dbhorst, 2024-01-17:
Same problem here with kostal-plenticore. I've already set set the minsoc and maxsoc values, but the battery minsoc gets set to 5%. Is there a problem with my meters config?

```
meters:
- type: template
  template: kostal-plenticore
  id: 71
  host: 10.11.11.21
  port: 1502
  usage: pv
  modbus: tcpip
  name: pv1
- type: template
  template: kostal-plenticore
  id: 71
  host: 10.11.11.21
  port: 1502
  usage: battery
  modbus: tcpip
  endianness: little
  minsoc: 15
  maxsoc: 100
  name: battery2
- type: template
  template: kostal-ksem-inverter
  id: 71
  host: 10.11.11.21
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid3
```

andig, 2024-01-18:
> I assume this goes 

Use 

    evcc configure