evcc-io/evcc #11735: Internal Server Error Opel / PSA

bbr111, 2024-01-17:
### Discussed in https://github.com/evcc-io/evcc/discussions/11589

<div type='discussions-op-text'>

<sup>Originally posted by **bbr111** January 10, 2024</sup>
Hallo zusammen,

hat noch jemand Probleme mit PSA / Opel?

[main  ] ERROR 2024/01/10 11:40:20 creating vehicle corsae failed: cannot create vehicle type 'template': cannot create vehicle type 'opel': login failed: oauth2: "server_error" "Internal Server Error"

Benutze auch noch den psa_car_controller und der funktioniert. die API scheint also zu funktionieren.</div>

nuiler, 2024-01-17:
Das Problem haben mehrere 

https://github.com/evcc-io/evcc/discussions/11589

agt189, 2024-01-17:
Ich habe auch das gleiche Problem...

chr-engwer, 2024-01-17:
The API (see https://developer.groupe-psa.io/webapi/b2c/api-reference/specification) has changed in november:

https://developer.groupe-psa.io/webapi/b2c/api-reference/changelog/#article

so this would fit to updating the app...
 1. update the API
 2. update the app in downwards compatible way
 3. update server

andig, 2024-01-18:
Depends on https://github.com/flobz/psa_car_controller/issues/733

andig, 2024-01-19:
Bitte Zugangsdaten für Test an info@evcc.io

bbr111, 2024-01-19:
> Bitte Zugangsdaten für Test an [info@evcc.io](mailto:info@evcc.io)

Kommt gerade durch.

Ich glaube das man noch mehr machen muss als nur den authorize.

die API wurde von api.groupe-psa auf api-key.groupe-psa und die neue URL benötigt ein Zertifikat.
Dieses ist auch in der neuen APK hinterlegt im assest ordner. Der Key dazu findet sich ebenfalls in der APK.

ritschi86, 2024-01-23:
gibt es hier schon etwas Neues?

stereotone, 2024-01-23:
Werden noch Zugangsdaten benötigt?

andig, 2024-01-23:
Im Moment muss jemand rausfinden wie die neue API aussieht.

chr-engwer, 2024-01-23:
Wie schon angemerkt wurde, wird das Problem auch in flobz/psa_car_controller#733 behandelt. Da gibt es zwar Fortschritte, aber die Änderungen am Protokoll sind nicht ganz so simple umzusetzen. Wer helfen möchte, sollte dort am besten schauen.

torstenmichl, 2024-01-24:
Hallo,
anscheinend gleiches Problem bei Peugeot nach Neustart durch das Update:
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/01/24 17:30:47 evcc 0.123.9
[main  ] INFO 2024/01/24 17:30:47 using config file: /config/evcc.yaml
[main  ] ERROR 2024/01/24 17:30:48 creating vehicle 308 failed: cannot create vehicle type 'template': cannot create vehicle type 'peugeot': login failed: oauth2: "server_error" "Internal Server Error"
[main  ] ERROR 2024/01/24 17:30:48 creating vehicle 2008 failed: cannot create vehicle type 'template': cannot create vehicle type 'peugeot': login failed: oauth2: "server_error" "Internal Server Error"

APP funktioniert. 
Mehr kann ich leider nicht beisteuern, da keine Ahnung vom Programmieren.

ThiloBaWue, 2024-01-28:
Keine Lösung in Sicht?

Sillium007, 2024-01-28:
Diskussion läuft in dem flobz issue. Solange es dort denke ich keine Lösung gibt passiert hier nichts

torstenmichl, 2024-01-31:
Was soll ich antworten? Hatte angefragt, warum die Api nebst Authentifizierung geändert wurde.

Sehr geehrter Herr Michl,

vielen Dank für Ihre Mitteilung.

Wir bedauern die Unannehmlichkeiten, die durch die Änderungen im Zugriff auf die API entstanden sind, und entschuldigen uns für etwaige Unannehmlichkeiten.

Um Ihr Anliegen bestmöglich bearbeiten zu können, benötigen wir weitere Informationen. Könnten Sie uns bitte detaillierter erläutern, wofür Sie die neue API-Schnittstellenbeschreibung inklusive Authentifizierungsmechanismus benötigen? 

Diese zusätzlichen Informationen helfen uns, Ihr Anliegen besser zu verstehen und Ihnen die benötigten Informationen gezielter zukommen zu lassen. Wir sind daran interessiert, Ihnen so schnell wie möglich zu helfen und die Probleme zu lösen.

Vielen Dank im Voraus für Ihre Kooperation. Bei weiteren Fragen stehen wir Ihnen gerne zur Verfügung.

Mit freundlichen Grüßen,

Stanislava Bako 

Peugeot Deutschland Kundenbetreuung

Telefon 0614 2293 5000

bbr111, 2024-01-31:
Die detaillierte Beschreibung ist die GitHub Seite von evcc / flopz / ...

Wir wollen einfach nur wieder mit 3rd Party Tools auf die Daten der Schnittstelle zugreifen.
Mir mich ein Kaufkriterium. Muss meinen Opel im Oktober zurückgeben. Mal schauen ob ich umsteigen muss.

ThiloBaWue, 2024-01-31:
Es ist ja nicht so, als könnten ThirdPartyApps nicht auf die Daten zugreifen. Ich bin sicher tronity kann nicht zaubern, und ich habe keine Ausfallzeit wahrnehmen können. Die hatten sich wohl vorbereitet. Was aber kein Vorwurf ab die Programmierer hier sein soll. Wahrscheinlich haben Sie es einfach nicht kommen sehen. Ihr schafft das! Aber eine Rückmeldung über den Fortschritt wäre für die zahlenden User schon nett.

Sillium007, 2024-02-01:
flobz hat bei Stellantis einen Issue aufgemacht, wenn ihr das unterstützen wollt bitte hier entlang: https://github.com/Stellantis/stellantis.github.io/issues/128

andig, 2024-02-03:
Normale oauth Anmeldung. Leider steht die uns nicht zur Verfügung.

andig, 2024-02-03:
Hier gehts um PSA, nicht um Tronity. Wer Tronity verwenden möchte kann das bereits jetzt schon. Das PSA API hat das gleiche Problem das wir mit Tesla auch bald haben werden: es ist kostenpflichtig.

> Wäre es nicht denkbar auf PSA in ähnlicher Form wie auf Tronity zuzugreifen, wenn man sich jeder individuell vorher bei PSA die entsprechenden OAUTH2 Credentials holt? 

Wenn Du es schaffst, die zu bekommen können wir das gerne ausprobieren.

Bubbler-der-Bub, 2024-02-04:
Rein aus Neugier: Wenn sich Opel / Stellantis so ziert; wo ist denn die ursprüngliche  API-Schnittstellenbeschreibung inklusive des Authentifizierungsmechanismus hergekommen?
Und für alle, die kurz vorm Wechsel oder einer Neuanschaffung stehen: Gibt es eine Liste von kooperativen Herstellern?

iphikles3000, 2024-02-06:
Hallo zusammen, ich habe mir wohl oder übel auch nen kostenpflichtigen Tronity Account zugelegt. Die Einrichtung läuft mühelos. In der Config habe ich lediglich die ID und die secret Id hinterlegt.

So sieht meine config diesbezüglich aus:
- name: ev3
  title: Peugeot Tronity
  icon: car
  type: template
  template: tronity
  clientid: 5117e23c-0c49-4410-b698-XXXxxxxxxxXXX # Einrichtung unter https://app.tronity.tech
  clientsecret: 671e376e-388e-4b6d-b664-xXXxxXXXXXxxx # Einrichtung unter https://app.tronity.tech
  capacity: 45 # AkkukapazitÃ¤t in kWh (optional)

Damit funktioniert die SOC-Abfrage in evcc wieder ohne Probleme. Natürlich hoffe ich weiter auf einen anderen Lösungsweg für Opel/Peugeot...

Sillium007, 2024-02-08:
Bei flobz haben sie es wohl grundlegend wieder zum Laufen bekommen, aber noch nicht ganz automatisierbar wohl:
https://github.com/flobz/psa_car_controller/issues/733#issuecomment-1931515174

kon1992, 2024-02-10:
Workaround über PSA Car controller funktioniert: PSA Car controller installieren in der Version von 0x3dlux (link oben) und anschlissend im EVCC config file das Fahrzeug auf das psa_controller template umstellen. Dann wir die Reichweite und Ladestand wieder angezeigt.

ostrocka, 2024-02-10:
Ich musste gerade mein Passwort zurücksetzen, das wurde mir angezeigt:

"Sehr geehrte Kundin, sehr geehrter Kunde, aufgrund technischer Änderungen in unseren Systemen zur Verbesserung der Sicherheit aller Konten, müssen Sie Ihr Passwort aktualisieren!
Eine E-Mail zum Zurücksetzen des Passworts wurde an Sie gesendet an:"

Mein altes Passwort war >16 Zeichen und hatte alles drin was man sich so wünscht.

Das ging nun nicht mehr da >16 Zeichen als "Weak Password" bezeichnet wird -- sobald man auf 16 Zeichen runterkürzt ist's dann "very strong".

Das ist ein Saftladen....

phen0, 2024-02-13:
Also ich habe den Workaround mit dem psa_car_controller umgesetzt. Das funktioniert und ich kann in evcc wieder den Ladestand sehen!
[image]

docsnyder7, 2024-02-13:
@phen0 was heisst workaround? Du hast zusätzlich den selbstständigen Psa Controler installiert und greifst von dem die Daten ab für EVCC

Sillium007, 2024-02-13:
Genau, das funktioniert derzeit https://github.com/evcc-io/evcc/issues/11735#issuecomment-1936929156
Evcc intern gibt es noch keine Änderungen

phen0, 2024-02-13:
@docsnyder7 ja genau. Ich habe zusätzlich den psa controller installiert, damit es überhaupt mit dem Ladestand funktioniert. Ist nicht schön, aber funktioniert. Wer Hilfe dabei braucht, kann sich gerne melden.

docsnyder7, 2024-02-13:
OK.. ich hab keine Oberfläche nur die Commandzeile.. wie komm ich ins Dev Tool? auf nem normalen PI4 kein docker oder ähnliches

Sillium007, 2024-02-13:
psa controller starten und vom Desktop aus mit dem Browser drauf verbinden etc.

Und auch in dem Browser nutzt Du die DevTools, das wird nicht auf dem Pi gemacht

docsnyder7, 2024-02-13:
```
 WARNING: The script psa-car-controller is installed in '/home/docsn/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.

```
na der install bekomm diese warnung und der versuch des starts der web conf geht mit "Komando nicht gefunden" ins leere

phen0, 2024-02-13:
Versuche mal dich von der Shell aus- und wieder einzuloggen. Mit echo $PATH kannst du sehen, ob das Verzeichnis enthalten ist. Falls nicht musst du den PATH in der .bashrc oder .profile ergänzen.

PR #13612 (2024-05-01): PSA: change authentication from user/password to token
Fixes #11735
Refs https://github.com/evcc-io/evcc/discussions/11589#discussioncomment-9239565

Die Umsetzung könnte sicherlich noch verbessert werden, aber zumindest lässt sich die direkte PSA Integration so schonmal wieder nutzen.