evcc-io/evcc #11754: Polestar: api changed

andig, 2024-01-18:
### Discussed in https://github.com/evcc-io/evcc/discussions/11751

<div type='discussions-op-text'>

<sup>Originally posted by **Demophobie** January 18, 2024</sup>
Moin,

seit heute zeigt EVCC mir kein SoC mehr an, ich glaube der Km-Stand ist auch betroffen. Habe nichts geändert, lief vorher problemlos.

`Wallbox Carport: vehicle soc: Message: 400 Bad Request; body: \"{\\\"errors\\\":[{\\\"message\\\":\\\"Cannot query field \\\\\\\"getBatteryData\\\\\\\" on type \\\\\\\"Query\\\\\\\".\\\",\\\"extensions\\\":{\\\"code\\\":\\\"GRAPHQL_VALIDATION_FAILED\\\",\\\"exception\\\":{\\\"stacktrace\\\":[\\\"GraphQLError: Cannot query field \\\\\\\"getBatteryData\\\\\\\" on type \\\\\\\"Query\\\\\\\".\\\",\\\" at Object.Field (/var/task/src/functions/gateway/handler.js:79:15333)\\\",\\\" at Object.enter (/var/task/src/functions/gateway/handler.js:48:25080)\\\",\\\" at Object.enter (/var/task/src/functions/gateway/handler.js:79:12452)\\\",\\\" at Pdr (/var/task/src/functions/gateway/handler.js:48:24254)\\\",\\\" at fgr (/var/task/src/functions/gateway/handler.js:79:60357)\\\",\\\" at f (/var/task/src/functions/gateway/handler.js:287:40742)\\\",\\\" at YRr (/var/task/src/functions/gateway/handler.js:287:39031)\\\",\\\" at runMicrotasks (<anonymous>)\\\",\\\" at processTicksAndRejections (node:internal/process/task_queues:96:5)\\\",\\\" at async dBe (/var/task/src/functions/gateway/handler.js:288:2270)\\\"]}}}]}\\n\", Locations: [], Extensions: map[code:request_error]`

in der evcc.yaml sieht es wie folgt aus:
```
vehicles:
  - name: polestar                      # Fahrzeug Polestar 2
    type: template
    template: polestar
    title: Polestar 2
    user: BLABLABLA@googlemail.com
    password: YOYOYOYOYOYO
    vin: LPSVSXXXXXXXXXX
    capacity: 75
```
</div>

andig, 2024-01-19:
Könnt ihr bitte mal probieren, ob https://github.com/leeyuentuen/polestar_api auf HA für Euch funktioniert?

Claudweb, 2024-01-19:
> Könnt ihr bitte mal probieren, ob https://github.com/leeyuentuen/polestar_api auf HA für Euch funktioniert?

Ja, das funktioniert mit der 1.6.0.

andig, 2024-01-19:
@Claudweb darf ich um Zugangsdaten für info@evcc.io bitten? Ich habe immer noch das Gefühl, dass das von Fzg zu Fzg unterschiedlich ist.

loebse, 2024-01-19:
> Könnt ihr bitte mal probieren, ob https://github.com/leeyuentuen/polestar_api auf HA für Euch funktioniert?

Funktioniert bei mir auch mit der 1.6.0 wieder zuverlässig.

Demophobie, 2024-01-19:
@andig Zugangsdaten kann ich wieder ändern?

andig, 2024-01-19:
Ja. Leider besteht das Problem weiter. Das nüsste funktionieren, tut es aber nicht und ich verstehe nicht, warum 😢

andig, 2024-01-19:
@Demophobie das Repo oben funktioniert bei Dir?

Demophobie, 2024-01-19:
@andig Achso, nein, ich dachte nur du hättest es hinbekommen, weil du was geändert hast. Hab's noch nicht probiert.

Kann mir meinem Wissen nur wenig beitragen 😢

andig, 2024-01-19:
@loebse hättest du Zugangsdaten für info@evcc.io?

Demophobie, 2024-01-19:
Den Gedanken mit der Zwei Faktor Authentifizierung hatte ich auch schon. Habe jetzt mal die Anweisungen bzgl. dieser Authentifizierung befolgt und meine Handynummer im Profil ergänzt. Die war nämlich gar nicht drin...

Irgendwas muss ja anders sein.... Hm...

loebse, 2024-01-20:
> @loebse hättest du Zugangsdaten für info@evcc.io?

Vielleicht eine blöde Frage, aber was für Zugangsdaten hättest du gerne?

loebse, 2024-01-20:
> Den Gedanken mit der Zwei Faktor Authentifizierung hatte ich auch schon. Habe jetzt mal die Anweisungen bzgl. dieser Authentifizierung befolgt und meine Handynummer im Profil ergänzt. Die war nämlich gar nicht drin...
> 
> 
> 
> Irgendwas muss ja anders sein.... Hm...

Mit der 2FA hat es nichts zu tun. Aber Polestar hat den API Endpunkt von `/my-star` zu `/mystar-v2` geändert.

andig, 2024-01-20:
Das ist im PR schon umgesetzt, funktioniert aber trotzdem nicht. Unklar ist warum. Wenn bei jemandem https://github.com/leeyuentuen/polestar_api geht, der PR aber nicht dann muss es irgendeinen Unterschied geben. Nur welchen? Lässt sich nur mit Zugangsdaten testen.

sgr002, 2024-01-20:
Mein API-Login geht, ich teste das gerne, ist die Anpassung schon im normalen Release und kann mit 

sudo apt --only-upgrade install -y evcc

installiert werden?

Demophobie, 2024-01-20:
Nein, ist sie leider noch nicht.

sgr002, 2024-01-20:
Ich habe Zeit, kein Stress

sgr002, 2024-01-20:
Habt ihr jetzt schon API-Zugangsdaten bekommen? Wenn nicht kann ich Euch die gegeben und nach Abschluss das Passwort wieder ändern, kein Problem.

MASHpony, 2024-01-21:
> @loebse hättest du Zugangsdaten für [info@evcc.io](mailto:info@evcc.io)?

Hi,

hast du jetzt schon die Zugangsdaten? Sonst sag Bescheid ich schick dir meine wenn du sie brauchst

Demophobie, 2024-01-21:
Ich denke @andig kann gerade alle Zugangsdaten gut gebrauchen um den Fehler zu finden. Schick sie ihm ruhig. Meine hat er auch.

leeyuentuen, 2024-01-21:
> Just wanted to join in - I have the same problem
> 
> Could it have something to do with Polestar introducing two-factor authentication?
> 
> `Domi: vehicle soc: Message: 400 Bad Request; body: \"{\\\"errors\\\":[{\\\"message\\\":\\\"Cannot query field \\\\\\\"getBatteryData\\\\\\\" on type \\\\\\\"Query\\\\\\\".\\\",\\\"extensions\\\":{\\\"code\\\":\\\"GRAPHQL_VALIDATION_FAILED\\\",\\\"exception\\\":{\\\"stacktrace\\\":[\\\"GraphQLError: Cannot query field \\\\\\\"getBatteryData\\\\\\\" on type \\\\\\\"Query\\\\\\\".\\\",\\\" at Object.Field (/var/task/src/functions/gateway/handler.js:79:15333)\\\",\\\" at Object.enter (/var/task/src/functions/gateway/handler.js:48:25080)\\\",\\\" at Object.enter (/var/task/src/functions/gateway/handler.js:79:12452)\\\",\\\" at Pdr (/var/task/src/functions/gateway/handler.js:48:24254)\\\",\\\" at fgr (/var/task/src/functions/gateway/handler.js:79:60357)\\\",\\\" at f (/var/task/src/functions/gateway/handler.js:287:40742)\\\",\\\" at YRr (/var/task/src/functions/gateway/handler.js:287:39031)\\\",\\\" at runMicrotasks (<anonymous>)\\\",\\\" at processTicksAndRejections (node:internal/process/task_queues:96:5)\\\",\\\" at async dBe (/var/task/src/functions/gateway/handler.js:288:2270)\\\"]}}}]}\\n\", Locations: [], Extensions: map[code:request_error]`

this error is more that the graphql query is not correctly

andig, 2024-01-22:
Polestar has a new api- `mystar-v2`. Old api only returns HTTP 400. Apparently the old authentication isn't valid for the new api. Switching to new authentication (similar to web ui) returns "BAD USER INPUT" when calling the API. If anyone if proficient with graphQL please take a look at the PR. I'm out of ideas what might be wrong with the fix.

andig, 2024-01-22:
Mir fehlen immer noch Zugangsdaten für einen Fall wo https://github.com/evcc-io/evcc/issues/11754#issuecomment-1901960249 definitiv funktioniert.

chfaurie, 2024-01-24:
Im Homematic funktioniert Polestar wieder nur evcc nicht.

Demophobie, 2024-01-29:
Ich glaube @andig sucht Zugangsdaten, bei denen die verlinkte Home Assistand App funktioniert.

Ich hoffe das Problem wird gelöst, man merkt erst wie verwöhnt man war wenn auf einmal die SoC nicht mehr zur Verfügung steht... ;-)

chfaurie, 2024-01-29:
Ich kann funktionierende Zugangsdaten zuverfügung stellen. @andig ich muss nur wissen ob das zielführend ist, dann muss ich noch wissen wohin mit den Daten.

hyperbart, 2024-01-31:
> Polestar has a new api- `mystar-v2`. Old api only returns HTTP 400. Apparently the old authentication isn't valid for the new api. Switching to new authentication (similar to web ui) returns "BAD [...]

@andig I think it has to do (underlying) with the introduction or alert we got a while back about Polestar transforming to strong authentication.

https://www.polestar.com/en-kw/polestar-support/polestar-id/

-->

> Two-factor authentication adds an extra layer of security to your account. In addition to your Polestar ID username and password, you’ll need to enter a code (sent to you via email or SMS) to log in.
> 
> Two-factor authentication will be launched in January 2024. To prepare, please go to [polestarid.eu.polestar.com/Account/landing-page](https://polestarid.eu.polestar.com/Account/landing-page) and make sure that you:
> 
> • can log in to your account.
> 
> • have a valid mobile number or email address linked to your account.
> 
> Note: Your Polestar ID username is the email address or mobile number that will receive the code when two-factor authentication launches.

Following this because also experiencing this problem.

andig, 2024-02-09:
Leider kein Fortschritt. Meine Implementierung sieht richtig aus, bekommt aber keine Tokens. Jetzt bräuchte es ein anderes Tool, dessen HTTP Verkehr ich 1:1 vergleichen könnte. Leider kann man in HA den HTTP Traffic nicht mit schneiden.

andig, 2024-02-09:
Gute Idee! Leider kann ich Node nicht dazu überreden über Proxyman zu gehen:

    HTTPS_PROXY=localhost:9000 NODE_TLS_REJECT_UNAUTHORIZED=0 node polestar.js

Falls da jemand eine Idee hat...

loebse, 2024-02-09:
@andig ich habe jetzt ein einfaches Python Beispiel, welches den Auth Token erfolgreich ausliest:
https://github.com/loebse/polestar_python_example/blob/main/polestarAuth.py

Im Endeffekt sind es 3 Calls:

1. Get resume path (wichtig ist, dem redirect NICHT zu folgen)
2. Get Code
3. Get Token

Alle drei Requests müssen natürlich in der gleichen Session abgesetzt werden.

Hilft das weiter?

andig, 2024-02-10:
Schaffst Du es, das mal durch Proxyman zu jagen?

loebse, 2024-02-10:
Ja, kann ich machen:

[image]

Was brauchst du genau @andig ?

andig, 2024-02-10:
Den kompletten Durchlauf bitte an info@evcc.io zusammen mit Deinen Zugangsdaten ;). Es muss einen Unterschied geben...

loebse, 2024-02-10:
Ich habe Dir das Proxyman-Log geschickt, möchte aber meine Zugangsdaten nicht per Mail teilen. Vielleicht können @hyperbart oder @chfaurie nochmal behilflich sein?

Demophobie, 2024-02-11:
@andig Meine Zugangsdaten sind unverändert, falls es hilft.

closed 2024-02-11 by andig