evcc-io/evcc #11772: DSMR parser to accept W, negative consumption and telegrams w/o CRCs

Aeroid, 2024-01-19:
**Is your feature request related to a problem? Please describe.**
The DSMR parser currently seems to
* require CRCs
* assumes "kW" only (i.e. ignores "W" units) and
* requires both "consumption" (1-0:1.7.0) and "delivery" (1-0:2.7.0) and applies substraction.

Telegrams like this don't work:

     /ESY5Q3DB1024 V3.04

    1-0:0.0.0*255(0272031312565)
    1-0:1.8.0*255(00066982.7369255*kWh)
    1-0:2.8.0*255(00025261.5976087*kWh)
    1-0:21.7.0*255(000210.92*W)
    1-0:41.7.0*255(000225.08*W)
    1-0:61.7.0*255(000144.75*W)
    1-0:1.7.0*255(000580.75*W)
    1-0:96.5.5*255(80)
    0-0:96.1.255*255(1ESY1313002565)
    !

**Describe the solution you'd like**
Accept negative consumption values.
Parse units w/o kilo correctly
Accept CRC-less telegrams

**Describe alternatives you've considered**
My current workaround is parsing above telegrams and re-assembiling them in evcc-dsmr

**Additional context**
see also https://github.com/ndokter/dsmr_parser/pull/92

andig, 2024-01-19:
See https://github.com/evcc-io/evcc/blob/master/meter/dsmr.go for which parser library we‘re using. This will in parts need be implemented upstream.

andig, 2024-01-20:
@Aeroid ein einzelner Wert Bezug vs. Lieferung ist umgesetzt. Für die weiteren Änderungen müsstest Du bitte in https://github.com/basvdlei/gotsmart ein Issue oder PR aufmachen. Nach Fix Upstream können wir das hier übernehmen.

closed 2024-01-20 by andig

Aeroid, 2024-01-20:
Danke! Raise ich dort noch mal. 

Mein Workaround liegt solange hier: https://github.com/Aeroid/dsmr-tools