evcc-io/evcc #11785: Orno OR-WE-514 geht seit 0.123.9 nicht mehr

L-Online, 2024-01-21:
### Describe the bug

Hallo,
seit dem Update auf 0.123.9 geht mein Orno OR-WE-514 (Einphasiger Zähler) nicht mehr.
Anscheinend ist die Unterstützung für 
```
  type: modbus
  model: orno1p
```
herausgenommen worden:
```
~ $ evcc meter -l trace

...

[main  ] FATAL 2024/01/21 09:57:14 cannot create meter 'JuwelMeter': invalid meter type: modbus
```

die "neuere" Notation funktioniert aber anscheinend nur mit den dreiphasigen Ornos:
```
  type: template
  template: orno
```

```
~ $ evcc meter -l trace

...

JuwelMeter
----------
[modbus] TRACE 2024/01/21 10:44:09 modbus: send 04 03 00 1c 00 02 05 98
[modbus] TRACE 2024/01/21 10:44:09 modbus: recv 04 83 02 d0 f0
[modbus] TRACE 2024/01/21 10:44:09 modbus: send 04 03 01 08 00 02 44 60
[modbus] TRACE 2024/01/21 10:44:09 modbus: recv 04 83 02 d0 f0
[modbus] TRACE 2024/01/21 10:44:09 modbus: send 04 03 00 16 00 02 25 9a
[modbus] TRACE 2024/01/21 10:44:09 modbus: recv 04 83 02 d0 f0
Power:          read failed: modbus: exception '2' (illegal data address), function '3'
Energy:         read failed: modbus: exception '2' (illegal data address), function '3'
Current L1..L3: read failed: modbus: exception '2' (illegal data address), function '3'

```

Kann ich da was anderes machen, als downgraden?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
- name: ErichMeter
#  type: modbus
  type: template
  template: orno
  modbus: rs485serial
#  model: orno3p
  id: 3
  device: /dev/ttyUSB1 # serial RS485 interface
  baudrate: 9600
  comset: "8E1"
  usage: charge
#  energy: Sum # default value, optionally override
#  power: Power # default value, optionally override
#  currents:
#    - CurrentL1
#    - CurrentL2
#    - CurrentL3

- type: template #modbus
  template: orno
  modbus: rs485serial
#  model: orno1p
  id: 4
  device: /dev/ttyUSB1 # serial RS485 interface
  baudrate: 9600
  comset: "8E1"
#  energy: Sum # default value, optionally override
#  power: PowerL1 # default value, optionally override
  name: JuwelMeter
```

### Log details

```text
ErichMeter
----------
[modbus] TRACE 2024/01/21 10:44:08 modbus: send 03 03 00 1c 00 02 04 2f
[sma   ] TRACE 2024/01/21 10:44:08 send discover package
[sma   ] TRACE 2024/01/21 10:44:08 recv 192.168.178.186: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry]
[modbus] TRACE 2024/01/21 10:44:08 modbus: recv 03 03 04 3d bc 6a 7f 7a fb
[modbus] TRACE 2024/01/21 10:44:08 modbus: send 03 03 01 08 00 02 45 d7
[modbus] TRACE 2024/01/21 10:44:08 modbus: recv 03 03 04 45 4c 5c 00 34 28
[modbus] TRACE 2024/01/21 10:44:08 modbus: send 03 03 00 16 00 02 24 2d
[modbus] TRACE 2024/01/21 10:44:08 modbus: recv 03 03 04 3e a8 f5 c3 52 fa
[modbus] TRACE 2024/01/21 10:44:08 modbus: send 03 03 00 18 00 02 45 ee
[modbus] TRACE 2024/01/21 10:44:08 modbus: recv 03 03 04 3d a3 d7 0a fb 8a
[modbus] TRACE 2024/01/21 10:44:08 modbus: send 03 03 00 1a 00 02 e4 2e
[modbus] TRACE 2024/01/21 10:44:08 modbus: recv 03 03 04 3c a3 d7 0a fa 76
[modbus] TRACE 2024/01/21 10:44:08 modbus: send 03 03 00 0e 00 02 a4 2a
[modbus] TRACE 2024/01/21 10:44:08 modbus: recv 03 03 04 43 63 cc cd a8 fc
[modbus] TRACE 2024/01/21 10:44:08 modbus: send 03 03 00 10 00 02 c4 2c
[modbus] TRACE 2024/01/21 10:44:09 modbus: recv 03 03 04 43 66 80 00 4d a8
[modbus] TRACE 2024/01/21 10:44:09 modbus: send 03 03 00 12 00 02 65 ec
[modbus] TRACE 2024/01/21 10:44:09 modbus: recv 03 03 04 43 65 19 9a 57 93
Power:          92W
Energy:         3269.8kWh
Current L1..L3: 0.33A 0.08A 0.02A
Voltage L1..L3: 228V 230V 229V

JuwelMeter
----------
[modbus] TRACE 2024/01/21 10:44:09 modbus: send 04 03 00 1c 00 02 05 98
[modbus] TRACE 2024/01/21 10:44:09 modbus: recv 04 83 02 d0 f0
[modbus] TRACE 2024/01/21 10:44:09 modbus: send 04 03 01 08 00 02 44 60
[modbus] TRACE 2024/01/21 10:44:09 modbus: recv 04 83 02 d0 f0
[modbus] TRACE 2024/01/21 10:44:09 modbus: send 04 03 00 16 00 02 25 9a
[modbus] TRACE 2024/01/21 10:44:09 modbus: recv 04 83 02 d0 f0
Power:          read failed: modbus: exception '2' (illegal data address), function '3'
Energy:         read failed: modbus: exception '2' (illegal data address), function '3'
Current L1..L3: read failed: modbus: exception '2' (illegal data address), function '3'
```

### What type of operating system are you running?

Linux

### Version

0.123.9

StefanSchoof, 2024-01-21:
`type: modbus` ist jetzt `type: mbmd`, siehe https://github.com/evcc-io/evcc/discussions/11747

L-Online, 2024-01-21:
Super, danke!

Bei mir hat es erst funktioniert, nachdem ich die Zeile auch noch gelöscht habe:
`   modbus: rs485serial
`
So funktioniert es bei mir jetzt:
```
- type: mbmd
# type: modbus
#  template: orno
#  modbus: rs485serial
  model: orno1p
  id: 4
  device: /dev/ttyUSB1 # serial RS485 interface
  baudrate: 9600
  comset: "8E1"
  energy: Sum # default value, optionally override
  power: PowerL1 # default value, optionally override
  name: JuwelMeter
```
Hier ist die Doku noch nicht angepasst:
[https://docs.evcc.io/docs/reference/configuration/meters/#modbus](https://docs.evcc.io/docs/reference/configuration/meters/#modbus)

premultiply, 2024-01-21:
Dann passe sie doch bitte gerade an. Danke!

andig, 2024-01-22:
Schau mal hier: https://github.com/evcc-io/evcc/discussions/11747

closed 2024-01-22 by andig