evcc-io/evcc #11787: browser-Anzeige evcc stimmt nicht wenn tibber plan definiert wurde

WUhle, 2024-01-21:
### Describe the bug

Ich habe die beiden Anzeigen von der evcc- und Fronius Wechselrichterbilanz in ein Bild kopiert. laut evcc wird der Haushalt aus der Batterie versorgt, was falsch ist. die Sonne speist komplett den Haushaltsbedarf und die Batterie lädt.  
Ich vermute das dies irgendwie mit der Verwaltung der dynamischen Tarife zusammenhängt.
Der Ladezustand der Batterie ist laut Fronius 10% weiter ladend und in evcc immer 8% entladend. Alles bezieht sich auf v0.123.9
[image]
Der Plan ist zwar eingebbar aber die Istanzeige scheint eingefroren. Der Log über Status ist komisch.

### Steps to reproduce

1.
2.
3.
...einfach zwei Anzeigen vergleichen.

### Configuration details

```yaml
fronius symo GEN 24 mit 10kwh BYD HVS und angeschlossenem IONIQ 5. Laden über nrgkick (bluetooth) 3Pahsig. Tibber dynamischer Tarif und Plan für knapp 2 Stunden ab 13.00 Uhr auf 90% SOC am Auto zu kommen.
evcc läuft auf einem rpi 4
```

### Log details

```text
● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; preset: enabled)
     Active: active (running) since Sat 2024-01-20 11:47:47 CET; 23h ago
   Main PID: 2684451 (evcc)
      Tasks: 11 (limit: 3933)
        CPU: 2min 59.796s
     CGroup: /system.slice/evcc.service
             └─2684451 /usr/bin/evcc

Jan 20 17:28:50 IoBSVHome evcc[2684451]: time="2024-01-20T17:28:50+01:00" level=warning msg="MapToStruct: invalid field detected *device.Device1Properties.Bonded"
Jan 20 17:28:50 IoBSVHome evcc[2684451]: time="2024-01-20T17:28:50+01:00" level=warning msg="MapToStruct: invalid field detected *device.Device1Properties.Bonded"
Jan 20 17:28:53 IoBSVHome evcc[2684451]: time="2024-01-20T17:28:53+01:00" level=warning msg="MapToStruct: invalid field detected *device.Device1Properties.Bonded"
Jan 20 17:28:53 IoBSVHome evcc[2684451]: time="2024-01-20T17:28:53+01:00" level=warning msg="MapToStruct: invalid field detected *device.Device1Properties.Bonded"
Jan 20 23:29:32 IoBSVHome evcc[2684451]: [lp-1  ] DEBUG 2024/01/20 23:29:32 set session soc limit: 85
Jan 20 23:29:34 IoBSVHome evcc[2684451]: [lp-1  ] DEBUG 2024/01/20 23:29:34 set session soc limit: 85
Jan 20 23:29:34 IoBSVHome evcc[2684451]: [lp-1  ] DEBUG 2024/01/20 23:29:34 set session soc limit: 85
Jan 20 23:29:36 IoBSVHome evcc[2684451]: [lp-1  ] DEBUG 2024/01/20 23:29:36 set session soc limit: 85
Jan 21 11:13:23 IoBSVHome evcc[2684451]: [lp-1  ] DEBUG 2024/01/21 11:13:23 set session soc limit: 90
Jan 21 11:16:55 IoBSVHome evcc[2684451]: [site  ] DEBUG 2024/01/21 11:16:55 set ev5 plan soc: 90 @ 2024-01-21 14:07:00 +0100 CET
```

### What type of operating system are you running?

Linux

### Version

v0.123.9

StefanSchoof, 2024-01-21:
Vermutlich ein Fehler bei deinen meter. 

Am besten du postest einmal deine Config: https://docs.evcc.io/docs/guides/setup/#etwas-funktioniert-nicht-was-nun

WUhle, 2024-01-21:
Nein, es lief jaschon seit 1.1.2024 (tibber-start) ausgezeichnet und jetzt nach 0.123.9+1705803575 einspielen ist auch wieder alles im Lot. Selbst der in der Vorversion definierte Plan wurde übernommen. Da ist irgendwo ein dynamischer Klemmer. Aktuell ist die Anzeige evcc wieder konsistent zum Istzustand des Wechselrichters und des Fronius-Meters. 
Ich denke die nächtliche Sequenz das Ladelimit zu verstellen, hat die Istanzeige irgendwie geerdet.

WUhle, 2024-01-21:
Aktuell läuft das Billigladen für knapp zwei Stunden:

● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-01-21 11:56:40 CET; 6min ago
   Main PID: 2790756 (evcc)
      Tasks: 10 (limit: 3933)
        CPU: 2.966s
     CGroup: /system.slice/evcc.service
             └─2790756 /usr/bin/evcc

Jan 21 12:02:39 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:02:39 charge total import: 9693.424kWh
Jan 21 12:02:41 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:02:41 charger status: C
Jan 21 12:02:43 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:02:43 charge start energy: 9.69e+03kWh
Jan 21 12:02:43 IoBSVHome evcc[2790756]: [lp-1  ] INFO 2024/01/21 12:02:43 start charging ->
Jan 21 12:02:43 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:02:43 wake-up timer: stop
Jan 21 12:02:43 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:02:43 soc estimated: 62.00% (vehicle: 62.00%)
Jan 21 12:02:43 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:02:43 vehicle soc: 62%
Jan 21 12:02:43 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:02:43 vehicle soc limit: 100%
Jan 21 12:02:43 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:02:43 vehicle range: 232km
Jan 21 12:02:45 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:02:45 plan: charge 1h57m51s starting at 2024-01-21 12:02:45 +0100 CET until 2024-01-21 14:07:00 +0100 C>
~
pi@IoBSVHome:~ $ systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-01-21 11:56:40 CET; 6min ago
   Main PID: 2790756 (evcc)
      Tasks: 10 (limit: 3933)
        CPU: 3.145s
     CGroup: /system.slice/evcc.service
             └─2790756 /usr/bin/evcc

Jan 21 12:03:04 IoBSVHome evcc[2790756]: [site  ] DEBUG 2024/01/21 12:03:04 site power: 10598W
Jan 21 12:03:05 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:03:05 charge currents: [15.8 15.7 15.7]A
Jan 21 12:03:05 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:03:05 detected active phases: 3p
Jan 21 12:03:09 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:03:09 charge total import: 9693.512kWh
Jan 21 12:03:11 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:03:11 charger status: C
Jan 21 12:03:11 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:03:11 soc estimated: 62.09% (vehicle: 62.00%)
Jan 21 12:03:11 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:03:11 vehicle soc: 62%
Jan 21 12:03:11 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:03:11 vehicle soc limit: 100%
Jan 21 12:03:11 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:03:11 vehicle range: 232km
Jan 21 12:03:13 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:03:13 plan: charge 1h57m28s starting at 2024-01-21 12:03:13 +0100 CET until 2024-01-21 14:07:00 +0100 C>
pi@IoBSVHome:~ $
pi@IoBSVHome:~ $
pi@IoBSVHome:~ $ systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-01-21 11:56:40 CET; 9min ago
   Main PID: 2790756 (evcc)
      Tasks: 10 (limit: 3933)
        CPU: 4.008s
     CGroup: /system.slice/evcc.service
             └─2790756 /usr/bin/evcc

Jan 21 12:05:29 IoBSVHome evcc[2790756]: [site  ] DEBUG 2024/01/21 12:05:29 site power: 10270W
Jan 21 12:05:31 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:05:31 charge currents: [15.6 15.6 15.5]A
Jan 21 12:05:31 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:05:31 detected active phases: 3p
Jan 21 12:05:35 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:05:35 charge total import: 9693.939kWh
Jan 21 12:05:37 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:05:37 charger status: C
Jan 21 12:05:37 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:05:37 soc estimated: 62.62% (vehicle: 62.00%)
Jan 21 12:05:37 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:05:37 vehicle soc: 63%
Jan 21 12:05:37 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:05:37 vehicle soc limit: 100%
Jan 21 12:05:37 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:05:37 vehicle range: 232km
Jan 21 12:05:39 IoBSVHome evcc[2790756]: [lp-1  ] DEBUG 2024/01/21 12:05:39 plan: charge 1h55m19s starting at 2024-01-21 12:05:39 +0100 CET until 2024-01-21 14:07:00 +0100 CET

WUhle, 2024-01-21:
Ich habe im laufenden Ladeprozess mit ursprünglichen Limit 90% eine Ladeendzeit von 14.07 Uhr angezeigt bekommen. Ich habe das Limit auf 85% im Plan reduziert und die Plan-Nutzung hat dies auch übernommen, allerding nicht die jetzt reduzierte Endzeitangabe. Beispiel:
[image]

closed 2024-01-22 by andig

WUhle, 2024-01-22:
Ich wollte mir nicht immer wieder die Mühe machen alle Credencials zu editieren. Nun die config-yaml ohne PIN, Passworten und Tokens:

# open evcc at http://evcc.local:7070
sponsortoken: <sponsor-token>.

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: dda0b39e2a9c7ef84af690c31250f0f8353dce286c373c19affe9aadb609e407

interval: 29s # control cycle interval

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: GW15/evcc # root topic for publishing, set empty to disable publishing
  clientid: evccGW15
  user: aquser
  password: mosquittoPW

meters:
- type: template
  template: fronius-gen24
  usage: grid
  host: 192.168.------
  port: 502
  name: Mgrid
- type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.-------
  port: 502
  name: 81pv
- type: template
  template: fronius-gen24
  usage: battery
  host: 192.168.------
  port: 502
  name: BYDbattery

chargers:
- type: template
  template: nrgkick-bluetooth
  mac: '00:1E::::77:81'
  pin: 1234
  name: NRGkick

vehicles:
- type: template
  template: hyundai
  title: IONIQ 5 WU203R
  user: 'wu@sv-uhlenberg.de'
  password: 'bluelinkPW'
  vin: K-------------7
  capacity: 72
  phases: 3
  icon: car
  cache: 15m
#  mode: pv
#  minSoc: 10
#  targetSoc: 99
  minCurrent: 6
  maxCurrent: 17
  language: en
  name: ev5

loadpoints:
- title: "EVC-Garage"
  charger: NRGkick
  mode: pv # Nur Überschuss aus PV oder BYD
  phases: 3
  mincurrent: 6
  maxcurrent: 18
#  resetOnDisconnect: false
#  prioritySoc: 50 # Hausbatterie bekommt bis zum Soc 50% Priorität beim laden
#  bufferSoc: 70 # Hausbatterie wird oberhalb Soc 70% als Puffer genutzt
#  bufferStartSoc: 80 # hat die Hausbatterie Soc 80& erreicht startet der ladevorgang
#  residualPower: 290 # 290W Hausverbrauch

# dynamischer Strombezugstarif bei Tibber

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: tibber
    token: "aOleh8--tibber token ----- R0E4" # access token wu@sv-u

  feedin:
    type: fixed
    price: 0.08 # [currency]/kWh

site:
  title: "GW15-27252"
  meters:
    grid: Mgrid
    pv:
      - 81pv
    battery:
      - BYDbattery
#  prioritySoc: 50 # Hausbatterie bekommt bis zum Soc 50% Priorität beim laden
#  bufferSoc: 70 # Hausbatterie wird oberhalb Soc 70% als Puffer genutzt
#  bufferStartSoc: 80 # hat die Hausbatterie Soc 80& erreicht startet der ladevorgang
  residualPower: 290 # 290W Hausverbrauch

andig, 2024-01-22:
> Ich wollte mir nicht immer wieder die Mühe machen alle Credencials zu editieren. 

Dafür gibts

    evcc discuss

dann ist es sogar gleich lesbar formatiert.

WUhle, 2024-01-28:
Output von evcc discuss zur yaml:
<details><summary>Konfiguration (/etc/evcc.yaml)</summary>

```yaml
# open evcc at http://evcc.local:7070
sponsortoken: *****

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 29s # control cycle interval

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: GW15/evcc # root topic for publishing, set empty to disable publishing
  clientid: evccGW15
  user: *****
  password: *****

meters:
- type: template
  template: fronius-gen24
  usage: grid
  host: 192.168.53.141
  port: 502
  name: Mgrid
- type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.53.141
  port: 502
  name: 81pv
- type: template
  template: fronius-gen24
  usage: battery
  host: 192.168.53.141
  port: 502
  name: BYDbattery

chargers:
- type: template
  template: nrgkick-bluetooth
  mac: *****
  pin: *****
  name: NRGkick

vehicles:
- type: template
  template: hyundai
  title: IONIQ 5 WU203R
  user: *****
  password: *****
  vin: *****
  capacity: 72
  phases: 3
  icon: car
  cache: 15m
#  mode: pv
#  minSoc: 10
#  targetSoc: 99
  minCurrent: 6
  maxCurrent: 17
  language: en
  name: ev5

loadpoints:
- title: "EVC-Garage"
  charger: NRGkick
  mode: pv # Nur Überschuss aus PV oder BYD
  phases: 3
  mincurrent: 6
  maxcurrent: 18
#  resetOnDisconnect: false
#  prioritySoc: 50 # Hausbatterie bekommt bis zum Soc 50% Priorität beim laden
#  bufferSoc: 70 # Hausbatterie wird oberhalb Soc 70% als Puffer genutzt
#  bufferStartSoc: 80 # hat die Hausbatterie Soc 80& erreicht startet der ladevorgang
#  residualPower: 290 # 290W Hausverbrauch

# dynamischer Strombezugstarif bei Tibber

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: tibber
    token: *****

  feedin:
    type: fixed
    price: 0.08 # [currency]/kWh

site:
  title: "GW15-27252"
  meters:
    grid: Mgrid
    pv:
      - 81pv
    battery:
      - BYDbattery
#  prioritySoc: 50 # Hausbatterie bekommt bis zum Soc 50% Priorität beim laden
#  bufferSoc: 70 # Hausbatterie wird oberhalb Soc 70% als Puffer genutzt
#  bufferStartSoc: 80 # hat die Hausbatterie Soc 80& erreicht startet der ladevorgang
  residualPower: 290 # 290W Hausverbrauch

```

</details>
Version: `0.123.9 (f6f4e30f)`