evcc-io/evcc #11789: Chromium -> Sprachauswahl absturz

ANforEVCC, 2024-01-21:
### Describe the bug

Ich nutze Fullpage-OS, um auf einem Raspi 3B mit angeschlossenem Display das EVCC Interface anzuzeigen.

Leider wird der Browser nicht von EVCC erkannt, so das die automatische Sprachauswahl nicht funktioniert.
Beim Versuch, die Browsersprache in der EVCC Gui zu ändern, stürzt die GUI ab. (Beim Klick auf "Automatisch" öffnet sich ein weißes Feld und die GUI ist tot)

Am Edge-Browser funktioniert das problemlos, also die automatische Sprachauswahl und das Dropdaun-Menue für die manuelle Sprachauswahl.

Gibt es einen Fix oder lässt dich die Sprache per Parameter in der Config oder im Browseraufruf festlegen?

### Steps to reproduce

FullpageOS auf Raspi installieren
Raspi mit angeschlossenem Display booten, EVCC über http-Call aufrufen

### Configuration details

```yaml
n/a
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.123.9

andig, 2024-01-22:
Leider fehlen Details. Gibts Fehler in der Browserkonsole? Logfile? Screenshot?

ANforEVCC, 2024-01-22:
Leider stürzt der Browser einfach nur kommentarlos ab.
Beim benutzten Betriebssystem gibt es leider keine Konsole (BS für Digital-Signage, ruft einfach nur eine Webseite auf dem angeschlossenen Touch-Display auf)

Wirkt für mich wie ein HTML-Problem.

Daher die Bitte, die Sprache per Parameter in der Konfig festsetzen zu können.

andig, 2024-01-22:
@naltatis können wir da noch irgendwas machen? YAML-Settings als Workaround für abstürzende Browser sind sicher kein Lösungsweg.

naltatis, 2024-01-22:
@andig Nein, nen Workaround über ne fixe Sprache in der yaml würd ich nicht machen. Da hängt sehr viel dran. Siehe vorheriger PR.

@ANforEVCC hast du Kontrolle über den Browser? evcc wählt die Sprache anhand der im Browser definierten Sprachliste (navigator.language). Vmtl kannst du die mit startparametern oder Konfiguration setzen.

naltatis, 2024-01-22:
Bin nicht tief genug im FullpageOS drin. Aber da scheints Override Möglichkeiten zu geben:

https://github.com/guysoft/FullPageOS/blob/762c5d886162e4ab764287cea897541aeb6f0f9d/src/modules/fullpageos/config#L5-L8

naltatis, 2024-01-24:
I've reopened because there might actually be a way to at least solve the "tesla == nl" case (see https://github.com/evcc-io/evcc/issues/10459) and maybe this as well by reading accept-language header

g-mocken, 2024-01-24:
The Tesla Chrome browser does in fact (for me in Germany) send a correct "Accept-Language”-Header "de" (checked with https://www.whatismybrowser.com/detect/what-http-headers-is-my-browser-sending), however it's JavaScript engine seems to be fixed to "nl-NL" (checked by running the JS-snippet on https://stackoverflow.com/questions/8199760/how-to-get-the-browser-language-using-javascript).

PR #11867 (2024-01-26): Header-based auto-language detection
Fix #11789, fix #10459

Use `Accept-Language` header (Go) instead of `navigator.language(s)` (JS) for "Language: automatic" setting in UI.
closed 2024-01-26 by andig